lvgl/lvgl #4624: More life-like benchmark

kisvegabor, 2023-10-04:
### Introduce the problem

From the current benchmark it's hard to tell the  system's performance with a real UI. Besides it contains to many cases and hard to see the bottlenecks.

### Suggested solution

I suggest renaming the current `lv_demo_benchmark` to `lv_demo_draw_test` and create a new shorter and more targeted benchmark. I can imagine these steps:
 
1. **Empty screen, change background color**: measure fill rate and also acts as a reference FPS 
2. **Moving a screen sized image** (can be tiled to save flash): it shows the memory read performance too
3. **Empty screen with small rectangle changing color**: show partial refreshing performance
4. **A smaller ARGB image is rotated** image transformation performance 
4.  **Several moving containers with small radius, icon and texts**: real life UI test
5. **Same as 5) with with a semi transparent overlay changing color**: show screen size semi-transparent rendering for modals
5. **Same as 5) but the widget shas animated opa**: small opacity performance
6. **Same as 5) but the widgets has animated opa_layer**: layer rendering performance. 
7. **Same as 5) but widget as rotate** show layer performance with layer rotation
8. **A lot of text in a screen size container scrolled** large text rendering performance
9. **A lot of simple list items scrolled** test scrolling many widgets
11. **Create 200 buttons and delete them** test widget creation speed 
11. **A balanced UI with some widgets on tabview and changing tabs**: something like the current widgets demo.

As a result, for each step we should display:
- Time between two render starts (in includes flushing, layout recalculation, etc)
- The same as FPS as this is the perceived performance
- Pure rendering time without flushing
- Time of flushing

What do you think?

cc @GorgonMeducer @nicusorcitu

bjsylvia, 2023-10-05:
Can you also provide some test cases for measuring the performance of parallel rendering feature?

nicusorcitu, 2023-10-05:
1. That would be indeed an benchmark closer to the real life application. It will be very good to evaluate the performance overall.
Will be nice to have also few tests similar with the existing benchmark from v8 so that we can compare v8 vs v9. I will keep at least fill and blit from v8: simple rectangles (not rounded) and blit img RGB565/ARGB8888.

2. I would add CPU load to the benchmark results. It is important to see the accelerator benefit.
For that we might need to modify a bit the way of calculating the CPU usage - maybe having something OS specific will be really nice. For FreeRTOS I already have the support to get the CPU Idle time.

GorgonMeducer, 2023-10-05:
@kisvegabor What you have listed looks more like functional tests. In addition, for scenario-based benchmark tests, I‘d like to propose some scenarios like the following:

- Watch faces (for smartwatch)
    - rotating 3 pointers
    - a dynamic background (below the 3 pointers)
    - Digits (A8 or A4) to show 10 milliseconds (0.xx Sec)

- A Calendar / Themostatus Panel (for smart controller / white goods) 
- A Smart Meter Panel (for industrial applications)
- A  Dashboard Panel (for automobile-like applications)

> I would add CPU load to the benchmark results. It is important to see the accelerator benefit.
For that we might need to modify a bit the way of calculating the CPU usage - maybe having something OS specific will be really nice. For FreeRTOS I already have the support to get the CPU Idle time.

Good idea. Because we cannot control the target software system that might have a lot of application tasks, measuring the RTOS idle time will be affected by many factors out of our control.   Instead of measuring the CPU Idle time, I would suggest measuring 1) the average CPU cycles used by the render for each frame and 2) intervals between each frame. Using the [(average CPU Cycles of each frame )/(Intervals between each frame)} * 100%, we can get clean CPU usage info (based on percentage).

nicusorcitu, 2023-10-05:
> I would suggest measuring 1) the average CPU cycles used by the render for each frame
>
Is not measuring also the GPU time? part of the frame is done by GPU.
I know what you mean, that the CPU does not wait any-longer for the GPU. But is not true in all the cases.
Imagine the very first draw task is a fill screen (with the gray light color). This is a task that can be done by the GPU and could take up to 3 ms. During this time, the CPU is idle, all the other tasks are waiting for the fill to complete because they are the screen children.
3ms from a frame time of 30ms means 10% difference.

GorgonMeducer, 2023-10-05:
> During this time, the CPU is idle, all the other tasks are waiting for the fill to complete because they are the screen children.

Since the CPU is idle, that means the LVGL doesn't take the CPU time and this CPU time can be used by applications (i.e. other tasks). So I don't think taking GPU time into consideration is necessary.

kisvegabor, 2023-10-05:
Based on your suggestions I've updated the first comment. Let me know what you think.

> Good idea. Because we cannot control the target software system that might have a lot of application tasks, measuring the RTOS idle time will be affected by many factors out of our control.

I think during benchmarking we can assume that there are no many running application tasks. Or if there are, the benchmark will show the real CPU usage, e.g. when due to the running application tasks the CPU usage is 30% by default.

nicusorcitu, 2023-10-06:
> Here, we are measuring CPU load, not rendering time for each frame. They are different.
>
Exactly my point. When I say task I mean graphic draw task.
We are running only the benchmark app that will report CPU usage as well. Getting the CPU load from OS Idle task is ideal in this case. We can not simply use a start frame time and stop frame time to measure the elapsed time of the frame, just because part of the frame is handled by the GPU.

kisvegabor, 2023-10-06:
> We can not simply use a start frame time and stop frame time to measure the elapsed time of the frame, just because part of the frame is handled by the GPU.

What do you mean by stop frame time? `FLUSH_START` (render ready), or `FLUSH_END` (the display is fully updated).

BTW, now we have [a lot of events](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_event.h#L105-L112) on the displays for various stages of rendering, however `LV_EVENT_FLUSH_FINISH` might be misleading. It's fired when the `flush_cb` returns and not when `lv_display_flush_ready()` is called. As  `lv_display_flush_ready()` can be called from an interrupt, we cannot send an event to the display from there. So we should figure out something else for it, like saving a `flush_ready_timestamp` in `lv_display_t`.

nicusorcitu, 2023-10-06:
> What do you mean by stop frame time?
>
I mean start draw frame and stop draw frame - that we had it in [lvgl 8.x](https://github.com/lvgl/lvgl/blob/73fa50afa0f075a0ff5875ee6cd314b6404967a1/src/core/lv_refr.c#L348)
I see that now it changed, so in v9 the frame draw time would be between the events `LV_EVENT_REFR_START` and `LV_EVENT_RENDER_READY`. But again, this time is composed by CPU time + GPU time.

kisvegabor, 2023-10-06:
Ah I see, and yes, it works as you hava described in v9.

GorgonMeducer, 2023-10-06:
In general, for CPU Load, the method I proposed has the following advantages:

- No dependency on RTOS, which means we don't have to rely on certain RTOS porting
- The two fundamental values are easy to get:
    * The average CPU time used for rendering one frame
    * The average intervals between each frame
- The generated CPU usage info is purely for the LVGL

I would call the CPU usage info generated in the RTOS Idle task the Overall CPU Usage.

**I suggest that we provide both the Overall CPU Usage and the LVGL CPU Usage at the same time.** 

It is very useful for system performance/workload analysis. For example, suppose the Overall CPU Usage is 80%, and the pure LVGL CPU Usage is 30%; then the developer can immediately understand that they should not blame LVGL for consuming the big trunk of CPU time and focusing on optimizing the Application part. Suppose we don't provide the pure LVGL CPU Usage info, then the developer will have trouble understanding whether it is LVGL or other application tasks that consume so much CPU time.

In brief, the overall CPU and the pure LVGL load information are important for performance analysis. If we don't provide both, the customers will be forced to find that information by themselves anyway if they have concerns about the CPU load info.

GorgonMeducer, 2023-10-06:
> Based on your suggestions I've updated the first comment. Let me know what you think.

It is great. I would suggest other scenarios:

- Text Reader scenario
    * Big Font Size
    * Text takes the full screen
    * Text scrolling down slowly (full-screen flushing)
    * Floating Navigation Bar (buttons, progress bar etc.)

- Navigation Scenario
    - Showing part of a map on the full screen in a visual window
    - Move and rotate the visual windows on the map following the movement of the Person in the centre of the visual window
    - This scenario shows the performance of rotation, scaling and perspective transform (optionally)

- MP3 player Scenario
    - Showing Waveform
    - Showing FFT
    - Showing lyrics (scrolling up slowly according to the music)
    - Showing a rotating record cover ( ARGB rotation )

kisvegabor, 2023-10-08:
What if during rendering LVGL's task was interrupted by a higher priority long lasting application task. If just measure the time of rendering we measure the time spent in the application task as well. 

> Text Reader scenario

I'm sure if it will have so much added value assuming we will have these:
> **Multiple text moving** text rendering + parallel rendering
> **A lot of text in a screen size container scrolled** large text rendering performance

> Navigation Scenario

Full screen transformation can be really interesting.

> MP3 player Scenario

I like it! Rotating a circle (not a line, like a clock handle) is really interesting. A good looking line and bar chart drawing is also great! However, I think the lyrics part is covered by the text tests I mentioned above.

nicusorcitu, 2023-10-09:
When we say CPU load we mean the CPU load of the system. All the time.

If we look at the CPU load while running the Benchmark App only, we know that graphic application is running, so there are no extra OS tasks to run. In Benchmark we can have a separate row with CPU load next to the FPS so we can see the difference between CPU only and CPU + GPU. CPU load can be different while running different benchmark tests.

If we run any custom application + graphic application and we look at the CPU load (by enabling `LV_USE_PERF_MONITOR`) then we assume that the CPU load can not be only the CPU used by draw app tasks but also for other tasks.

Either way, the CPU load must be reported correctly. And the only way to do that is to taken it from OS idle task. (if any OS is enabled)

GorgonMeducer, 2023-10-09:
@nicusorcitu 
OK, I give up.

nicusorcitu, 2023-10-09:
Sorry, I do not mean to have the final decision. I don't have to. :)

I am only trying to signal some issues that I faced and the solution I found while determining the CPU load while running the benchmark app in LVGL v8.

GorgonMeducer, 2023-10-09:
> Sorry, I do not mean to have the final decision. 

Don't get me wrong. You don't have to feel sorry. I give up because I am sure you understand my points, and it is an effective communication. My purpose is the same as you, to express what I have seen and my thoughts about LVGL. 

> What if during rendering LVGL's task was interrupted by a higher priority long lasting application task. If just measure the time of rendering we measure the time spent in the application task as well.

Actually, there are methods to remove the noise introduced by RTOS scheduling. It depends on whether you want to try this method, and we can have a dedicated discussion about it. If in general, you don't want a PURE LVGL load info, then we don't have to go that deep.

nicusorcitu, 2023-10-09:
If the purpose is to measure the CPU load ONLY by LVGL tasks this can be achieved very easy on FreeRTOS by hacking the `traceTASK_SWITCHED_IN()` and `traceTASK_SWITCHED_OUT()` and measure the cycles only by lvgl main task and lvgl draw tasks.

If you want to see the CPU idle time then the same mechanism can be applied to measure the cycles spend in Idle task. Then CPU load = 100 - CPU Idle percentage. This will be the overall CPU load of the system.

Measuring the time between the events `LV_EVENT_REFR_START` and `LV_EVENT_RENDER_READY` will not give us any clue about CPU time if there is an additional draw unit (GPU is also enabled).

GorgonMeducer, 2023-10-09:
> If the purpose is to measure the CPU load ONLY by LVGL tasks this can be achieved very easy on FreeRTOS by hacking the traceTASK_SWITCHED_IN() and traceTASK_SWITCHED_OUT() and count the cycles only by lvgl main task and lvgl draw tasks.

Yes, that's method I wanted to say. In fact, almost all mainstream RTOS provide similar hooking interfaces (including RT-Thread).

kisvegabor, 2023-10-10:
I'm planning to start implementing the "Basic tests" and "Complex tests" next week and use the widget demo first as a real life example. After that we can gradually add more real life examples and fine time measurements methods. 

What do you think?

bjsylvia, 2023-10-14:
If you implement it as a real life example, could you also consider incorporating the ability for the demo to exit gracefully? This scenario is crucial in multi-instance versions.

XuNeo, 2023-10-14:
Hi Gabor,
For watch face example, we can pick some from Xiaomi Watch S1 pro. I have talked with designers that we can use the original PNG files if you think they are appropriate as benchmark.

kisvegabor, 2023-10-16:
> If you implement it as a real life example, could you also consider incorporating the ability for the demo to exit gracefully?

Sure!

> Hi Gabor,
For watch face example, we can pick some from Xiaomi Watch S1 pro. I have talked with designers that we can use the original PNG files if you think they are appropriate as benchmark.

Having a watch face from S1 pro would be really nice. I'll need a written notice about it's really allowed by Xiaomi :slightly_smiling_face:

We need to keep in mind that the demo should support not only square/round, but landscape and portrait displays as well. In this case the watch face should be displayed in the middle of the screen, but the other pages should be more responsive.

kisvegabor, 2023-10-16:
I've created the basic framework for the benchmark. See [here](https://github.com/lvgl/lvgl/tree/feat/new-benchmark). I've updated `lv_sysmon` and now any modules can subscribe to the performance data. Finally it's super clean I'm really happy with it. 

Now it shows only some basic performance data which is now available, but we can improve it later.

nicusorcitu, 2023-10-17:
I am very busy 2-3 weeks but I will get back to lvgl asap and I will let you know.

kisvegabor, 2023-10-17:
Okay, thanks for letting us know. In 2-3 weeks I'm sure we will have something that you can try out when you are back. :slightly_smiling_face:

kisvegabor, 2023-10-18:
I've pushed the basic and complex tests to https://github.com/lvgl/lvgl/tree/feat/new-benchmark

Although it's in early stage, please check it out, try it out and give me feedback. :slightly_smiling_face:

XuNeo, 2023-10-19:
> I'll need a written notice about it's really allowed by Xiaomi

No problem, I will write the notice and email to you.

kisvegabor, 2023-10-25:
I've merged the first version of the new benchmark into master: a61d87f84

nicusorcitu, 2023-10-30:
I just tried to run the benchmark, CPU only, on the target and I get the below assert.
One more note about logging: I have seen that logs are nly printed from the main lvgl thread, any other log happening in draw thread will not get print.
Shall we have a callback register set also in the draw thread?
```
lv_log_register_print_cb(print_cb);
```

[image]

kisvegabor, 2023-10-30:
> I just tried to run the benchmark, CPU only, on the target and I get the below assert.

Please attach your `lv_conf.h`

> One more note about logging: I have seen that logs are nly printed from the main lvgl thread, any other log happening in draw thread will not get print.

LVGL just calls the registered `print` function. Maybe a `flush` needs to be added too.

nicusorcitu, 2023-10-30:
> LVGL just calls the registered print function. Maybe a fflush needs to be added too.
>
The idea is that it gets registered _only_ for the thread where the registration api `lv_log_register_print_cb` is called.

nicusorcitu, 2023-10-30:
Attached lv_conf.h (txt extension)
[lv_conf.txt](https://github.com/lvgl/lvgl/files/13204833/lv_conf.txt)

kisvegabor, 2023-10-30:
It's working on my end. At which step did it crash for you? Can you try with `LV_OS_NONE` and pure SW render?

nicusorcitu, 2023-10-30:
I feel there is something more. Sometimes I receive a wrong task from dispatcher - which is handled only by the lvgl main thread. I put one breakpoint at the beginning of `_vglite_execute_drawing` (in VGLite thread) and `execute_drawing` (in SW thread). The second task is invalid, I remember I have see this in the past where `lv_draw_get_next_available_task` returned invalid task...
```
(gdb) p *t
$2 = {next = 0x0 <lv_refr_now>, type = LV_DRAW_TASK_TYPE_FILL, area = {x1 = 0, y1 = 0, x2 = 719, y2 = 1279}, clip_area_original = {x1 = 0, y1 = 0, x2 = 0, y2 = 0}, clip_area = {x1 = 0, y1 = 0, x2 = 719, y2 = 1279}, state = 2, draw_dsc = 0x80386538 <work_mem_int+7880>, preferred_draw_unit_id = 2 '\002', preference_score = 80 'P'}
(gdb) 
(gdb) p *t
$4 = {next = 0x804160f0 <ucHeap+71288>, type = 161, area = {x1 = 12290, y1 = 21444, x2 = -32712, y2 = 24816}, clip_area_original = {x1 = -32703, y1 = 28, x2 = 0, y2 = -10727}, clip_area = {x1 = 12290, y1 = -9747, x2 = 12290, y2 = -9103}, state = 805734821, draw_dsc = 0x0 <lv_refr_now>, preferred_draw_unit_id = 0 '\000', preference_score = 0 '\000'}
```

kisvegabor, 2023-10-30:
Oh, it's really strange: 
- Have you seen this issue with SW render only as well?
- Do you see it `LV_OS_NONE`?
- Do you do some RTOS calls from interrupt, or could it be an issue that you call `lv_*` functions from interrupt?
-

kisvegabor, 2023-10-30:
@nicusorcitu please also try enabling all asserts and LV_MEM_ADD_JUNK in lv_mem.c.

nicusorcitu, 2023-10-31:
I think I found the issue [here](https://github.com/lvgl/lvgl/blob/e08314df940413f6de08ee3c05c6f5b3bd0e7671/src/draw/lv_draw.c#L297)
It might be that both t_prev and layer->draw_task_head to be null, and the task will be wrong at the end.

kisvegabor, 2023-10-31:
It just came to my mind that if `lv_draw_get_next_available_task` is called in parallel it cab mess up the task list which isa common resource (used by all draw units). 

Can you try adding a mutex in `lv_draw_get_next_available_task`?

nicusorcitu, 2023-10-31:
I was thinking on that one but is not the case.
`lv_draw_get_next_available_task` is called from sw_dispatcher or vglite_dispatcher functions.
But both dispatcher functions are called from the same thread, lvgl_main thread.
https://github.com/lvgl/lvgl/blob/e08314df940413f6de08ee3c05c6f5b3bd0e7671/src/draw/lv_draw.c#L248

kisvegabor, 2023-10-31:
True!
Is there a test which I can use to reproduce the issue?

nicusorcitu, 2023-10-31:
Just the benchmark test I run with VgLite. (no matter the Sw thread exist or not).
I can not figure out what is happening, I do not get a consistent behavior.
Right now I comment out all the tests except the "Empty screen" and it seems to run complete, hence I see nothing on the screen - screen just lights on on black color.

nicusorcitu, 2023-10-31:
I just test with a simple blit image (vglite only) and there is nothing on the screen even the tasks have been completed successfully.

kisvegabor, 2023-10-31:
It seems the opacity of fill is not correct for some reason. Let's have a very simple UI: empty red screen and a blue rectangle. 
1. Do you see both draw tasks in the debugger?
2. Is the opacity of both set to 0xff?
3. Do you see any issue (flickering or crash) if you call `lv_obj_invalidate(lv_scr_act())` right after `lv_timer_handler()`?
4. Can you send a video or photo about the result?

nicusorcitu, 2023-11-01:
Everything looks fine, for some reasons it will just stop doing anything. It has to be a race somewhere.
I did 3 fixes (above the vglite specific only patches), I will upload them here as I think shall be applied on other units (includins SW unit) too. It still does not fix the problem but I believe they are correct.
Please let me know your thoughts on them.
[patches.zip](https://github.com/lvgl/lvgl/files/13225210/patches.zip)

nicusorcitu, 2023-11-01:
> Do you see both draw tasks in the debugger?
>
Yes, both are processed with success.

> Is the opacity of both set to 0xff?
>
Yes.

> Do you see any issue (flickering or crash) if you call lv_obj_invalidate(lv_scr_act()) right after lv_timer_handler()?
>
No, same behavior.

> Can you send a video or photo about the result?
>
There is basically no picture to show, just the screen with lights on, but nothing on the screen at all. It does not look to get stuck somewhere.

kisvegabor, 2023-11-02:
I've merged the FreeRTOS fix but what's purpose if the `unit_count` update?

To debug the rendering issue I would do this: Instead of using VGLite to fill, write a very simple software fill (just `for` loops). and use it instead. If it work we will know that the issue is in the VGLite driver somehow.

nicusorcitu, 2023-11-02:
I finally found the issue. I was using the `layer->buf_stride` when in fact this stride was not initialized.
I will prepare another PR as soon as possible with the fixes.

> what's purpose if the unit_count update?
>
If you have SW draw thread and VGLITE draw thread, if VGLite does not support a task, it shall not break the while because the SW can cover it.
Maybe it shall not be a count over there, but it has to be a method to figure out if another "unit type" is available.

kisvegabor, 2023-11-02:
> I finally found the issue. I was using the layer->buf_stride when in fact this stride was not initialized.
I will prepare another PR as soon as possible with the fixes.

Fantastic! A lot of things are still changing in v9 and I'm sure it's difficult to follow all these. I hope these wil lcome to an end soon. FYI my plan is to 
- finish the larger scale LVGL changes (API, SDL render, float support) this week
- add the minor changes in November
- update the docs in December

> If you have SW draw thread and VGLITE draw thread, if VGLite does not support a task, it shall not break the while because the SW can cover it.

What if we add a warning and drop the draw task if no one could take it after calling the `evaluate_cb`s?

nicusorcitu, 2023-11-02:
There is a difference if a dispatcher returns 0 or -1.

https://github.com/lvgl/lvgl/blob/ba3960e867b8613916577b9349be760142f3d608/src/draw/lv_draw.c#L248
If VGLITE dispatcher does not report a supported task, it shall return 0 so that SW dispatcher shall run.
If it return -1 then it will break the loop and that task will not be rescheduled.

kisvegabor, 2023-11-03:
I meant dropping the task [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw.c#L116) if `task->preferred_unit == 0`

nicusorcitu, 2023-11-03:
But you mentioned there will be some tasks that anyone can take it. E.G. tasks with `task->preferred_unit == 0`.

kisvegabor, 2023-11-03:
Yeah, we need to handle the **ANY** case somehow. I'll look into it.

nicusorcitu, 2023-11-03:
What I mean is to make a distinguish in a dispatcher if there are "no tasks" versus "task is not supported".

nicusorcitu, 2023-11-03:
I'm sorry to trouble you again, but with CPU only, one draw thread, the benchmark get stuck after a while into this mutex get:
https://github.com/lvgl/lvgl/blob/b95d74bfcb5c3372c43625c16ce3ebcf0632e888/src/stdlib/builtin/lv_mem_core_builtin.c#L141

Once the benchmark gets to the "John Smith" that mutex seems to be messed up. I do not believe is from freertos as I debug it and everything seems ok. Only the mutex pointer get changed so we are trying to get an inexistent mutex.

I have moved the init mutex from here:
https://github.com/lvgl/lvgl/blob/b95d74bfcb5c3372c43625c16ce3ebcf0632e888/src/stdlib/builtin/lv_mem_core_builtin.c#L93C1-L93C1

to the beginning of the file, because the mutex will be used in this line:
https://github.com/lvgl/lvgl/blob/b95d74bfcb5c3372c43625c16ce3ebcf0632e888/src/stdlib/builtin/lv_mem_core_builtin.c#L88C26-L88C41

but it does not fix the problem. Anyway the freertos code would take care if trying to use without initialization first.

kisvegabor, 2023-11-06:
> I finally found the issue. I was using the layer->buf_stride when in fact this stride was not initialized.
I will prepare another PR as soon as possible with the fixes.

@nicusorcitu @XuNeo 
I fixed it here 0bdd54abe and also applied the patches.

> What I mean is to make a distinguish in a dispatcher if there are "no more tasks" versus "task is not supported". I will leave it to you this one.

It shouldn't happen that a task is not supported in the dispatcher as `evaluate_cb` already checked if a task is supported. 

> Once the benchmark gets to the "John Smith" that mutex seems to be messed up. I do not believe is from freertos as I debug it and everything seems ok. Only the mutex pointer get changed so we are trying to get an inexistent mutex.

The mutex address shouldn't change, so I think it's an out of bounds write somewhere. Please try setting:
- `LV_LAYER_MAX_MEMORY_USAGE 512*104`
- `LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE 24*1024`
- `LV_MEM_SIZE  1024 * 1024`

And use SW render only. 

How does it work now?

nicusorcitu, 2023-11-06:
I am testing now SW only.
Got the latest changes from today, and it still stops at the same image of "John Smith". Sometimes it gets stop in the invalid mutex, otherwise in a different context. I agree this is a memory corruption somewhere while blending the layers.
I put these LV_ASSETS in the sw code but did not stop in any:
```
nxf68195@lw15016 ~/bitbucket/lvgl_mcu/lvgl (feature/nxp_v9)$ git diff
diff --git a/src/draw/sw/lv_draw_sw_img.c b/src/draw/sw/lv_draw_sw_img.c
index c71a743d1..ac8ff5bc8 100644
--- a/src/draw/sw/lv_draw_sw_img.c
+++ b/src/draw/sw/lv_draw_sw_img.c
@@ -54,6 +54,13 @@ void lv_draw_sw_layer(lv_draw_unit_t * draw_unit, const lv_draw_image_dsc_t * dr
      *In this case just return. */
     if(layer_to_draw->buf == NULL) return;
 
+    uint32_t stride = lv_draw_buf_width_to_stride(lv_area_get_width(&layer_to_draw->buf_area),
+                                                  layer_to_draw->color_format);
+    LV_ASSERT_MSG(stride == layer_to_draw->buf_stride, "Layer stride invalid");
+
+    void * buf = lv_draw_buf_align(layer_to_draw->buf, layer_to_draw->color_format);
+    LV_ASSERT_MSG(buf == layer_to_draw->buf, "Layer buf invalid");
+
     lv_image_dsc_t img_dsc;
     img_dsc.header.w = lv_area_get_width(&layer_to_draw->buf_area);
     img_dsc.header.h = lv_area_get_height(&layer_to_draw->buf_area);
nxf68195@lw15016 ~/bitbucket/lvgl_mcu/lvgl (feature/nxp_v9)$ 
```
With this setup it still crashes:
> LV_LAYER_MAX_MEMORY_USAGE 512*104
> LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE 128*1024
> LV_MEM_SIZE  1024 * 1024

nicusorcitu, 2023-11-08:
Tried to run again (SW only), this time setting the default value for buf alignment.
```
#define LV_DRAW_BUF_ALIGN                       4
```
And this time it get stuck at the first container test - still with John Smith images but on a different place.
The scene is invalid - 805933661.
[image]

kisvegabor, 2023-11-08:
Does it happen on the first John Smith scene? 
Are there any changes without OS?

nicusorcitu, 2023-11-10:
Yes, it happens on the first John Smith scene
It happens also without OS.

[39 more replies]

closed 2024-03-23 by lvgl-bot