lvgl/lvgl #4626: Disable LV_LABEL_LONG_SCROLL_CIRCULAR in lv_list_add_btn(...)

Vetal7979, 2023-10-04:
### Introduce the problem
I want disable scroll text in lv_list_add_btn, how i can do it?
in func lv_list_add_btn:
...
   if(txt) {
        lv_obj_t * label = lv_label_create(obj);
        lv_label_set_text(label, txt);
        lv_label_set_long_mode(label, LV_LABEL_LONG_SCROLL_CIRCULAR);
        lv_obj_set_flex_grow(label, 1);
....

Vetal7979, 2023-10-04:
I found a solution:
        btn = lv_list_add_btn(list, NULL, text);   
        lv_obj_t *txt = lv_obj_get_child(btn,-1);     
        lv_label_set_long_mode(txt, LV_LABEL_LONG_DOT);

kisvegabor, 2023-10-05:
What is not working exactly?

Please use 
````
```c
your 
code
```
````

to format code blocks

Vetal7979, 2023-10-05:
```c 
 for (int i=0;i<menu->listLength;i++)  
    {
       btn = lv_list_add_btn(list, NULL, menu->list[i].title);     
        lv_obj_t *txt = lv_obj_get_child(btn,-1);        
        lv_label_set_long_mode(txt, LV_LABEL_LONG_DOT);   
     }
```
this code makes 1-st button as LV_LABEL_LONG_WRAP, but not LV_LABEL_LONG_DOT
second and another button work normally
[image]

kisvegabor, 2023-10-06:
I've tested with this code an it works well for me. 

```c
  lv_obj_t * list = lv_list_create(lv_scr_act());
  lv_obj_t * btn;
  const char * texts[10] = {
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
          "aaaaaaaa bbbbbbbbbbbbbbbbbb ccccccccccc dddddddddddddddd",
  };

  for (int i=0;i<10;i++) {
      btn = lv_list_add_btn(list, NULL, texts[i]);
      lv_obj_t *txt = lv_obj_get_child(btn,-1);
      lv_label_set_long_mode(txt, LV_LABEL_LONG_DOT);
  }
```

Could you modify it to show the issue?

Vetal7979, 2023-10-06:
[image]

kisvegabor, 2023-10-08:
That's what you get without any modification?

closed 2023-10-30 by lvgl-bot