lvgl/lvgl #4630: Tiny_TTF updates

kisvegabor, 2023-10-05:
### Introduce the problem

In master (v9-dev) now we support stride and memory aligned buffers via [lv_draw_buf](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_buf.h). To be sure the SW render works well with non 1 stride and align settings it would be great to set these to e.g. 64 in the unit tests.

One of the blocking issue is that testing Tiny_TTF is part of our CI but the bitmaps of the glyphs don't use `lv_draw_buf_malloc()`, so the malloced arrays are not aligned correctly.

### Suggested solution

Tiny TTF should be updated to use 
- `lv_draw_buf_malloc()`
- `lv_draw_buf_free()`
- `lv_draw_buf_align()`

when allocating buffer and should also consider the stride.

Besides in v9 we have implemented a generic cache which shall be used for Tiny_TTF too. See [lv_cache.c](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_cache.c), and [lv_cache_builtin.c](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_cache_builtin.c), and [lv_lodepng.c](https://github.com/lvgl/lvgl/blob/master/src/libs/lodepng/lv_lodepng.c) as a reference.

@codewitch-honey-crisis do you have time and interest to take look at this improvements?

codewitch-honey-crisis, 2023-10-05:
I'll take a look. I'm not clear on what allocations they mean. but maybe lodepng will shed some light on it.

codewitch-honey-crisis, 2023-10-05:
Okay wait a minute. I need clarification. What am I using regular malloc for, and what is this lv_draw_buf_malloc() stuff? what do I allocate, the bitmaps? also, I'm not implementing the cache.

kisvegabor, 2023-10-05:
Thank you for the prompt reply! :blush: 

There are three things:
1. For image like buffers (e.g. glyph bitmaps) we need to use `lv_draw_buf_malloc`
2. When rendering the bitmap we need to consider stride. See `lv_draw_buf_width_to_stride()`
3. Maybe I'm wrong but now we have some kind of caching in Tiny_TTF. If so, we should use `lv_cache_..` instead.

codewitch-honey-crisis, 2023-10-05:
Well in that case I'll try to implement the cache.

codewitch-honey-crisis, 2023-10-05:
Since typedef void * (*lv_draw_buf_malloc_cb)(size_t size, lv_color_format_t color_format); wants a color format, and the fonts are an alpha transparency map, what color format do I use?

kisvegabor, 2023-10-05:
> Well in that case I'll try to implement the cache.

Amazing! :hugs: 

> Since typedef void * (*lv_draw_buf_malloc_cb)(size_t size, lv_color_format_t color_format); wants a color format, and the fonts are an alpha transparency map, what color format do I use?

`LV_COLOR_FORMAT_A8`

codewitch-honey-crisis, 2023-10-05:
Where are the instructions for building this again? Your CMakeLists.txt doesn't work out of the box.

kisvegabor, 2023-10-05:
It works for me. What error do you see?

codewitch-honey-crisis, 2023-10-05:
A bunch of file not founds. Like massive error list. 

No worries. I dragged it into a platformIO project and I'm building it that way.

codewitch-honey-crisis, 2023-10-05:
man squareline's latest update export is no good. Wish I would have known before I upgraded. This will take me awhile since I can't use squareline. It no longer just gives me UI files on export. Now it's folders and folders of crap. Nonsense.

kisvegabor, 2023-10-05:
> man squareline's latest update export is no good.

Do you mean code formatting? If so we are aware of that, and it will be fixed in the next version.

codewitch-honey-crisis, 2023-10-05:
Wow. Everything is broken.

static void lvgl_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map) 

That won't compile because two of the types no longer exist.

The latest docs are for 8.3

Am I just supposed to figure this out by osmosis? I need to hook this up to the ESP Panel API in order to test it.

You guys changed everything. Now what?

codewitch-honey-crisis, 2023-10-05:
Let me know if you update the examples or the documentation to reflect the new driver initialization and hookup procedure.

I'm blocked until I understand it.

kisvegabor, 2023-10-06:
Sorry, I should have linked [this comment](https://github.com/lvgl/lvgl/issues/4011#issue-1595034440) where I summarized the changes. I will update the docs before the release when there won't be new changes.

codewitch-honey-crisis, 2023-10-06:
That's just what I needed! Thanks so much.

codewitch-honey-crisis, 2023-10-06:
What am I doing wrong here? It's crashing my TTGO on lv_display_create() - and yeah VRES/HRES are reversed. It's intentional.
```cpp
void setup() {
    Serial.begin(115200);
    lcd_panel_init();
    // crashes:
    disp = lv_display_create(LCD_VRES, LCD_HRES);
    lv_display_set_flush_cb(disp, lvgl_flush);
    lv_display_set_draw_buffers(disp, transfer_buffer1, transfer_buffer2, sizeof(transfer_buffer1), LV_DISPLAY_RENDER_MODE_PARTIAL);
    ui_init();
}
```

codewitch-honey-crisis, 2023-10-06:
I just realized I didn't. Sorry it's early, and I'm on my first cup of coffee - I've not quite arrived yet.

codewitch-honey-crisis, 2023-10-06:
Oh no. This is the same problem I ran into before. There's something in your .bss seg that prevents LVGL from working on an ESP32.

c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\test\firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 225592 bytes

I can find the offending unintialized static var if you need. You are allocating too much.

kisvegabor, 2023-10-06:
The 2 `transfer_buffer`s?

codewitch-honey-crisis, 2023-10-06:
I mean, I can remove one, but that's only 64k. You're overflowing by over 200kB. That's not my code. I ran into this before. I just forgot about it. My workaround was to not run it on an ESP32, set up a build environment and run the tests, but frankly, I want to avoid that again. Too many problems. This needs to work on an ESP32 or it's a bug.

codewitch-honey-crisis, 2023-10-06:
I set the transfer buffers to 16kB each just to check. Still overflows by 180+kB as I expected.

I'm trying to remember how to get a copy of the .bss dump so I can find the offending vars, but although I've done it before, I can't remember for the life of me what I did, so I'm sifting through google at the moment.

codewitch-honey-crisis, 2023-10-06:
I got the firmware.map but it's huge, and there are about 1300 .bss entries. That's overwhelmingly large. I think this pursuit might be a dead end.

If it helps, it doesn't cause a linker error unless I call lv_init(). My guess is maybe the memory pool? Perhaps the cache?

kisvegabor, 2023-10-06:
Please check LV_MEM_SIZE in lv_conf.h

codewitch-honey-crisis, 2023-10-06:
Thank you. That did it, although I had to set it to 40kB. That doesn't leave a lot of memory. Let's hope it works.

I think this TTGO comes with 384kB of RAM rather than the 512kB you get on some. That is probably why I didn't see this error with my M5Stack fire which I tried to use initially, but hardware failure. =( cheap widgets.

codewitch-honey-crisis, 2023-10-06:
Never mind. Autocomplete saved the day. I figured it out. Thanks anyway. =)

codewitch-honey-crisis, 2023-10-06:
I have a question about the cache. Lodepng appears to fail if the cache cannot allocate the memory. Is this how I should handle the font code, or should I fallback to not using the cache?

codewitch-honey-crisis, 2023-10-06:
I have another question. I need to key the cache by codepoint and the lv_font_t pointer.

Unfortunately your cache is not generic. I think it should have simply taken some key-data and a hash function. It doesn't. So now I can't key it appropriately. I guess I could resolve the pointer to a string with itoa but come on. That's ugly and slow. There has to be a better way.

codewitch-honey-crisis, 2023-10-06:
I have an idea. If I can use the cache pointer entry type, maybe I can add the codepoint to the value of the pointer. That will render the pointer invalid. I'm assuming that's safe.

codewitch-honey-crisis, 2023-10-06:
I think i figured out the cache maybe, but this call crashes
```cpp
    size_t szb = h*stride;
    tent->buffer = lv_draw_buf_malloc(szb, LV_COLOR_FORMAT_A8);
```

codewitch-honey-crisis, 2023-10-06:
Ran into another problem. Using lv_draw_buf_malloc is not really compatible with the cache.

The reason being is I need to know when the cache gets expired so I can free my buffer.

I would just use the cache allocation, but you want me to use this draw buf allocation and the cache just isn't compatible with it as far as I can tell.

Also it still crashes.

kisvegabor, 2023-10-07:
It crashes because the cache data (return value of `lv_cache_get_data()`) is not allocated by default. This way you have the opportunity to either allocate it manually or assign any buffer there.

I recommend this:
```c
lv_cache_entry_t* entry = lv_cache_add(szb);  //The size of the cache data is the size of the bitmap
ttf_cache_entry_t *tent = lv_malloc(sizeof(ttf_cache_entry_t));
uint8_t* buffer = lv_draw_buf_malloc(szb, LV_COLOR_FORMAT_A8);

entry->data = buffer;
entry->free_data = 1;  //Automatically free buffer on cache invalidation

entry->user_data = tent;
entry->free_user_data = 1;  //It's not implemented in the cache now, but can be easily added to automatically free the user_data on invalidation.
```

codewitch-honey-crisis, 2023-10-07:
Thank you!

I'm a bit surprised at the API. The thing that threw me was you give it the size. That made me think it allocates the data, you know? Otherwise it seems odd to give it the size. That's fine, but I'd document the heck out of that. The other thing that is strange to me, is it knows how to free the data you allocated. What if you allocate PSRAM or something and that platform requires a special call to free it? I mean, this is just feedback primarily so you can document it. This is just the stuff that jumped out at me.

codewitch-honey-crisis, 2023-10-07:
Oh I don't need the user data stuff. the only reason for tent was I thought I needed it.

codewitch-honey-crisis, 2023-10-07:
I have a question about logging. I am hesitant to log within the bitmap retrieval callback because of the amount of spew it would generate. On the other hand right now it doesn't report errors. This is the last bit before I wrap up the code. I only changed one file fortunately, but I'd like your feedback on logging within the bitmap routine before I do a pull request.

codewitch-honey-crisis, 2023-10-07:
I went ahead and added logging to that routine. I figure it's easy enough to remove

codewitch-honey-crisis, 2023-10-07:
Okay I'm a github idiot. I've updated my local fork with the changes but I don't know how to submit a PR.

codewitch-honey-crisis, 2023-10-07:
Anyway I also ran into the line ending problem. I've tried everything with git and it won't fix it.

codewitch-honey-crisis, 2023-10-07:
I think i fixed the line endings? but it's failing on the 64 bit tests and I'm not sure why. I can't make heads or tails of this: https://github.com/codewitch-honey-crisis/lvgl/actions/runs/6443833589

kisvegabor, 2023-10-09:
It was fast, thank you! 🥇 
You can create a PR as described here: https://docs.lvgl.io/8.3/CONTRIBUTING.html#pull-request

Don't worry about the failing CI, sometimes the code coverage report doesn't find its server.

codewitch-honey-crisis, 2023-10-09:
Do I have to create a new pull request every time this build fails?

This is taking forever and a day.

codewitch-honey-crisis, 2023-10-09:
Anyway, it probably would work, but line breaks. Github absolutely hates that I contribute to your code and always screws up my line breaks no matter what i do. This time I copied the file through the github web to the github web and it STILL screwed it up.

So if someone else wants to fix my PR fine. I am done.

kisvegabor, 2023-10-10:
I hope you and GitHub will reconcile soon :slightly_smiling_face: To be more serious I have no idea why code formatter wasn't working on your end. 

Anyway the PR looks great, and I'll merge in a few minutes. 

Thank you for your quick help! :100:

closed 2023-10-10 by kisvegabor