lvgl/lvgl #4640: Rendering verification

kisvegabor, 2023-10-09:
### Introduce the problem

Now it's hard to verify that rendering works correctly in a new environment with GPU support other customization.

### Examples and cases
@nicusorcitu is developing VG-Lite and PXP GPU support for LVGL v9 and now there is no standard way to say that whether the GPU integration is working well or not.

### Suggested solution

1. Create an `lv_demo_render_test` with several pages for each draw task type (there should be normal, opa, non-normal blend mode variants for each):
    1.  fill (radius + gradient too)
    2. border (each sides + radius)
    3. outline (radius)
    4. box shadow (offset, size, spread)
    5. text (normal text + underline/strike through, placeholder)
    6. triangle (just some rectangles)
    7. image (various formats + transformation)
    8. line (various angles + line caps)
    9. arc (some arcs + caps)
    10. vector (later)
    11. layer (blend mode, transformation)
    12. mask bitmap (not implemented SW render yet)
    13. mask rectangle 
2. Allow page changing by Next/Previous buttons so that the test results can examined by a human
3. Provide an interface to save the screenshots of each page into a PNG file and/or a simple C array. These will be the reference data.
4. Create an automated mode for the demo in which it automatically compares each page with the related reference data.

Note that
- This way SW render, VG-Lite, PXP, Arm-2D, SDL, etc, all will have their own set of reference images. 
- Where possible (SW, SDL, ?) the automated tests should be parts of the CI, similarly to the current unit tests
- We don't run all unit tests on MCUs because we assume that e.g. if styles were working on PC they will work on an embedded hardware as well.
- The target resolution should be around 800x480 
- The benchmark demo in #4624 is independent from this 

What do you think?

cc @bjsylvia, @GorgonMeducer, @nicusorcitu, @XuNeo.

GorgonMeducer, 2023-10-11:
Is this an improved version of the existing benchmark?

nicusorcitu, 2023-10-11:
The scope is to have a continuous integration test for GPUs similar to the existing tests that runs on the CPU.
The performance is not the purpose of those tests but rather the image quality. There should be a image comparation (similar to what is already done in unity support).

GorgonMeducer, 2023-10-11:
> There should be a image comparation (similar to what is already done in unity support).

This is really a good feature.

kisvegabor, 2023-10-11:
I would like to focus on #4624 first because I'm very excited to have some benchmark about v9 :smile:

bjsylvia, 2023-10-14:
Apart from the basic CI verification, do you also provide validation for parallel rendering and related comparative data reports here?

kisvegabor, 2023-10-16:
In the above listed points parallel rendering can be used too. In this case the reference images can be slightly different (e.g. because the GPU rotates slightly differently than SW).

So yes, we should have reference images for various configurations, e.g. in case of NXP:
- SW only
- SW + PXP
- SW + VG-Lite
- SW + PXP + VG-Lite

nicusorcitu, 2023-10-17:
I-ll copy paste my comment for the other issue:
> We can do otherwise: run the tests and dump the FB (screenshot) as png and transfer them over the serial on the host in a folder having the name of the test. The comparation with the reference images can be done on the host after the tests runs completely. We don;t have to make the png diff on target.
> I know there is no 1:1 comparation between CPU output and GPU output. We might have some scripts to evaluate the PSNR (peak signal-to-noise ratio) or SSIM (structural similarity index measure) in order to measure the similarity between images.
> 
Having said that, it will not be a 1 to 1 comparation on the images so we can have a "threshold" set in order to measure the similarities with the reference which can be SW only image.

kisvegabor, 2023-10-18:
I'm open to it as well, but we need to discuss the advantages and disadvantages.

Let's get back to this when we have something from the benchmark.

kisvegabor, 2023-10-25:
I've added the a basic version of the render test:
3ff625483

It runs in 480x272 to make it easy to use on more displays.
Only fill and border is add yet, but I'm planning something similar for the others too.
[image]

[image]

kisvegabor, 2023-10-26:
Box shadow, text, and image tests are added as well.

[image]
[image]
[image]
[image]

kisvegabor, 2023-10-26:
Aaand finished! All main rendering types are covered and it's also integrated with the unit tests. See
- unit test: https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/draw/test_render.c 
- reference images: https://github.com/lvgl/lvgl/tree/master/tests/ref_imgs/draw/render

What o you think?

nicusorcitu, 2023-10-30:
Thank you very much. I will give it a try but still have something to do in the queue.

kisvegabor, 2023-11-14:
Not stale.

nicusorcitu, 2023-11-29:
We will start running the tests on NXP board in December.

anaGrad, 2023-12-12:
I tried running the redering tests and I get fatal error on png.h as it is not included in the library.

nicusorcitu, 2023-12-12:
This was the main issue from the beginning. The question is how to make the tests independent of that dinamic library? Can we have a static library to replace png.h?
________________________________
De la: anaGrad ***@***.***>
Trimis: Tuesday, December 12, 2023 3:59:26 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] Rendering verification (Issue #4640)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

I tried running the redering tests and I get fatal error on png.h as it is not included in the library.

kisvegabor, 2023-12-13:
You can just run the demo by calling `lv_demo_render()`. After that it's up to you how to compare the rendered image with the reference images. 

In the future we can develop a unified solution that works on embedded system too, not only in our CI.

kisvegabor, 2023-12-30:
I consider the rendering verification complete by using `lv_demo_remder`. If needed a new issue can be opened for using test images on embedded hardware

closed 2024-01-14 by kisvegabor