lvgl/lvgl #4641: LVGL with FATFS does not work, no image loaded from SD-Card

epikao, 2023-10-10:
Hello,

I’m using LVGL 8.3.8 with STM32H750 and STM32CubeIDE and HAL.
With a "Driver-Test Code", see below, I can confirm that my SD-Card hardware is working properly.

Now I'm attempting to integrate all of this with LVGL, but without success. I've enabled LV_USE_FATFS and LV_USE_BMP in lv.conf file.
I'm using following code, as seen below in “LVGL-Code”. 
I'm not receiving any log warnings from LVGL, but only the background color is displayed, but the image from the SD-Card isn't being loaded.
With standard debugging methods, I'm unable to pinpoint what's going wrong here.

Does anyone have any hints (also for debugging) or is this possibly a bug? Am I overlooking something?
I can't find any examples using the included FATFS driver at this link: "https://docs.lvgl.io/8.3/overview/file-system.html".

Thank you very much

**LVGL-Code:**
`      MX_SDMMC1_SD_Init();
      MX_FATFS_Init();
#if 1
      lv_fs_fatfs_init();

      static lv_style_t style;
      lv_style_init(&style);
      lv_style_set_bg_color(&style, lv_color_make(0xFF,0x00,0xFF));
      lv_style_set_bg_opa(&style, LV_OPA_COVER);
    
    
      img1 = lv_img_create(lv_scr_act());
      lv_obj_set_size(img1, 320, 320);
      lv_img_set_src(img1, "S:color.bmp");
      lv_obj_set_pos(img1, 0, 0);
      lv_obj_add_style(img1, &style, 0);
#endif
      `
      
      `/*API for FATFS (needs to be added separately). Uses f_open, f_read, etc*/
#define LV_USE_FS_FATFS 1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'S'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_FATFS_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

/*PNG decoder library*/
#define LV_USE_PNG 0

/*BMP decoder library*/
#define LV_USE_BMP 1`

**Driver-Test Code:**
`
MX_SDMMC1_SD_Init();
  MX_FATFS_Init();

#if 0
  FATFS fs;
  FRESULT res;
  UART_Printf("Ready!\r\n");

  // mount the default drive
  res = f_mount(&fs, "", 0);
  if(res != FR_OK) {
      UART_Printf("f_mount() failed, res = %d\r\n", res);
      return;
  }

  UART_Printf("f_mount() done!\r\n");

  uint32_t freeClust;
  FATFS* fs_ptr = &fs;
  res = f_getfree("", &freeClust, &fs_ptr); // Warning! This fills fs.n_fatent and fs.csize!
  if(res != FR_OK) {
      UART_Printf("f_getfree() failed, res = %d\r\n", res);
      return;
  }

  UART_Printf("f_getfree() done!\r\n");

  uint32_t totalBlocks = (fs.n_fatent - 2) * fs.csize;
  uint32_t freeBlocks = freeClust * fs.csize;

  UART_Printf("Total blocks: %lu (%lu Mb)\r\n", totalBlocks, totalBlocks / 2000);
  UART_Printf("Free blocks: %lu (%lu Mb)\r\n", freeBlocks, freeBlocks / 2000);

  DIR dir;
  res = f_opendir(&dir, "/");
  if(res != FR_OK) {
      UART_Printf("f_opendir() failed, res = %d\r\n", res);
      return;
  }

  FILINFO fileInfo;
  uint32_t totalFiles = 0;
  uint32_t totalDirs = 0;
  UART_Printf("--------\r\nRoot directory:\r\n");
  for(;;) {
      res = f_readdir(&dir, &fileInfo);
      if((res != FR_OK) || (fileInfo.fname[0] == '\0')) {
          break;
      }

      if(fileInfo.fattrib & AM_DIR) {
          UART_Printf("  DIR  %s\r\n", fileInfo.fname);
          totalDirs++;
      } else {
          UART_Printf("  FILE %s\r\n", fileInfo.fname);
          totalFiles++;
      }
  }

  UART_Printf("(total: %lu dirs, %lu files)\r\n--------\r\n", totalDirs, totalFiles);

  res = f_closedir(&dir);
  if(res != FR_OK) {
      UART_Printf("f_closedir() failed, res = %d\r\n", res);
      return;
  }

  UART_Printf("Writing to log.txt...\r\n");

  char writeBuff[128];
  snprintf(writeBuff, sizeof(writeBuff), "Total blocks: %lu (%lu Mb); Free blocks: %lu (%lu Mb)\r\n",
      totalBlocks, totalBlocks / 2000,
      freeBlocks, freeBlocks / 2000);

  FIL logFile;
  res = f_open(&logFile, "log.txt", FA_OPEN_APPEND | FA_WRITE);
  if(res != FR_OK) {
      UART_Printf("f_open() failed, res = %d\r\n", res);
      return;
  }

  unsigned int bytesToWrite = strlen(writeBuff);
  unsigned int bytesWritten;
  res = f_write(&logFile, writeBuff, bytesToWrite, &bytesWritten);
  if(res != FR_OK) {
      UART_Printf("f_write() failed, res = %d\r\n", res);
      return;
  }

  if(bytesWritten < bytesToWrite) {
      UART_Printf("WARNING! Disk is full, bytesToWrite = %lu, bytesWritten = %lu\r\n", bytesToWrite, bytesWritten);
  }

  res = f_close(&logFile);
  if(res != FR_OK) {
      UART_Printf("f_close() failed, res = %d\r\n", res);
      return;
  }

  UART_Printf("Reading file...\r\n");
  FIL msgFile;
  res = f_open(&msgFile, "log.txt", FA_READ);
  if(res != FR_OK) {
      UART_Printf("f_open() failed, res = %d\r\n", res);
      return;
  }

  char readBuff[128];
  unsigned int bytesRead;
  res = f_read(&msgFile, readBuff, sizeof(readBuff)-1, &bytesRead);
  if(res != FR_OK) {
      UART_Printf("f_read() failed, res = %d\r\n", res);
      return;
  }

  readBuff[bytesRead] = '\0';
  UART_Printf("```\r\n%s\r\n```\r\n", readBuff);

  res = f_close(&msgFile);
  if(res != FR_OK) {
      UART_Printf("f_close() failed, res = %d\r\n", res);
      return;
  }

  // Unmount
  res = f_mount(NULL, "", 0);
  if(res != FR_OK) {
      UART_Printf("Unmount failed, res = %d\r\n", res);
      return;
  }

  UART_Printf("Done!\r\n");
#endif`

kisvegabor, 2023-10-10:
> What does following mean exactly /*API for FATFS (needs to be added separately). Uses f_open, f_read, etc*/ ?
I've a "ff.c" file which already include all these function "f_open...", but do I need to link it somewhere with LVGL's included FATFS driver??

It means that FATFS is not part of LVGL, but you need to add it to the project yourself. So it's what you did.

Regarding the BMP image, be sure that the color depth of the image is the same as `LV_COLOR_DEPTH`.

epikao, 2023-10-11:
Thank you. I'm using a 16-bit color depth and have now adjusted the image accordingly, but unfortunately, without success. 
LVGL doesn't return any error messages... Is that normal?

kisvegabor, 2023-10-11:
Have you enabled [logging](https://docs.lvgl.io/8.3/porting/log.html)?

epikao, 2023-10-12:
Yes, but from the LVGL log I don't get a related message, 
I have added log in fs_open in "lv_fs_fatfs.c" and now I am encountering the error 12 = FR_NOT_ENABLED,			/* (12) The volume has no work area */
I checked with another name (color.bmp) or with S:/ , but error is still the same...

```
static void * fs_open(lv_fs_drv_t * drv, const char * path, lv_fs_mode_t mode)
{
    LV_UNUSED(drv);
    uint8_t flags = 0;

    if(mode == LV_FS_MODE_WR) flags = FA_WRITE | FA_OPEN_ALWAYS;
    else if(mode == LV_FS_MODE_RD) flags = FA_READ;
    else if(mode == (LV_FS_MODE_WR | LV_FS_MODE_RD)) flags = FA_READ | FA_WRITE | FA_OPEN_ALWAYS;

    FIL * f = lv_mem_alloc(sizeof(FIL));
    if(f == NULL) return NULL;

    FRESULT res = f_open(f, path, flags);
    if(res == FR_OK) {
        return f;
    }
    else {
    	UART_Printf("f_open() failed, res = %d\r\n", res);
        lv_mem_free(f);
        return NULL;
    }
}
```

epikao, 2023-10-12:
where does lvgl mount the sd-card ?

kisvegabor, 2023-10-13:
LVGL doesn't mount the SD card. It need to be done in your application. LVGL just calls `f_open`.

epikao, 2023-10-16:
Ok, added mount, and now I'm encountering following error, when I call my sd-card demo the first time:
`f_read() failed, res = 2.   /* ..assertion failed */`

In my code, I can switch between different lvgl-demos including my sd-card picture demo.
What's interesting is that, when I switch a second time to my sd-card demo, it successfully loads the picture.
But when I switch to another demo afterward, the program freezes, and I receive the following message (from lvgl widget demo): 

`  [Error]     (137.541, +128)  block_locate_free: Asserted at expression: block_size(block) >= size   (in lv_tlsf.c line #775)`
  
 So it appears that I have some assertion failures here... 
Could these be related to some RAM problem or overflow?
 Any hints that I can try?
  
 Thank you

kisvegabor, 2023-10-16:
> Ok, added mount, and now I'm encountering following error, when I call my sd-card demo the first time:
f_read() failed, res = 2.   /* ..assertion failed */

Does FatFS work without LVGL?

epikao, 2023-10-17:
According my Code in the first post, the basic commands works, so I would say yes.
And as mentioned, it also works with LVGL, but not in all cases, sometimes it can load the image, sometimes not ...
And I don't know how to test image load without LVGL....

epikao, 2023-10-17:
sometimes I get also following Warn:

`_lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125`

kisvegabor, 2023-10-17:
Maybe increasing LV_MEM_SIZE?

epikao, 2023-10-18:
ok thanks finally, f_mount, and increase LV_MEM_SIZE helped.
BTW: 
By what rules do I actually define LV_MEM_size? The images I load are 1024x600x16-bit in size.

kisvegabor, 2023-10-18:
I feel like the issue is related to the specific application/hardware you are using.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2023-10-18 by kisvegabor

dbtronics, 2024-06-05:
Hey I'm using LVGL (3.8.11) implemented code generated from SquareLine Studio. I'm also trying to load the image from SD-Card as well but I'm completely lost as to how to implement this. I'm trying to load C array file images from sd card. There doesn't seem to be any examples or resource that will point me as to how this method is implemented. @kisvegabor any examples you know that will help with this when the code is generated through squareline studio?

kisvegabor, 2024-06-05:
Hi,
For SquareLine related issues please use https://forum.squareline.io/