lvgl/lvgl #4642: Is background-size considered to be supported?

C-Ackerman, 2023-10-10:
Hi Gabor:
  Have you considered supporting background-size?

https://developer.mozilla.org/en-US/docs/Web/CSS/background-size

kisvegabor, 2023-10-11:
Hey, 

I've just opened #4648 to discuss some related questions.

bjsylvia, 2023-10-14:
@kisvegabor  There are also [`background-position`](https://developer.mozilla.org/en-US/docs/Web/CSS/background-position) and [`background-repeat`](url) related to the background. Maybe we should think about them together?

kisvegabor, 2023-10-16:
This issue is also related: https://github.com/lvgl/lvgl/issues/4102

I think we can unify what settings we have for `lv_image` and the background. I'm thinking about making the background a new `LV_PART_BACKGROUND` so we would have all the x, y, width, height, align, image recolor, etc style options.

kisvegabor, 2023-10-31:
Closing in favor of #4733.

closed 2023-10-31 by kisvegabor