lvgl/lvgl #4647: Offset bug when zoom is applied.

Kirbyrawr, 2023-10-10:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

This bug is a bit similar to this one https://github.com/lvgl/lvgl/issues/1737 but not exactly the same, to reproduce it you can do the following:
1. Convert this image to C array (Truecolor with Chroma, pure green)
[image]

2. Add this code:
```c
    lv_obj_t* img = lv_img_create(lv_scr_act());
    lv_img_set_src(img, &munchlax);
    lv_obj_align(img, LV_ALIGN_CENTER, 0, 0);
    lv_img_set_antialias(img, false); 
    lv_obj_set_size(img, 24, 24); 
    lv_img_set_zoom(img, 256*4);
    lv_img_set_pivot(img, 0, 0);
    lv_img_set_offset_x(img, -24 * 4 * 2); //0, 1 works ok, other than that it won't work.
```
You can see that the image is cut, this is actually because the offset it's wrapped, but it's probably not taking in consideration the zoom
I also tried with non-negative offsets.

### Expected behavior

The offset should be applied properly when the img is zoomed.

### Screenshots or video

https://github.com/lvgl/lvgl/assets/3259762/99143ea5-c915-49b1-8ee5-b661af3b9fff

Thanks.

kisvegabor, 2023-10-11:
Hi,

I've tried it out an I think the issues is this:
- The width of the image is 120. 
- When you set an offset larger than 120 (e.g. -24*4*2) LVGL calculate from the non-zoomed image where it should start. In the example  -24 * 4 * 2 = -192, if we repeat the  120 px wide image, we can calculate that the firs pixel is at -192 + 120 = -72, -72 + 120 = 48. So the actual image offset will be 48.

It's really not intuitive, an unfortunately I don't have a good idea to solve it in a simple way. Probably using a 5 separate image would  the easiest.

Kirbyrawr, 2023-10-11:
> Hi,
>
> I've tried it out an I think the issues is this:
> - The width of the image is 120.
> - When you set an offset larger than 120 (e.g. -24*4*2) LVGL calculate from the non-zoomed image where it [...]

Thanks for the reply, i agree with you on the separated images, that was my alternative if this didn't work.

Thanks again for take a look and confirm what i suspected!

kisvegabor, 2023-10-11:
No problem :slightly_smiling_face: 
In v9, this part of `lv_img` wil be reworked so I hope we will have something more consistent.

closed 2023-10-11 by kisvegabor