lvgl/lvgl #4648: Float, matrix transformation and non-uniform scale support

kisvegabor, 2023-10-11:
### Introduce the problem
The end goal is to support non uniform scale (different scale horizontally than vertically). To do so, we should introduce describing the image transformation with a matrix. However, to use a matrix in a simple way we should introduce floats into LVGL. However using floats should be optional, as it's slow on systems without FPU.  

### Examples and cases
Non uniform scale could be used to make an image fit into it's parent, even if the the parent doesn't have the same proportion as the image. It's also required for 9-patch image rendering.

Using a matrix opens the door to describe perspective transformations and skew transformations as well.

Introducing float could be useful on APIs like 
- image rotation
- image scale
- arc start end value
- line points

### Suggested solution
To introduce non uniform scale in the end we should consider how to handle each step.

**Float**
- I suggest having a `LV_USE_FLOAT` config option and an `lv_precise_value_t`types which is either `int32_t` or `float` depending on `LV_USE_FLOAT`
- If `LV_USE_FLOAT` is disabled we can say that `lv_precise_value_t` is always upscaled by 256
- We can gradually migrate the API of the widgets to support `lv_precise_value_t`

**Matrix support**
- We need to figure out how make it simple for the user to configure simple image rotation and scale with matrix. So we still should have `lv_image_set_scale/rotation()` functions but also an `lv_image_set_transformation_matrix()`. Same for styles.

**Non unified scale**
- It means using the matrix during image rendering 

What do you think?
@C-Ackerman @GorgonMeducer @nicusorcitu @PeterBee97

GorgonMeducer, 2023-10-11:
In Arm-2D, we use four corner points to actually determine the result of a transformation, so it could be:

- Rotation
- Scaling
- Rotation with Scaling
- Stretch 
- etc

It all depends on how to generate the four anchor points. 
I hope this info can help.

kisvegabor, 2023-10-11:
The 4 corner points also can be used for ["skew" transformation.](https://css-tricks.com/almanac/properties/t/transform/#aa-skew)

I've found this nice demo for 4 corner transformation: https://codepen.io/MNK/pen/dZqGvd

Actually I really like it, especially if we allow using the relative values, e.g. 50% of the original image size or the parent size. 
With the 4 corner description the matrix can be used to calculate the values of the corner points. 

However we still need to be able to describe rotation and scale in an explicit way, as having 4 arbitrary points could be too complex for most of the GPUs even if they are calculated from simple scale and/or rotation only.

The only thing I miss is the ability to describe perspective transformation. E.g. this is not a real perspective transformation as pixels should be "denser" on the right (far):
[image]

However maybe perspective transformation can be described differently.

So in the end we could have these to describe an image transformation:
1. scale and/or rotation + pivot explicitly
2. 4 corner points, if set 1) will be ignored
3. perspective transformation: I don't know how it fits yet :slightly_smiling_face:

GorgonMeducer, 2023-10-11:
@kisvegabor 
4 corner points are the internal form, and normal scale and/or rotation is translated to 4 points formation.

For perspective transform, it needs a different style of description. So please do not mix it with 4 corner points solution

kisvegabor, 2023-10-12:
Ah, I see. Now it's much clearer in my mind. It seems it won't be as complicated as I thought initially. 

I think finally all we need are `scale_x` and `scale_y` style and image properties. Maybe some GPUs can handle the scale only if these 2 are the same, but in SW render it's simple to handle different x and y scale.

The normal `scale` property just sets scale_x` and `scale_y` to the same value.

So we don't need the matrix  now and we can postpone the perspective transformation too.

The floating point support is still important, but not blocking for this. 

I'll find some time to try out scale_x` and `scale_y` hopefully next week.

bjsylvia, 2023-10-14:
> We can gradually migrate the API of the widgets to support lv_precise_value_t

So Do you mean LVGL would support float for all position/size style (x, y, width, height ...) ? 

If so, this sounds like good news for devices with an FPU. When performing animation operations, using float precision can effectively improve rendering performance.

kisvegabor, 2023-10-16:
> So Do you mean LVGL would support float for all position/size style (x, y, width, height ...)

I mean to introduce `lv_precise_value_t` where it makes sense. We need to figure out where it makes sense. :slightly_smiling_face:  Arc angle, scale, rotation, and line coordinates are good starting points. 

If a GPU supports sub pixel rendering it's very useful, otherwise, I can't see how it can help with X, Y, width and hegiht.

nicusorcitu, 2023-10-17:
VGLite GPU does support subpixel rendering (antialiasing).
It also support scale and zoom to decimals 0.1 factor or 0.1 degree.
So having a matrix 3x3 with floats will really help on vglite case.

kisvegabor, 2023-10-18:
I'm into the new benchmark now, but I'll look into it as well ASAP.

kisvegabor, 2023-10-20:
I've pushed an update to support `scale_x` and `scale_y` for `lv_image` and `style_transform_scale_x/y` too. 

It opens the door for a more CSS like image sizing as discussed here #4102.

But before that I would like to add the basics of the float support. This is my idea in more detail
- Add `LV_USE_FLOAT`
- Add `lv_value_precise_t` which if `int32_t` (upscaled by 256) or `float` depending on `LV_USE_FLOAT`
- Add
```c
//Optional macro to always use float on the API.
//E.g. `lv_image_set_scale(image, lv_precise(1.3));`
//Used only for the examples for float-independence. 
//In real UIs you can use `1.3` directly if floats are enabled.
//Suggestion for a shorter name would be welcome
#if LV_USE_FLOAT
    #define lv_precise(v) (float) v
#else 
    #define lv_precise(v) (int32_t) v * 256
#endif
```
- Use `lv_value_precise_t` in the following places at first:
    - scale
    - rotation
    - arc angle
    - line points  
    - `lv_sin`, `lv_cos`
-

kisvegabor, 2023-10-25:
I've added the first version of float support. See 84c8cf857

As an example I've enhanced arc with float support.

kisvegabor, 2023-10-31:
Now `lv_line` also supports `lv_value_precise_t`. I started to work on using `lv_value_precise_t` for rotation too (instead of the current 0.1 degree resolution) but it's larger update affection lv_arc, lv_scale, and all the transformations.

kisvegabor, 2023-11-02:
Having float for image rotation is is more complicated than I anticipated. Mainly because it will produce slightly different result than integers, so all the unit tests needed to be duplicated.  

I leave it for now, and consider the introduction of `float`s ready by having them in arc angles and line points. 

Later we can get back to having float for image rotation and pivot too.

closed 2023-11-02 by kisvegabor