lvgl/lvgl #4652: Scrolling exception under RTL

fuxiujun, 2023-10-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

After setting` lv_obj_set_style_base_dir(lv_scr_act(), LV_BASE_DIR_RTL, 0)`, the scrolling of some objs is affected and cannot scroll left and right correctly.

### To Reproduce

```c
void lv_example_scroll_1(void)
{
    /*Create an object with the new style*/
    lv_obj_t * panel = lv_obj_create(lv_scr_act());
    lv_obj_set_size(panel, 200, 200);
    lv_obj_center(panel);

    lv_obj_t * child;
    lv_obj_t * label;

    child = lv_obj_create(panel);
    lv_obj_set_pos(child, 0, 0);
    lv_obj_set_size(child, 70, 70);
    label = lv_label_create(child);
    lv_label_set_text(label, "Zero");
    lv_obj_center(label);

    child = lv_obj_create(panel);
    lv_obj_set_pos(child, 160, 80);
    lv_obj_set_size(child, 80, 80);

    lv_obj_t * child2 = lv_button_create(child);
    lv_obj_set_size(child2, 100, 50);

    label = lv_label_create(child2);
    lv_label_set_text(label, "Right");
    lv_obj_center(label);

    child = lv_obj_create(panel);
    lv_obj_set_pos(child, 40, 160);
    lv_obj_set_size(child, 100, 70);
    label = lv_label_create(child);
    lv_label_set_text(label, "Bottom");
    lv_obj_center(label);
}

void test(void)
{
  lv_obj_set_style_base_dir(lv_scr_act(), LV_BASE_DIR_RTL, 0);
  lv_example_scroll_1();
}

```

### Expected behavior

Can scroll left and right normally

```c
lv_coord_t lv_obj_get_scroll_right(lv_obj_t * obj)
{
------

    /*With RTL base dir can't scroll to the object out on the right.
     *So simply use the current scroll position as "right size"*/
    if(lv_obj_get_style_base_dir(obj, LV_PART_MAIN) == LV_BASE_DIR_RTL) {
        if(obj->spec_attr == NULL) return 0;
        return obj->spec_attr->scroll.x;
    }

-------
}
```
It might have something to do with this code but I don't understand why it does this

### Screenshots or video

https://github.com/lvgl/lvgl/assets/49239561/857611b6-a1f1-4917-a2f0-3a1d6fd25c82

fuxiujun, 2023-10-11:
https://github.com/lvgl/lvgl/assets/49239561/841dcd54-c846-4eae-978d-ea4488641d5c

kisvegabor, 2023-10-11:
Actually it's not a bug but a features :smile: 

When I implemented the scrolling I checked how HTML works as a reference. And it turned out there is not scroll to the left with LTR direction, or to the right with RTL direction. See these examples:
- LTR: https://codepen.io/kisvegabor/pen/OJrqZmy
- RTL: https://codepen.io/kisvegabor/pen/zYybjZj

So I went with it as it simplified several things.

closed 2023-10-13 by fuxiujun