lvgl/lvgl #4658: How to disable drag scroll but show scrollbar?

lanistor, 2023-10-14:
### Introduce the problem

Epaper painting speed is slow, and drag experience is really bad. So we add buttons to turn page, by `lv_obj_scroll_by ` function.
But the drag operation still effect, if we use `lv_obj_clear_flag(content, LV_OBJ_FLAG_SCROLLABLE)`, the scrollbar won't be seen.

### Examples and cases

```
lv_obj_set_scrollbar_mode(content, LV_SCROLLBAR_MODE_AUTO);

void scroll_but_click_cb(lv_event_t* e, scroll_but_type type) {
  auto obj = (fl_scroll_cont*)e->user_data;
  if (!obj || !is_widget_valid<fl_scroll_cont>(obj)) {
    return;
  }

  auto content = obj->get_content();
  if (!content) {
    return;
  }

  auto content_height = lv_obj_get_content_height(content);
  auto scroll_top     = lv_obj_get_scroll_top(content);
  auto scroll_bottom  = lv_obj_get_scroll_bottom(content);

  auto scroll_by = content_height - 60;
  if (type == scroll_but_type::pre) {
    if (scroll_by > scroll_top) {
      scroll_by = scroll_top;
    }
    if (scroll_by < 0) {
      scroll_by = 0;
    }
    scroll_by = -scroll_by;
  } else {
    if (scroll_by > scroll_bottom) {
      scroll_by = scroll_bottom;
    }
    if (scroll_by < 0) {
      scroll_by = 0;
    }
  }

  if (scroll_by == 0) {
    return;
  }

  lv_obj_scroll_by(content, 0, -scroll_by, LV_ANIM_OFF);
}
```

### Suggested solution

I think where will be a way to show scrollbar but disable drag operation currently, but i don't know how to implement it.

kisvegabor, 2023-10-16:
Hey,

Do you use LVGL v8 or v9-dev?

In v9 we could this:
- renaming `LV_SCROLLBAR_MODE_AUTO` to `LV_SCROLLBAR_MODE_OVERFLOW` which just display the scrollbar if the content is too large 
- use `LV_SCROLLBAR_MODE_AUTO` same as now: `LV_SCROLLBAR_MODE_OVERFLOW` but hide the scrollbar if the object is not scrollable.

 
What do you think?

lanistor, 2023-10-16:
@kisvegabor Thanks for reply, i'm using v8, `LV_SCROLLBAR_MODE_AUTO` will enabel `drap operation` to the scroll container, which is expected.

lanistor, 2023-10-17:
> I don't get it. 🙂 So does enabling `LV_SCROLLBAR_MODE_AUTO` solve the issue?

I don't think this is a problem, it's a feature of common UI Libraries.

The behavior i want is: showing `scroll bar`, ang disable `drap operation` at the same time.

I don't know if there is a way to implement it in lvgl.

kisvegabor, 2023-10-18:
I couldn't find a simple way to do it, but here you can easily tailor LVGL to your needs :wink: 
https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_obj_scroll.c#L475-L490

closed 2023-11-09 by lvgl-bot