lvgl/lvgl #4663: Why weighted FPS is 33386

chimeiw, 2023-10-17:
I ported lvgl8.3.5 to my own MCU platform to run lv_demo_benchmark(), weighted FPS is 33386. It can run normally, but the interface will run slowly, which seems to be an error.

kisvegabor, 2023-10-17:
It might be an under or overflow somewhere. By eye, what is the real FPS approxiamtelly?

chimeiw, 2023-11-02:
> It might be an under or overflow somewhere. By eye, what is the real FPS approxiamtelly?

Thanks for you reply! Now, the FPS is 44FPS in my platform.
I'm Sorry, as a beginner in lvgl, I had trouble finding the solution to my problem for a long time.This issue was that the time base in lvgl was not accurate, so I had to modify the code according my paltform:

#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
/*Header for the system time function*/
#define LV_TICK_CUSTOM_INCLUDE "stdint.h"
/*Expression evaluating to current system time in ms*/
#define LV_TICK_CUSTOM_SYS_TIME_EXPR (custom_tick_get())
#endif    /* LV_TICK_CUSTOM */
uint32_t custom_tick_get(void)
{
    static uint32_t start_ms = 0;
	  static uint32_t now_ms;
    if(start_ms == 0) 
		{
				start_ms = k_uptime_get_32();
    }
    now_ms = k_uptime_get_32();

    uint32_t time_ms = now_ms - start_ms;
    return time_ms;
}

kisvegabor, 2023-11-02:
It's great that you have found the problem. I close this issue then.

closed 2023-11-02 by kisvegabor