lvgl/lvgl #4679: Some Bugs IN textarea and keyboad(V)

lhdjply, 2023-10-20:
When textarea is in single line mode, the keyboard presses LV_SYMBOL_NEW_LINE or LV_SYMBOL_OK will trigger LV_EVENT_READY event. But I found that when using LV_SYMBOL_NEW_LINE triggers LV_EVENT_READY, and then close the keyboard. If you press LV_SYMBOL_NEW_LINE position happens to have other widgets, which can cause accidental touch. Same operation, press LV_SYMBOL_OK, there won't be this problem.

Here is the example code:
```

lv_obj_t *keyboard;
static void passedit_event_cb(lv_event_t *e)
{

    lv_event_code_t code = lv_event_get_code(e);
    if (code == LV_EVENT_FOCUSED)
    {
        keyboard = lv_keyboard_create(lv_scr_act());
        lv_obj_set_size(keyboard, LV_HOR_RES, LV_VER_RES/2);
        lv_keyboard_set_mode(keyboard, LV_KEYBOARD_MODE_TEXT_LOWER);
        lv_keyboard_set_textarea(keyboard, e->target);
    }
    else if (code == LV_EVENT_DEFOCUSED || code == LV_EVENT_READY || code == LV_EVENT_CANCEL)
    {
        lv_obj_del(keyboard);
        lv_indev_reset(NULL, e->target);
        lv_obj_clear_state(e->target, LV_STATE_FOCUSED);
    }
}

void lv_example_textarea_1(void)
{
    lv_obj_t *textarea;
    lv_obj_t *dropdown1,*dropdown2;
    textarea = lv_textarea_create(lv_scr_act());
    lv_obj_set_width(textarea, LV_HOR_RES);
    lv_textarea_set_one_line(textarea, true);
    lv_obj_align(textarea,LV_ALIGN_TOP_LEFT, 0, 0);
    lv_obj_add_event_cb(textarea, passedit_event_cb, LV_EVENT_ALL, NULL);

    dropdown1 = lv_dropdown_create(lv_scr_act());
    lv_dropdown_set_options(dropdown1,
                            "1\n"
                            "2");
    lv_obj_align(dropdown1,LV_ALIGN_RIGHT_MID,0,LV_HOR_RES/10);

    dropdown2 = lv_dropdown_create(lv_scr_act());
    lv_dropdown_set_options(dropdown2,
                            "1\n"
                            "2");
    lv_obj_align(dropdown2,LV_ALIGN_BOTTOM_RIGHT,0,0);
}

```

kisvegabor, 2023-10-20:
Hi,

You can fix it by adding `lv_indev_wait_release(lv_indev_get_act());` instead of ` lv_indev_reset(NULL, e->target);`

lhdjply, 2023-10-21:
> Hi,
> 
> You can fix it by adding `lv_indev_wait_release(lv_indev_get_act());` instead of ` lv_indev_reset(NULL, e->target);`

Thank you. I gave it a try and it did solve the problem

kisvegabor, 2023-10-22:
Cool, I close this issue than :slightly_smiling_face:

closed 2023-10-22 by kisvegabor