lvgl/lvgl #4709: Wrong image in widgets examples

anaGrad, 2023-10-26:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I ran the animimg and imgbtn examples and the images didn't look right. You can see in the pictures below:

[image]
[image]

I first thought that it was because when setting the source and checking for the type (in lv_image_decoder_get_info() ) it returned that the image was a file and not a variable, but that behaviour doesn't happen anymore, so I am unsure about the cause of this images.

### To Reproduce

The code for the animimg widget is the one provided in the lvgl examples:
```c
LV_IMAGE_DECLARE(animimg001)
LV_IMAGE_DECLARE(animimg002)
LV_IMAGE_DECLARE(animimg003)

static const lv_image_dsc_t * anim_imgs[3] = {
    &animimg001,
    & animimg002,
    & animimg003,
};

void lv_example_animimg_1(void)
{
    lv_obj_t * animimg0 = lv_animimg_create(lv_screen_active());
    lv_obj_center(animimg0);
    lv_animimg_set_src(animimg0, (const void **) anim_imgs, 3);
    lv_animimg_set_duration(animimg0, 1000);
    lv_animimg_set_repeat_count(animimg0, LV_ANIM_REPEAT_INFINITE);
    lv_animimg_start(animimg0);
}
```
The one for imgbtn widget is also from the lvgl examples:
```c
void lv_example_imgbtn_1(void)
{
    LV_IMAGE_DECLARE(imgbtn_left);
    LV_IMAGE_DECLARE(imgbtn_right);
    LV_IMAGE_DECLARE(imgbtn_mid);

    /*Create a transition animation on width transformation and recolor.*/
    static lv_style_prop_t tr_prop[] = {LV_STYLE_TRANSFORM_WIDTH, LV_STYLE_IMAGE_RECOLOR_OPA, 0};
    static lv_style_transition_dsc_t tr;
    lv_style_transition_dsc_init(&tr, tr_prop, lv_anim_path_linear, 200, 0, NULL);

    static lv_style_t style_def;
    lv_style_init(&style_def);
    lv_style_set_text_color(&style_def, lv_color_white());
    lv_style_set_transition(&style_def, &tr);

    /*Darken the button when pressed and make it wider*/
    static lv_style_t style_pr;
    lv_style_init(&style_pr);
    lv_style_set_image_recolor_opa(&style_pr, LV_OPA_30);
    lv_style_set_image_recolor(&style_pr, lv_color_black());
    lv_style_set_transform_width(&style_pr, 20);

    /*Create an image button*/
    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_screen_active());
    lv_imgbtn_set_src(imgbtn1, LV_IMGBTN_STATE_RELEASED, &imgbtn_left, &imgbtn_mid, &imgbtn_right);
    lv_obj_add_style(imgbtn1, &style_def, 0);
    lv_obj_add_style(imgbtn1, &style_pr, LV_STATE_PRESSED);

    lv_obj_align(imgbtn1, LV_ALIGN_CENTER, 0, 0);

    /*Create a label on the image button*/
    lv_obj_t * label = lv_label_create(imgbtn1);
    lv_label_set_text(label, "Button");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, -4);
}
```

### Expected behavior

### Screenshots or video

kisvegabor, 2023-10-27:
Thanks for reporting it. 

FYI, I've fixed the images, but I'm planing to update `lv_animimg` a little bit in v9 to make it use `lv_anim` in a more transparent way. It will allow more customization for the user.

closed 2023-10-27 by kisvegabor

anaGrad, 2023-11-13:
Hi, sorry to bother you again with this, but the problem keeps happening on our side. 

[image]

kisvegabor, 2023-11-13:
I'm sorry I fixed n ly the `animimg` assets... :sweat: 

Should be fine now. See db4b26407

anaGrad, 2023-11-14:
Yes. it's perfect. Thank you!