lvgl/lvgl #4713: Textarea cannot defocused normally.

lanistor, 2023-10-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The textarea which has default focus state by `lv_obj_add_state(ta, LV_STATE_FOCUSED)`, cannot trigger `DEFOCUSED` event when other textarea clicked.

### To Reproduce

Can be reproduced with [lvgl examples](https://github.com/lvgl/lvgl/blob/v8.2.0/examples/widgets/textarea/lv_example_textarea_2.c).

Add focus state by `lv_obj_add_state(ta, LV_STATE_FOCUSED)` to the first textarea, the textarea will have cursor, and can receive input, but when you click the second textarea, the first textarea won't trigger `DEFOCUSED` event. If you click the first textarea, then the `DEFOCUSED` event will have expected behavior.

And another problems: `lv_keyboard_set_textarea` cannot resolve the upper problem.

```c
void lv_example_textarea_2(void)
{
    /*Create the password box*/
    lv_obj_t * pwd_ta = lv_textarea_create(lv_scr_act());
    lv_textarea_set_text(pwd_ta, "");
    lv_textarea_set_password_mode(pwd_ta, true);
    lv_textarea_set_one_line(pwd_ta, true);
    lv_obj_set_width(pwd_ta, lv_pct(40));
    lv_obj_set_pos(pwd_ta, 5, 20);
    lv_obj_add_event_cb(pwd_ta, ta_event_cb, LV_EVENT_ALL, NULL);

    /*Create a label and position it above the text box*/
    lv_obj_t * pwd_label = lv_label_create(lv_scr_act());
    lv_label_set_text(pwd_label, "Password:");
    lv_obj_align_to(pwd_label, pwd_ta, LV_ALIGN_OUT_TOP_LEFT, 0, 0);

    /*Create the one-line mode text area*/
    lv_obj_t * text_ta = lv_textarea_create(lv_scr_act());
    lv_textarea_set_one_line(text_ta, true);
    lv_textarea_set_password_mode(text_ta, false);
    lv_obj_set_width(text_ta, lv_pct(40));
    lv_obj_add_event_cb(text_ta, ta_event_cb, LV_EVENT_ALL, NULL);
    lv_obj_align(text_ta, LV_ALIGN_TOP_RIGHT, -5, 20);

    /*Create a label and position it above the text box*/
    lv_obj_t * oneline_label = lv_label_create(lv_scr_act());
    lv_label_set_text(oneline_label, "Text:");
    lv_obj_align_to(oneline_label, text_ta, LV_ALIGN_OUT_TOP_LEFT, 0, 0);

    /*Create a keyboard*/
    kb = lv_keyboard_create(lv_scr_act());
    lv_obj_set_size(kb,  LV_HOR_RES, LV_VER_RES / 2);

    lv_obj_add_state(pwd_ta, LV_STATE_FOCUSED);
    // lv_keyboard_set_textarea(kb, pwd_ta); /*Focus it on one of the text areas to start*/
}

static void ta_event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * ta = lv_event_get_target(e);
    if(code == LV_EVENT_CLICKED || code == LV_EVENT_FOCUSED) {
        /*Focus on the clicked text area*/
        if(kb != NULL) lv_keyboard_set_textarea(kb, ta);
    }

    else if(code == LV_EVENT_READY) {
        LV_LOG_USER("Ready, current text: %s", lv_textarea_get_text(ta));
    }
}
```

### Expected behavior

1. When click another textarea, the focused textarea trigger `DEFOCUSED` event.

### Screenshots or video

[video](https://github.com/lanistor/assets/blob/master/lvgl/textarea-1.mov)

kisvegabor, 2023-10-27:
You can FOCUSED state to any widgets but LVGL doesn't keep track of them. 

To make LVGL track the focused widgets you should use [`lv_group`](https://docs.lvgl.io/8.3/overview/indev.html#groups)

lanistor, 2023-10-27:
Thanks a lot, i resoveled it by `lv_group_focus_obj(obj)`.

closed 2023-10-27 by lanistor