lvgl/lvgl #4716: slider indicator deforms when slider is moved to the bottom

kdschlosser, 2023-10-27:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
look at the screen recording at the bottom 

### To Reproduce
paste this code into the simulator running version 9 of LVGL. I have not tested to see if the same issue exists in other versions
```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv
import display_driver

scrn = lv.scr_act()
scrn.set_style_bg_color(lv.color_hex(0x000000), lv.PART.MAIN)

slider = lv.slider(scrn)
slider.center()
slider.set_mode(lv.slider.MODE.SYMMETRICAL)
slider.set_range(10, 245)
slider.set_size(250, 150)
print(slider.get_style_radius(lv.PART.INDICATOR))
slider.set_style_outline_color(lv.color_hex(0x484848), lv.PART.MAIN)
slider.set_style_outline_width(1, lv.PART.MAIN);
slider.set_style_bg_color(lv.color_hex(0xCD3F3F), lv.PART.MAIN)
slider.set_style_bg_color(lv.color_hex(0x851D1D), lv.PART.INDICATOR)
slider.set_style_bg_opa(255, lv.PART.MAIN)
slider.set_style_bg_color(lv.color_hex(0xFFFFFF), lv.PART.KNOB)
slider.set_style_border_color(lv.color_hex(0x851D1D), lv.PART.MAIN)
slider.set_style_border_side(lv.BORDER.SIDE_TOP, 0)
slider.set_style_border_width(3, 0)
```

### Expected behavior
for the indicator not to distort

### Screenshots or video
[image]

kdschlosser, 2023-10-27:
Maybe fixing the overhang of the knob can be done at the same time.

kdschlosser, 2023-10-27:
on a side note. version 9 examples are non functioning because of lv.screen_active() not existing. It looks like the examples were updated to support the new API but the simulator was not updated. Also I am not able to save scripts in the simulator otherwise I would have posted a link.

when I click the save button I get the following error
```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://forum.lvgl.io:8443/save_file. (Reason: CORS request did not succeed). Status code: (null).
ERROR: Network Error
```

and here is the request/response data when I try and save

CORS error:
```json
{
  "log": {
    "version": "1.2",
    "creator": {
      "name": "Firefox",
      "version": "118.0.2"
    },
    "browser": {
      "name": "Firefox",
      "version": "118.0.2"
    },
    "pages": [
      {
        "id": "page_3",
        "pageTimings": {
          "onContentLoad": -3308061,
          "onLoad": -3307964
        },
        "startedDateTime": "2023-10-27T06:28:31.982-06:00",
        "title": "https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/ef1885f670e7162ce0c4c595bd35316c6a6e24da/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/ef1885f670e7162ce0c4c595bd35316c6a6e24da/examples/widgets/slider/lv_example_slider_1.py"
      }
    ],
    "entries": [
      {
        "startedDateTime": "2023-10-27T06:28:31.982-06:00",
        "request": {
          "bodySize": 0,
          "method": "OPTIONS",
          "url": "https://forum.lvgl.io:8443/save_file",
          "httpVersion": "",
          "headers": [
            {
              "name": "Host",
              "value": "forum.lvgl.io:8443"
            },
            {
              "name": "User-Agent",
              "value": "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/118.0"
            },
            {
              "name": "Accept",
              "value": "*/*"
            },
            {
              "name": "Accept-Language",
              "value": "en-US,en;q=0.5"
            },
            {
              "name": "Accept-Encoding",
              "value": "gzip, deflate, br"
            },
            {
              "name": "Access-Control-Request-Method",
              "value": "POST"
            },
            {
              "name": "Access-Control-Request-Headers",
              "value": "content-type"
[... 341 lines cut ...]
          ],
          "cookies": [],
          "queryString": [],
          "headersSize": 0
        },
        "response": {
          "status": 0,
          "statusText": "",
          "httpVersion": "",
          "headers": [],
          "cookies": [],
          "content": {},
          "redirectURL": "",
          "headersSize": 0,
          "bodySize": -1
        },
        "cache": {},
        "timings": {},
        "time": 0,
        "pageref": "page_1"
      }
    ]
  }
}
```

kdschlosser, 2023-10-27:
OH, your security certificate has expired on 9/29/2023.

kisvegabor, 2023-10-27:
It's working in C. I think just the simulator was not updated. 
[image]

I updated the certificate. We really need figure out something for this. 
For v9 we already have GitHub CodeSpace support. See here:
[image]

I wonder if something like this could be utilized for Micropython too.

closed 2023-11-18 by lvgl-bot