lvgl/lvgl #4720: How to modify the width of LV_CHART_TYPE_BAR （V8.3.10）

lhdjply, 2023-10-28:
When the number of bar charts is relatively large, the width of the bar chart will become very thin. How can I modify the width of the bar chart.

Program
```
#define BAR_CNT 100
void lv_example_chart_bar(void)
{
    lv_obj_t * chart;
    chart = lv_chart_create(lv_scr_act());
    lv_obj_set_size(chart, 400, 200);
    lv_obj_center(chart);
    lv_chart_set_type(chart, LV_CHART_TYPE_BAR);
    lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, 0, 100);
    lv_chart_set_range(chart, LV_CHART_AXIS_SECONDARY_Y, 0, 400);
    lv_chart_set_point_count(chart, BAR_CNT);

    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_X, 10, 5, BAR_CNT, 3, true, 40);
    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_Y, 10, 5, 6, 2, true, 50);

    lv_chart_set_zoom_x(chart, LV_IMG_ZOOM_NONE*BAR_CNT/5);

    lv_chart_series_t * ser1 = lv_chart_add_series(chart, lv_palette_lighten(LV_PALETTE_GREEN, 2), LV_CHART_AXIS_PRIMARY_Y);

    for(int i=0;i<BAR_CNT;i++)
    {
        lv_chart_set_next_value(chart, ser1, i);
    }
    lv_chart_refresh(chart);
}
```

kisvegabor, 2023-10-29:
Here the problem is that the gap between the bars are also scaled up so much that the bars will be very thin. I suggest setting a smaller gap:
```c
    lv_obj_set_style_pad_column(chart, 1, LV_PART_MAIN);
```

[image]

lhdjply, 2023-10-30:
> ```c
> lv_obj_set_style_pad_column(chart, 1, LV_PART_MAIN);
> ```

Thank you. This really solves the problem. But I found another problem, if there is a series 2, and the number is greater than 200, the series 2 will fill the entire chart。
```
#define BAR_CNT 200
void lv_example_chart_bar(void)
{
    lv_obj_t * chart;
    chart = lv_chart_create(lv_scr_act());
    lv_obj_set_size(chart, 400, 200);
    lv_obj_center(chart);
    lv_chart_set_type(chart, LV_CHART_TYPE_BAR);
    lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, 0, 200);
    lv_chart_set_range(chart, LV_CHART_AXIS_SECONDARY_Y, 0, 200);
    lv_chart_set_point_count(chart, BAR_CNT);
    lv_obj_set_style_pad_column(chart, 1, LV_PART_MAIN);

    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_X, 10, 5, BAR_CNT, 3, true, 40);
    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_Y, 10, 5, 6, 2, true, 100);
    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_SECONDARY_Y, 10, 5, 6, 2, true, 100);

    lv_chart_set_zoom_x(chart, LV_IMG_ZOOM_NONE*BAR_CNT/5);

    lv_chart_series_t * ser1 = lv_chart_add_series(chart, lv_palette_lighten(LV_PALETTE_GREEN, 2), LV_CHART_AXIS_PRIMARY_Y);

    for(int i=0;i<BAR_CNT;i++)
    {
        lv_chart_set_next_value(chart, ser1, i);
    }
    lv_chart_series_t * ser2 = lv_chart_add_series(chart, lv_palette_lighten(LV_PALETTE_YELLOW, 2), LV_CHART_AXIS_PRIMARY_Y);

    for(int i=0;i<BAR_CNT;i++)
    {
        lv_chart_set_next_value(chart, ser2, i+100);
    }
    lv_chart_refresh(chart);
}
```
[image]

lhdjply, 2023-10-30:
> Here the problem is that the gap between the bars are also scaled up so much that the bars will be very thin. I suggest setting a smaller gap:
> 
> ```c
>     lv_obj_set_style_pad_column(chart, 1, LV_PART_MAIN);
> ```
> 
> [image]

Thank you. This really solves the problem. But I found another problem, if there is a series 2, and the number is greater than 200, the series 2 will fill the entire chart。
```
#define BAR_CNT 200
void lv_example_chart_bar(void)
{
    lv_obj_t * chart;
    chart = lv_chart_create(lv_scr_act());
    lv_obj_set_size(chart, 400, 200);
    lv_obj_center(chart);
    lv_chart_set_type(chart, LV_CHART_TYPE_BAR);
    lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, 0, 200);
    lv_chart_set_range(chart, LV_CHART_AXIS_SECONDARY_Y, 0, 200);
    lv_chart_set_point_count(chart, BAR_CNT);
    lv_obj_set_style_pad_column(chart, 1, LV_PART_MAIN);

    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_X, 10, 5, BAR_CNT, 3, true, 40);
    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_PRIMARY_Y, 10, 5, 6, 2, true, 100);
    lv_chart_set_axis_tick(chart, LV_CHART_AXIS_SECONDARY_Y, 10, 5, 6, 2, true, 100);

    lv_chart_set_zoom_x(chart, LV_IMG_ZOOM_NONE*BAR_CNT/5);

    lv_chart_series_t * ser1 = lv_chart_add_series(chart, lv_palette_lighten(LV_PALETTE_GREEN, 2), LV_CHART_AXIS_PRIMARY_Y);

    for(int i=0;i<BAR_CNT;i++)
    {
        lv_chart_set_next_value(chart, ser1, i);
    }
    lv_chart_series_t * ser2 = lv_chart_add_series(chart, lv_palette_lighten(LV_PALETTE_YELLOW, 2), LV_CHART_AXIS_PRIMARY_Y);

    for(int i=0;i<BAR_CNT;i++)
    {
        lv_chart_set_next_value(chart, ser2, i+100);
    }
    lv_chart_refresh(chart);
}
```
[image]

kisvegabor, 2023-10-30:
`lv_obj_set_style_pad_column(chart, 0, LV_PART_ITEMS;` should do the trick.

lhdjply, 2023-10-30:
> `lv_obj_set_style_pad_column(chart, 0, LV_PART_ITEMS);` should do the trick.

Thank you!!

closed 2023-10-30 by lhdjply