lvgl/lvgl #4733: Image scaling and tiling

kisvegabor, 2023-10-31:
Continue #4642, #4102, and #3174

### Introduce the problem

It would be useful to have more standard CSS like features to scale images automatically according to the parent and some other settings.

### Suggested solution

We should consider the following CSS proeprties:
- [object-fit](https://css-tricks.com/almanac/properties/o/object-fit/)
- [background-attachment](https://css-tricks.com/almanac/properties/b/background-attachment/)
- [background-image](https://css-tricks.com/almanac/properties/b/background-image/)
- [background-origin](https://css-tricks.com/almanac/properties/b/background-origin/)
- [background-position](https://css-tricks.com/almanac/properties/b/background-position/)
- [background-repeat](https://css-tricks.com/almanac/properties/b/background-repeat/)
- [background-size](https://css-tricks.com/almanac/properties/b/background-size/)

My idea is to create `LV_PART_BACKGROUND` instead of the `style_bg_img` properties. This way we could use all the style properties on the background as well, including 
- `align` 
- `x`, `y`
- `width`, `height`
- `rotation`, , `scale_x`, `scale_y`, `pivot_x`, `pivot_y`
- `img_recolor`.
- `translate_x`, `translate_y`

Besides we should have some new style properties to automatically set these properties for the background and the normal `lv_image` widget too. These should be a subset of the  [object-fit](https://css-tricks.com/almanac/properties/o/object-fit/) property. I think the `repeat` properties should be part of `object-fit` as they are mutually exclusive. (you can't apply `cover` and `repeat` at the same time)

A helper function like `lv_image_calculate_fit(img1, LV_OBJECT_FIT_..., &scale_x, &scale_y)` would do the job.

I think tiling should be handled in the draw unit as it can be accelerated. 

What do you think?

@GorgonMeducer @nicusorcitu @XuNeo @W-Mai

GorgonMeducer, 2023-10-31:
> I think tiling should be handled in the draw unit as it can be accelerated.

I strongly agree with you!

kisvegabor, 2023-11-06:
I started to work on it and as it usually happens some questions came up.

I realized having an `LV_PART_BACKGROUND` is not a good idea as it wouldn't allow adding a background image to e.g. and arc knob or a slider knob or indicator. I guess there is a reason why it's no a pseudo-element in CSS either but normal properties.

So to have a really flexible background we need to duplicate a lot of style properties: align, x, y, scale, rotation, pivot, tiling.

The other thing I was thinking about what properties shell we have from `object-fit`. Finally I concluded that having an `lv_image_align_t` type with:
- `LV_IMAGE_ALIGN_TOP_MID`
- `LV_IMAGE_ALIGN_TOP_LEFT`
- etc
- `LV_IMAGE_ALIGN_STRETCH`: set the scale to fill the object in both direction. 
- `LV_IMAGE_TILE`: repeat in both directions

I think it should cover most of the use cases. What do you think? 

And the last thing that I don't know if it's required to support multiple backgrounds. A use case could be to create a bar indicator from:
1. a tiling pattern
2. some animated particles on top of the pattern
3. a glowing light on the right

In CSS it's supported and it really can be quite useful in some cases.

We could convert all background properties to pointers (arrays) and have `background-count` property. 

What do you think?

GorgonMeducer, 2023-11-07:
Sorry, I don't have sufficient experience to answer this question. Same as you, I would like to hear others' ideas first.

kisvegabor, 2023-11-07:
Finally I added the stretch and tile feature to `lv_image` only and left the background as it is. 

I realized that there are too many options with the background to handle it in a generic way so the suggested approach is to use a draw event and draw images manually if a complex background is needed. 

See d02562b79 

Just to illustrate a really complex case for the background:
- have multiple background on to of each other
- each positioned differently
- some of them might be tiled
- the offset a background is animated
- some background are blended with additive blending
- some background are transformed
- all these are added to an arc knob
- all these should completely different when the arc is pressed  
- a transition should be applied when the arc goes to presses state 

---

I consider this feature ready and I close this issue. Feel free to comment here or reopen it if you have any remarks.

closed 2023-11-07 by kisvegabor

W-Mai, 2023-11-08:
I think `cover` and `contain` in `obj-fit` are also very useful functions. Should we also add them?

I have simple realize them in this pr before https://github.com/lvgl/lvgl/pull/4106

kisvegabor, 2023-11-08:
From this point we can easily add the other options too. They are mainly just setting different scale values. We should also consider if they need to work with the alignment options.  E.g. in case of contain, the the image can be left/right/center aligned. 

Could you also mention a use case for `cover` and `contain` in `obj-fit`? On an embedded UI I couldn't imagine real life scenarios for these.

C-Ackerman, 2024-02-19:
> Finally I added the stretch and tile feature to `lv_image` only and left the background as it is.
> 
> I realized that there are too many options with the background to handle it in a generic way so the suggested approach is to use a draw event and draw images manually if a complex background is needed.
> 
> See [d02562b](https://github.com/lvgl/lvgl/commit/d02562b796fcc140fd8781f59b5d2cd340ac0ebd)
> 
> Just to illustrate a really complex case for the background:
> 
> * have multiple background on to of each other
> * each positioned differently
> * some of them might be tiled
> * the offset a background is animated
> * some background are blended with additive blending
> * some background are transformed
> * all these are added to an arc knob
> * all these should completely different when the arc is pressed
> * a transition should be applied when the arc goes to presses state
> 
> I consider this feature ready and I close this issue. Feel free to comment here or reopen it if you have any remarks.

hi gabor,
When I use lv_obj_get_style_bg_image_src to set the BackgroundImage, then how to set the background-size

kisvegabor, 2024-02-19:
Hi @C-Ackerman,

The style properties of background image are not so powerful yet. I suggest using an `lv_image` where you have more control, especially with the new `inner_align` style property.

C-Ackerman, 2024-03-12:
Hi @kisvegabor ,
There are several scenarios that need support
1. lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE); setting scale at the same time does not take effect
For example, to achieve the effect of background-size: 5%; background-repeat: repeat;
[image]

```
void lv_example_style_6(void)
{

    lv_obj_t * native = lv_obj_create(lv_screen_active());
    lv_obj_set_size(native, 450, 450);
    static lv_style_t style;
    lv_style_init(&style);
    // lv_style_set_bg_color(&style, lv_palette_lighten(LV_PALETTE_GREY, 3));
    lv_style_set_border_width(&style, 2);
    lv_style_set_border_color(&style, lv_palette_lighten(LV_PALETTE_YELLOW, 3));
    lv_obj_add_style(native, &style, 0);

    lv_obj_t * bgimg = lv_image_create(native);
    lv_image_set_src(bgimg, "/data/app/logo.png");
    // lv_obj_center(bgimg);
    // lv_image_set_pivot(bgimg, 50, 100);
    lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5);
    // lv_image_set_offset_x(bgimg, 50);
    // lv_obj_set_pos(bgimg,  260, 150);
    // lv_obj_set_x(bgimg, 100);
    // lv_image_set_scale_x(bgimg, LV_SCALE_NONE * 2);
    // int widgetWidth = lv_obj_get_style_width(bgimg, LV_PART_MAIN);
    // lv_obj_set_size(bgimg, 50, LV_SIZE_CONTENT);
    lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE);

    lv_obj_t * panel1 = lv_obj_create(bgimg);
    lv_obj_set_size(panel1, 440, 440);
    lv_obj_center(panel1);
    static lv_style_t style1;
    lv_style_init(&style1);
    lv_style_set_opa(&style1, 50);
    lv_style_set_border_width(&style1, 2);
    lv_style_set_border_color(&style1, lv_palette_lighten(LV_PALETTE_PINK, 3));
    lv_obj_add_style(panel1, &style1, 0);

    lv_obj_t * label = lv_label_create(bgimg);
    lv_label_set_text(label, "Widgets demo");
    lv_obj_set_size(label, 100, 200);
    lv_obj_align(label, LV_ALIGN_CENTER, 50, 100);
}
```
2. If I want to achieve the effect of background-size: 5%; background-repeat: repeat-x;, can it only be achieved through lv_obj_set_size(bgimg, 50, LV_SIZE_CONTENT); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE);, but modify the image like this The size will affect its child nodes, so the child nodes also need to set LV_OBJ_FLAG_OVERFLOW_VISIBLE to control the rendering area?

[image]

3.If I want to implement background-repeat: repeat-y; background-position: 25% 50%; effect,
lv_image_set_offset_x(bgimg, 50); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE); cannot take effect at the same time.
lv_obj_set_x(bgimg, 100); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE); can take effect at the same time, but this will also change the position of the child node, so you need to adjust the position of the child node again.

[image]

kisvegabor, 2024-03-13:
Please open a new issue for it for better visibility.

C-Ackerman, 2024-03-13:
> lv_image_set_offset_x(bgimg, 50); and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE); cannot take effect at the same time.
> lv_obj_set_x(bgimg, 100); and lv_image_set_align(bgimg, [...]

@kisvegabor 

Divided into three questions

https://github.com/lvgl/lvgl/issues/5858
https://github.com/lvgl/lvgl/issues/5859
https://github.com/lvgl/lvgl/issues/5860