lvgl/lvgl #4752: lv_display_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED) doesn't work.

ngc6589, 2023-11-04:
I created a sample to try out LVGL v9-dev (master), but a problem occurred.

- Raspberry Pi Pico(RP2040)
- ILI9341(MSP2807) 320x240 lcd with touch 16bit COLOR

■test procedure
screen = lv_obj_create(NULL);
lv_obj_set_style_bg_color(screen, lv_color_hex(0xFF0000), 0);
lv_display_load_scr(screen);

I changed the value of lv_color_hex to RGB and checked the value of the drawing buffer.

■LVGL 8.3.10
```
lv_conf.h
#define LV_COLOR_DEPTH 16
#define LV_COLOR_16_SWAP 0

R 00F8 00F8 00F8 00F8 ...
G E007 E007 E007 E007 ...
B 1F00 1F00 1F00 1F00 ...
```
```
lv_conf.h
#define LV_COLOR_DEPTH 16
#define LV_COLOR_16_SWAP 1

R F800 F800 F800 F800 ...
G 07E0 F800 F800 F800 ...
B 001F 001F 001F 001F ...
```
Buffer data can be displayed correctly with the combination of 
ILI9341 (MSP2807) Memory Access Control (36h) BGR settings.

■LVGL 9-dev
```
lv_conf.h
#define LV_USE_DEV_VERSION
#define LV_COLOR_DEPTH 16
#define LV_MEM_SIZE (48U * 1024U)
#define  LV_LAYER_MAX_MEMORY_USAGE 16

R 00F8 00F8 00F8 00F8 ...
G E007 E007 E007 E007 ...
B 1F00 1F00 1F00 1F00 ...
```
Same as lvgl 8.3.10
```
lv_conf.h
#define LV_USE_DEV_VERSION
#define LV_COLOR_DEPTH 16
#define LV_MEM_SIZE (48U * 1024U)
#define  LV_LAYER_MAX_MEMORY_USAGE 16
with
lv_display_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED);

R 4001 f000 ffff ffff 
G 0000 0000 0000 0000 
B 0000 0000 0000 0000 
```
It seems that the buffer data is not reversed when 
lv_display_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED); is specified.

I hope this issue will be resolved when v9 is released.

PGNetHun, 2023-11-05:
Hello!
This is a known issue: https://forum.lvgl.io/t/maintainer-wanted/12837/22?u=pgnet 
Not only ILI9341, but also ST7789 does not show colors correctly (I tested only MicroPython binding on real device).
Please till it is not fixed in v9-dev, use 8.3.

ngc6589, 2023-11-05:
It was reported on  Micropython in the VGL forum.
I understand.
thank you.

kisvegabor, 2023-11-06:
In d4613aecb I've added `lv_draw_sw_rgb565_swap()` you can call it in the `flush_cb` to swap the bytes in the buffer. 

I think it's a more transparent way of handling it, instead of having `LV_COLOR_FORMAT_NATIVE_REVERSED`. We can provide similar functions if some kind of hardware acceleration is available, so that you can pick the one fits the best for you.

ngc6589, 2023-11-06:
■LVGL-9dev commit d4613ae
LCD callback function with **lv_draw_sw_rgb565_swap()**
```
R F800 F800 F800 F800 ...
G 07E0 F800 F800 F800 ...
B 001F 001F 001F 001F ...
```
I can confirm that RBG565 is inverted, and the LCD display is now OK.
Thank you.

closed 2023-11-06 by kisvegabor

PGNetHun, 2023-11-06:
It's great, thank you Gabor!

I think also that `LV_COLOR_FORMAT_NATIVE_REVERSED` can be omitted from color formats, as it does not represent 'real' color format (like `LV_COLOR_FORMAT_NATIVE`)

kisvegabor, 2023-11-07:
It as removed in the same commit. See [here](https://github.com/lvgl/lvgl/commit/d4613aecbe960535702e55bb582f482a40975a7e#diff-7c5a79b9166b64c2b88bb889fec3ec27a44e62b94682b38221f47cf3683a1c05).

crosser, 2024-10-13:
> In [d4613ae](https://github.com/lvgl/lvgl/commit/d4613aecbe960535702e55bb582f482a40975a7e) I've added `lv_draw_sw_rgb565_swap()` you can call it in the `flush_cb` to swap the bytes in the buffer.
> 
> I think it's a more transparent way of handling it, instead of having `LV_COLOR_FORMAT_NATIVE_REVERSED`. We can provide similar functions if some kind of hardware acceleration is available, so that you can pick the one fits the best for you.

But it means that every byte of every  buffer is being written twice, correct?

Isn't it more efficient to fill the buffer with "correct" (for the hardware) byteorder from the start?

I tried to use this userspace hack

```
lv_color_t c_swap(lv_color_t o)
{
        return (lv_color_t) {
        .red =   ((o.green << 3) & 0b11100000) | ((o.blue >> 3)  & 0b00011000),
        .green = ((o.blue << 2)  & 0b11100000) | ((o.red >> 3)   & 0b00011100),
        .blue =  ((o.red << 3)   & 0b11000000) | ((o.green >> 2) & 0b00111000)
        };
}
```

and it almost works, except antialiasing produces partial pixels of wrong colour (which can be expected).

Is there a chance to move byte swapping to `lv_color_to_u16()` / `lv_color_to_u32()`?

kisvegabor, 2024-10-15:
It's really faster for fill, but for anti-aliasing and image blending it's slower. According to our benchmarks in real life use cases there is not noticeable performance change.

crosser, 2024-10-15:
I guess it's fair enough...

For or my use case, I do need some text and icons with antialiasing, but they are drawn rarely, and can be slow. The main display however is of a real time signal trace, basically "put these N pixels at there coordinates", and that should better be fast.  I will have to see how it goes. Thanks for the response, in any case!