lvgl/lvgl #4753: All the examples are broken

gitcnd, 2023-11-04:
Except the first, all others here https://docs.lvgl.io/master/examples.html say this:-

```
Running the SDL lvgl version
Traceback (most recent call last):
  File "<stdin>", line , in <module>
AttributeError: no such attribute

MicroPython v1.20.0-1030-gf94797b9c-dirty on 2023-08-31; JS with Emscripten
Type "help()" for more information.
>>> 
>>> 

```

... in the simulator.

e.g. here:-

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0bdd54abedf352e82acfbaabe8be48324f9435a2/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0bdd54abedf352e82acfbaabe8be48324f9435a2/examples/get_started/lv_example_get_started_2.py

kisvegabor, 2023-11-06:
@embeddedt Can you take a look at it? There were a larger refactoring in LVGL's API, and probably some MicroPython-JS parts were not updated. Most importantly `lv_disp_...` was renamed to `lv_display_...`

embeddedt, 2023-11-06:
Because lv_micropython v9 development is happening on the `feat/multi-instance` branch, rather than `master`, the changes are not merged to the simulator's branch automatically. After merging `feat/multi-instance` manually the examples appear to be functional again.

kisvegabor, 2023-11-07:
Thank you @embeddedt. Could you tell exactly which repo's which branch is used by the simulator?

> I'm interested in building this as a loadable binary .mpy for (at least) my esp32. Is there someone I can chat with about this effort?

I'll reply in the forum.

embeddedt, 2023-11-07:
The v9 simulator uses the `lvgl_javascript_v9` branch of the https://github.com/lvgl/lv_micropython repository.

The workflow that merges new commits on `master` into that branch is here: https://github.com/lvgl/lv_micropython/blob/master/.github/workflows/merge-to-js.yml

kisvegabor, 2023-11-08:
Clear, thank you! 

Closing this issue.

closed 2023-11-08 by kisvegabor