lvgl/lvgl #4756: Need help to display full screen on ILI9488 vs ESP32S3

f16g53, 2023-11-06:
Hi, i am new to LVGL.

### Describe the bug
I have a problem that the screen is not showing full size.
When I try printing the size in the flush_cb function (`ili9488_flush` in `il9488.c`), I see the size is missing 1

TFT: `ILI9488`
IDF version: `4.4.6`
LVGL version: `8.3.0`
Driver version: https://github.com/lvgl/lvgl_esp32_drivers/tree/develop/lvgl_7.11.0_idf_5.0 (modified to work on` eps32s3`)

Environment:
`Windows 11`
`Python: 3.11`
`ESP-IDF Tool: 2.23`

### ili9488.c
```
// Flush function based on mvturnho repo
void ili9488_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map) {
  uint32_t size = lv_area_get_width(area) * lv_area_get_height(area);

  lv_color16_t *buffer_16bit = (lv_color16_t *) color_map;
  uint8_t *mybuf;
  do {
    mybuf = (uint8_t *) heap_caps_malloc(3 * size * sizeof(uint8_t), MALLOC_CAP_DMA);
    if (mybuf == NULL) ESP_LOGW(TAG, "Could not allocate enough DMA memory!");
  }
  while (mybuf == NULL);

  uint32_t LD = 0;
  uint32_t j = 0;

  for (uint32_t i = 0; i < size; i++) {
    LD = buffer_16bit[i].full;
    mybuf[j] = (uint8_t) (((LD & 0xF800) >> 8) | ((LD & 0x8000) >> 13));
    j++;
    mybuf[j] = (uint8_t) ((LD & 0x07E0) >> 3);
    j++;
    mybuf[j] = (uint8_t) (((LD & 0x001F) << 3) | ((LD & 0x0010) >> 2));
    j++;
  }
  printf("x1: %d\n", area->x1); // <--- print to check
  printf("x2: %d\n", area->x2);
  printf("y1: %d\n", area->y1);
  printf("y2: %d\n", area->y2);

  /* Column addresses  */
  uint8_t xb[] = {
      (uint8_t) (area->x1 >> 8) & 0xFF,
      (uint8_t) (area->x1) & 0xFF,
      (uint8_t) (area->x2 >> 8) & 0xFF,
      (uint8_t) (area->x2) & 0xFF,
  };

  /* Page addresses  */
  uint8_t yb[] = {
      (uint8_t) (area->y1 >> 8) & 0xFF,
      (uint8_t) (area->y1) & 0xFF,
      (uint8_t) (area->y2 >> 8) & 0xFF,
      (uint8_t) (area->y2) & 0xFF,
  };

  /*Column addresses*/
  ili9488_send_cmd(ILI9488_CMD_COLUMN_ADDRESS_SET);
  ili9488_send_data(xb, 4);

  /*Page addresses*/
  ili9488_send_cmd(ILI9488_CMD_PAGE_ADDRESS_SET);
  ili9488_send_data(yb, 4);

  /*Memory write*/
  ili9488_send_cmd(ILI9488_CMD_MEMORY_WRITE);

  ili9488_send_color((void *) mybuf, size * 3);
  heap_caps_free(mybuf);

  lv_disp_flush_ready(drv);
}
```

monitor:

```
x1: 0
x2: 319
y1: 0
y2: 2
```

### main.cpp

```
#define TAG "demo"
#define LV_TICK_PERIOD_MS 1

static void lv_tick_task(void *arg);
static void guiTask(void *pvParameter);

extern "C" void app_main() {
  xTaskCreatePinnedToCore(guiTask, "gui", 4096 * 2, NULL, 0, NULL, 1);
}

SemaphoreHandle_t xGuiSemaphore;

static void guiTask(void *pvParameter) {
  (void) pvParameter;
  xGuiSemaphore = xSemaphoreCreateMutex();

  lv_init();
  lvgl_driver_init();

  auto *buf1 = heap_caps_malloc(DISP_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_DMA);
  assert(buf1 != NULL);

#ifndef CONFIG_LV_TFT_DISPLAY_MONOCHROME
  auto *buf2 = heap_caps_malloc(DISP_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_DMA);
  assert(buf2 != NULL);
#else
  static lv_color_t *buf2 = NULL;
#endif

  static lv_disp_draw_buf_t disp_buf;

  uint32_t size_in_px = DISP_BUF_SIZE;

#if defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_IL3820         \
    || defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_JD79653A    \
    || defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_UC8151D     \
    || defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_SSD1306
  size_in_px *= 8;
#endif

  lv_disp_draw_buf_init(&disp_buf, buf1, buf2, size_in_px);

  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.flush_cb = disp_driver_flush;
  disp_drv.ver_res = 480;
  disp_drv.hor_res = 320;
  
#ifdef CONFIG_LV_TFT_DISPLAY_MONOCHROME
  disp_drv.rounder_cb = disp_driver_rounder;
  disp_drv.set_px_cb = disp_driver_set_px;
#endif

  disp_drv.draw_buf = &disp_buf;
  lv_disp_drv_register(&disp_drv);

#if CONFIG_LV_TOUCH_CONTROLLER != TOUCH_CONTROLLER_NONE
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.read_cb = touch_driver_read;
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  lv_indev_drv_register(&indev_drv);
#endif

  const esp_timer_create_args_t periodic_timer_args = {
      .callback = &lv_tick_task,
      .name = "periodic_gui"
  };
  esp_timer_handle_t periodic_timer;
  ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
  ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, LV_TICK_PERIOD_MS * 1000));

  lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0x21242c), LV_PART_MAIN);

  static lv_style_t style;
  lv_style_init(&style);
  lv_style_set_text_font(&style, &lv_font_montserrat_48); // <--- you have to enable other font sizes in menuconfig

  lv_obj_t *label1 = lv_label_create(lv_scr_act());
  lv_label_set_long_mode(label1, LV_LABEL_LONG_WRAP);     /*Break the long lines*/
  lv_label_set_recolor(label1, true);                      /*Enable re-coloring by commands in the text*/
  lv_label_set_text(label1, "Hello, world");
  lv_obj_set_width(label1, 320);  /*Set smaller width to make the lines wrap*/
  lv_obj_set_style_text_align(label1, LV_TEXT_ALIGN_CENTER, 0);
  lv_obj_align(label1, LV_ALIGN_CENTER, 0, -40);
  lv_obj_add_style(label1, &style, 0);  // <--- obj is the label

  while (1) {
    vTaskDelay(pdMS_TO_TICKS(10));

    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY)) {
      lv_task_handler();
      xSemaphoreGive(xGuiSemaphore);
    }
  }

  free(buf1);
#ifndef CONFIG_LV_TFT_DISPLAY_MONOCHROME
  free(buf2);
#endif
  vTaskDelete(NULL);
}

static void lv_tick_task(void *arg) {
  (void) arg;
  lv_tick_inc(LV_TICK_PERIOD_MS);
}

```

### Expected behavior
Please let me know where to edit so that the screen is fully displayed.

Thanks a lot!

### ---------------
[image]

[image]

### Another example

```
static void set_x(void * var, int32_t v)
{
    lv_obj_set_x(var, v);
}

...
 lv_obj_t * obj = lv_obj_create(lv_scr_act());
  lv_obj_center(obj);

  lv_obj_t * label = lv_label_create(obj);
  lv_label_set_text(label, "Home");
  lv_obj_add_flag(label, LV_OBJ_FLAG_FLOATING); /*To be ignored by scrollbars*/
  lv_obj_update_layout(obj);    /*To have the correct size for lv_obj_get_width*/

  lv_anim_t a;
  lv_anim_init(&a);
  lv_anim_set_var(&a, label);
  lv_anim_set_values(&a, lv_obj_get_width(obj) + 20, -lv_obj_get_width(label) - 20);
  lv_anim_set_time(&a, 3000);
  lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
  lv_anim_set_exec_cb(&a, set_x);
  lv_anim_start(&a);

```
[image]

kisvegabor, 2023-11-06:
It seems you have provided 3 lines of buffer for LVGL What if you provide e.g. 20 lines?

f16g53, 2023-11-06:
Do you mean `ili9488_send_color((void *) mybuf, size * 3); <---- change here?`
sorry I'm not good at this part, I just use the library's default

f16g53, 2023-11-07:
well, when i replaced 3 with 20, custom buffer size in menuconfig was 4096 and kept 3 in `ili9488_send_color`, now it is working

```

// Flush function based on mvturnho repo
void ili9488_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map) {

  ...

  do {
    mybuf = (uint8_t *) heap_caps_malloc(20 * size * sizeof(uint8_t), MALLOC_CAP_DMA);
    if (mybuf == NULL) ESP_LOGW(TAG, "Could not allocate enough DMA memory!");
  }

  ...

  ili9488_send_color((void *) mybuf, size * 3);

  ...
}

```
**Thank you so much!**

[image]

closed 2023-11-07 by f16g53

kisvegabor, 2023-11-07:
Great, however, I'm not perfectly sure how it solved the issue. :smile: