lvgl/lvgl #4765: lv_conf_template.h Version description not updated

landaming, 2023-11-08:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

In v8.3.10, lv_conf_template.h The version description of the file has not been updated
### To Reproduce

### Expected behavior

### Screenshots or video

[image]

kisvegabor, 2023-11-08:
Thank you for pointing it out.  I think there is no good way to fix it after the the release. What we can do is setting it to v8.3.11 in the next release. :slightly_frowning_face: 

cc @GorgonMeducer

GorgonMeducer, 2023-11-09:
Sorry, I will double check next time.

kisvegabor, 2023-11-09:
No worries. :+1:

closed 2023-11-09 by kisvegabor