lvgl/lvgl #4771: obj's border opa display is not normal in lvgl8.3.10

guoqiang-gump, 2023-11-09:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-11-10:
Please use `lv_obj_set_style_opa_layered`. 

Background story: 
In v8.3 we have introduced a "layers" to allow correct blending with opacity and tranformation of any widgets. For these kind of operations LVGL creates a spanshot about he widget and blends it as a single image. In case of opacity it results in better result, but it's also slower and uses much more memory. So we have introduced the `opa_layered` property to explicitly tell that you want slower and more memory hungry mode.

guoqiang-gump, 2023-11-10:
Thanks for the reply.
lv_obj_set_style_opa_layered does have an effect. I've been able to tell the difference between lvgl 8.3.10 and lvgl 8.3.7, and learned that lv_obj_set_style_opa_layered() replaces lv_obj_set_style_opa().

But when I call lv_scr_load_anim(&ui_Home, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0):  lv_scr_load_anim --->opa_scale_anim ---->lv_obj_set_style_opa, which ends up calling This makes the FADE_ON or FADE_OUT screen switching strange.

kisvegabor, 2023-11-10:
Hm, I see. I think the best would be to have `LV_SCR_LOAD_ANIM_FADE_ON_LAYERED` too. What do you think?

guoqiang-gump, 2023-11-11:
Yes, I think adding LV_SCR_LOAD_ANIM_FADE_ON_LAYERED is necessary because after many developers upgraded from lvgl8.3.7 to lvgl8.3.10, they use LV_SCR_LOAD_ANIM_FADE_ON and they find that the scr opa displays weirdly. So need LV_SCR_ LOAD_ANIM_FADE_ON_LAYERED to correspond to lv_obj_set_style_opa.

Also I found the biggest upgrade surprise for lvgl 8.3.10, a huge frame rate increase using double fb and dirct_mode. I think lv_port_disp_template.c should be updated to guide developers to use double fb and dirct_mode.

kisvegabor, 2023-11-13:
>  So need LV_SCR_ LOAD_ANIM_FADE_ON_LAYERED to correspond to lv_obj_set_style_opa.

Almost. `LV_SCR_ LOAD_ANIM_FADE_ON_LAYERED` should correspond to `lv_obj_set_style_opa_layered` (which allcoates a new layer and has higher quality).

Could you help us by sending a PR?

> Also I found the biggest upgrade surprise for lvgl 8.3.10, a huge frame rate increase using double fb and dirct_mode. I think lv_port_disp_template.c should be updated to guide developers to use double fb and dirct_mode.

Sure! Approximately how much was the increase?

guoqiang-gump, 2023-11-14:
> Almost. LV_SCR_ LOAD_ANIM_FADE_ON_LAYERED should correspond to lv_obj_set_style_opa_layered (which allcoates a new layer and has higher quality).
Could you help us by sending a PR?
I've been very busy lately and won't be available until some time later. Until then, I'll have to see about submitting PR.

> Sure! Approximately how much was the increase?
I have a GPU inside my MCU, and direct_mode doubles the frame rate over full_reflesh, currently at 58FPS inside my clock_demo

kisvegabor, 2023-11-15:
> I've been very busy lately and won't be available until some time later. Until then, I'll have to see about submitting PR.

I'm quite busy too, so I leave this issue open for now.

> I have a GPU inside my MCU, and direct_mode doubles the frame rate over full_reflesh, currently at 58FPS inside my clock_demo

Really cool!

closed 2023-12-07 by lvgl-bot