lvgl/lvgl #4772: The opa set by the lv_obj_set_style_opa api does not display properly in lvgl8.3.10

guoqiang-gump, 2023-11-09:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The opa set by the lv_obj_set_style_opa api does not display properly in lvgl8.3.10.

### To Reproduce
    lv_obj_t *ui_PowerSlider = lv_slider_create(root);
    lv_obj_set_width(ui_PowerSlider, 100);
    lv_obj_set_height(ui_PowerSlider, 233);
    lv_slider_set_value(ui_PowerSlider, 0, LV_ANIM_OFF);
    lv_obj_set_x(ui_PowerSlider, 0);
    lv_obj_set_y(ui_PowerSlider, -30);
    lv_obj_set_align(ui_PowerSlider, LV_ALIGN_CENTER);
    lv_obj_set_style_radius(ui_PowerSlider, 40, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_color(ui_PowerSlider, lv_color_hex(0x202020), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_PowerSlider, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_color(ui_PowerSlider, lv_color_hex(0xFF3F64), LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_PowerSlider, 255, LV_PART_INDICATOR | LV_STATE_DEFAULT);
    LV_IMG_DECLARE(ui_img_powericon_png);                   // assets\powerIcon.png
    lv_obj_set_style_bg_img_src(ui_PowerSlider, &ui_img_powericon_png, LV_PART_KNOB | LV_STATE_DEFAULT);
    lv_obj_set_style_opa(ui_PowerSlider, LV_OPA_50, 0);

### Expected behavior
The opa of each part of obj is displayed normally

### Screenshots or video
[image]

strijar, 2023-11-10:
I also encountered the same problem. At the same time, it works for some objects, and for others it does not.

kisvegabor, 2023-11-10:
Should be related to https://github.com/lvgl/lvgl/issues/4771

closed 2023-12-02 by lvgl-bot