lvgl/lvgl #4779: Will lvgl support 16 levels of grayscale color?

lanistor, 2023-11-10:
Combining lvgl and ink screen, we face a problem: the pixel need to convert to color of epaper(16 levels of grayscale color), when using `set_px_cb`, the painting will be very slow, [see previous testing result](https://github.com/lvgl/lvgl/issues/3921#issuecomment-1407588802).

We are using [epdiy](https://github.com/vroland/epdiy/tree/s3_lcd), the color rule is this: only have 0~15 colors, represents different grayscale colors(Eink paper's rule), each pixel occupies 4 bits, the high 4 bits and low 4 bits of a byte represent two adjacent pixels respectively.

So how to resolve this problem to improve rendering speed?

### Suggested solution

I think out a way: to let lvgl support `16 levels of grayscale color`, so the pixel doesn't need to convert by `set_px_cb`.

Do you think it's a good way? or any other ways?

kisvegabor, 2023-11-10:
Hi,

For LVGL v8 (that you are using now) we are not planning to add such a features. In LVGL v9 (under development) we have much more flexible rendering architecture, where 4 bit rendering is not supported yet but it could be in the future. 

To make it faster with v8, I recommend using normal 16 bit rendering and convert it to 4 bit in the `flush_cb`.

closed 2023-12-02 by lvgl-bot

mataide, 2023-12-08:
@lanistor Were you able to combine Epdiy with Lvgl? I am trying to do the same.

lanistor, 2023-12-09:
> @lanistor Were you able to combine Epdiy with Lvgl? I am trying to do the same.

You can use this: https://github.com/martinberlin/lvgl_epaper_drivers