lvgl/lvgl #4782: The brower_file_event_handler interface has hidden risks

sunzigang, 2023-11-10:
<!  --
IMPORTANT
Issues that don't use this template will be ignored and closed.
-->

### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
<!  --
A clear and concise description of what the bug is.
-->
When clicking on an empty space in the file list, an illegal access is generated

### To Reproduce
<!  --
Provide a small, independent code sample that can be used to reproduce the issue.
Ideally this should work in the PC simulator unless the problem is specific to a platform.

Use this routine: LVGL/examples/others/file_explorer
/lv_example_file_explorer_1.c

### Expected behavior
<!  --
A clear and concise description of what you expected to happen.
-->
No illegal access is generated

### Screenshots or video
<!  --
If applicable, add screenshots to help explain your problem.
-->
[image]
 here return ""
[image]
str_fn = str_fn + 5;

sunzigang, 2023-11-10:
[image]
Should there be a judgment?

kisvegabor, 2023-11-10:
Nice catch! I think we should return `NULL` and also check the return value where needed. Could you send a Pull request with the fix?

closed 2023-12-02 by lvgl-bot