lvgl/lvgl #4785: sjpg can't load big .jpg

Jeeeeeiel, 2023-11-10:
### Introduce the problem

sjpg cannot load .jpg with size 240*240 on platform arduino+esp32 picoD4.
error happens here:
"sjpeg->frame_cache = (void *)lv_mem_alloc(sjpeg->sjpeg_x_res * sjpeg->sjpeg_single_frame_height * 3);"
can anybody help me?thx

### Examples and cases

### Suggested solution

Jeeeeeiel, 2023-11-11:
how can i load .jpg with sjpg segmented？i mean piece by piece

kisvegabor, 2023-11-13:
Seems like a out of memory issue. Try increasing `LV_MEM_SIZE` in `lv_conf.h`

Jeeeeeiel, 2024-02-11:
> Seems like a out of memory issue. Try increasing `LV_MEM_SIZE` in `lv_conf.h`

not a solution😓

kisvegabor, 2024-02-12:
Let's start form the beginning: 
- Which version of LVGL do you use?
- If v8, I recommend updating to v9. See the changelog [here](https://docs.lvgl.io/master/CHANGELOG.html)
- In v9 SJPG was replaced by TJPGD
- Please using your image in [PC simulator](https://docs.lvgl.io/master/integration/ide/pc-simulator.html) first to rule out as much hardware related issues as possible.

closed 2024-03-23 by lvgl-bot