lvgl/lvgl #4786: Bug with Roller widget when there is more than 36 items in infinite mode

rlagace-cld, 2023-11-10:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
If there is more than 36 items in a roller when in infinite mode, the rendering of the selected item is screwed up. In normal mode, there is no bug.

### To Reproduce
```c
roller1 = lv.roller(lv.scr_act())
roller1.set_options("\n".join([
    "01",
    "02",
    "03",
    "04",
    "05",
    "06",
    "07",
    "08",
    "09",
    "10",
    "11",
    "12",
    "13",
    "14",
    "15",
    "16",
    "17",
    "18",
    "19",
    "20",
    "21",
    "22",
    "23",
    "24",
    "25",
    "26",
    "27",
    "28",
    "29",
    "30",
    "31",
    "32",
    "33",
    "34",
    "35",
    "36",
    "37",
    "38",
    "39",
    "40",
    "41",
    "42",
    "43",
    "44",
    "45",
    "46",
    "47",
    "48",
    "49",
    "50",
    "51",
    "52",
    "53",
    "54",
    "55",
    "56",
    "57",
    "58",
    "59",
    ]),lv.roller.MODE.INFINITE)

roller1.set_visible_row_count(5)
roller1.center()
```

### Expected behavior
There should be no glitch in the rendering.

### Screenshots or video
With 36 items:
[image]
With 37 items:
[image]
With 59 items:
[image]

kisvegabor, 2023-11-10:
You should enable `LV_USE_LARGE_COORD` in `lv_conf.h`

rlagace-cld, 2023-11-10:
Ok, using LV_USE_LARGE_COORD correct the problem.
Thanks

closed 2023-11-10 by rlagace-cld