lvgl/lvgl #4803: Use lv_draw_buf_t where possible

XuNeo, 2023-11-14:
### Introduce the problem

Since we support stride now, all image/draw `buf` now comes with stride parameter.
But currently the `buf` pointer is provided separately with stride, thus makes it error prone.

Currently `stride` usage is not clear enough.
-  calculated based on `lv_draw_buf_width_to_stride`
-  calculated from area width,
    > fill_dsc.mask_stride = blend_dsc->mask_stride == 0  ? lv_area_get_width(blend_dsc->mask_area) : blend_dsc->mask_stride;

- use parameters provided together with buf(like `_lv_draw_sw_blend_fill_dsc_t`,  `lv_draw_sw_blend_dsc_t`).

### Many struct already has stride
 
For examples, these struct already has stride for both `src`, `dest`, `mask`. So it makes sense to replace them with unified `lv_draw_buf_t`
<details>
  <summary>Code already has stride together with `buf`</summary>

```c
typedef struct {
    const lv_area_t * blend_area;   /**< The area with absolute coordinates to draw on `layer->buf`
                                     *   will be clipped to `layer->clip_area` */
    const void * src_buf;     /**< Pointer to an image to blend. If set `fill_color` is ignored */
    uint32_t src_stride;
    lv_color_format_t src_color_format;
    const lv_area_t * src_area;
    lv_opa_t opa;                   /**< The overall opacity*/
    lv_color_t color;               /**< Fill color*/
    const lv_opa_t * mask_buf;      /**< NULL if ignored, or an alpha mask to apply on `blend_area`*/
    lv_draw_sw_mask_res_t mask_res;    /**< The result of the previous mask operation */
    const lv_area_t * mask_area;    /**< The area of `mask_buf` with absolute coordinates*/
    int32_t mask_stride;
    lv_blend_mode_t blend_mode;     /**< E.g. LV_BLEND_MODE_ADDITIVE*/
} lv_draw_sw_blend_dsc_t;

struct _lv_draw_unit_t;

typedef struct {
    void * dest_buf;
    int32_t dest_w;
    int32_t dest_h;
    int32_t dest_stride;
    const lv_opa_t * mask_buf;
    int32_t mask_stride;
    lv_color_t color;
    lv_opa_t opa;
} _lv_draw_sw_blend_fill_dsc_t;

typedef struct {
    void * dest_buf;
    int32_t dest_w;
    int32_t dest_h;
    int32_t dest_stride;
    const lv_opa_t * mask_buf;
    int32_t mask_stride;
    const void * src_buf;
    int32_t src_stride;
    lv_color_format_t src_color_format;
    lv_opa_t opa;
    lv_blend_mode_t blend_mode;
} _lv_draw_sw_blend_image_dsc_t;
```

</details>

### Against using `lv_draw_buf_t`

There are cases not all parameters in `lv_draw_buf_t` is used like the `w`, `h`. 
It may increase memory usage, but should be reasonable.

We may also need to add helpers around `lv_draw_buf_t`.
- Create draw buf from various known parameters.
   - lv_draw_buf_create(w, h, stride, align, cf)
   - lv_draw_buf_create_from_size(size, cf).  Now use global setting for `align`.
   - Others

### Suggested solution

- The stride parameter should only be calculated and assigned when `buf` creates, and we should not modify it elsewhere.
- Limit usage of `lv_draw_buf_width_to_stride`.

```c
typedef struct {
    lv_color_format_t cf;
    uint16_t flag;
    uint32_t w;
    uint32_t h;
    uint32_t stride;
} lv_image_header_t;

typedef struct {
    lv_image_header_t header;
    uint32_t flag; /* May need to mark it's a source(image) or a dest. */
    void * buf;
} lv_draw_buf_t;

```

XuNeo, 2023-11-14:
Now `lv_dwaw_sw_blend_t` could look like this.
```c
typedef struct {
    lv_draw_buf_t * src;
    lv_draw_buf_t * mask;
    const lv_area_t * src_area;
    const lv_area_t * mask_area;
    const lv_area_t * blend_area;
    lv_opa_t opa;
    lv_color_t color;
    lv_draw_sw_mask_res_t mask_res;
    lv_blend_mode_t blend_mode;
} lv_draw_sw_blend_dsc_t;

typedef struct {
    lv_draw_buf_t * dest;
    lv_draw_buf_t * mask;
    lv_color_t color;
    lv_opa_t opa;
} _lv_draw_sw_blend_fill_dsc_t;

typedef struct {
    lv_draw_buf_t * dest;
    lv_draw_buf_t * mask;
    lv_draw_buf_t * src;
    lv_opa_t opa;
    lv_blend_mode_t blend_mode;
} _lv_draw_sw_blend_image_dsc_t;
```

kisvegabor, 2023-11-14:
I'm not against to adding it back. Probably it will require some extra code the create `lv_draw_buf_t`s here and there. We will see how it looks in the practice. 

Tomorrow I'm planning to make the CI pass if the ALIGN and STRIDE settings are not 1.  Now the SW render fails in some cases. 

I think it would be better wait for it to let the tests help find the issues.

nicusorcitu, 2023-11-17:
Oh, it was removed already twice... are we going to add it again?

XuNeo, 2023-11-17:
Yes, I'm planning a new PR. And all buf+stride should be replaced by draw_buf_t.

kisvegabor, 2023-11-18:
FYI, now the CI uses these settings  
```
/*Use a large value be sure any issues will cause crash*/
#define LV_DRAW_BUF_STRIDE_ALIGN                64

/*Use non power of 2 to avoid the case when `malloc` returns aligned pointer by default, and use a large value be sure any issues will cause crash*/
#define LV_DRAW_BUF_ALIGN                       852
```
and `lv_demo_render`  runs with RGB565, RGB888, XRGB8888 and ARGB8888 display color formats. See [here](https://github.com/lvgl/lvgl/tree/master/tests/ref_imgs/draw/render).

So I think now we have a safe base to do any modifications.

XuNeo, 2023-12-03:
Not stale

kisvegabor, 2023-12-18:
In progress

XuNeo, 2024-01-04:
In progress.

kisvegabor, 2024-01-14:
Is something still missing from here?

XuNeo, 2024-01-15:
With https://github.com/lvgl/lvgl/pull/5254 being merged, I see no more urgent change now.
The software renderer internal APIs can be simplified now, since it's internal API only, I'll close the issue as completed.

closed 2024-01-15 by XuNeo