lvgl/lvgl #4813: Unified VG-Lite implementation

kisvegabor, 2023-11-15:
### Introduce the problem

VG-Lite is an IP of Think Silicon and used by several chip vendors. So instead of having a VG-Lite driver for each vendor we could have a unified vendor independent one which can optimized by all parties.  

### Examples and cases

@nicusorcitu is working on the VG-Lite driver from NXP's side, however @XuNeo and @bjsylvia also started to work on it from Xiaomi's side.

### Suggested solution

So I wonder if you agree to have a joint vendor independent VG-Lite driver. 
What do you think?

Of course it applies to all similar GPUs when a well known IP is used.

XuNeo, 2023-11-15:
I agree with this solution. @FASTSHIFT is currently porting vglite to v9 on a smartwatch hardware.

VG-Lite provides its own SDK. We have experience of dealing with several versions of vglite IP and  it looks like backward compatible

Also the SDK looks like license friendly, see [the one found on github.](https://github.com/nxp-appcodehub/an-vglite_porting_imxrt/blob/main/middleware/vglite/VGLiteKernel/vg_lite_kernel.h)

nicusorcitu, 2023-11-15:
NXP have the vglite Ip (hw and driver) from Verisilicon not from Think Silicon.

Is it the same driver that Xiaomi has it?
________________________________
De la: Neo Xu ***@***.***>
Trimis: Wednesday, November 15, 2023 4:32:18 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] Unified VG-Lite implementation (Issue #4813)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

I agree with this solution. @FASTSHIFT<https://github.com/FASTSHIFT> is currently porting vglite to v9 on a smartwatch hardware.

VG-Lite provides its own SDK. We have experience of dealing with several versions of vglite IP and it looks like backward compatible

Also the SDK looks like license friendly, see the one found on github.<https://github.com/nxp-appcodehub/an-vglite_porting_imxrt/blob/main/middleware/vglite/VGLiteKernel/vg_lite_kernel.h>

XuNeo, 2023-11-15:
It's also Verisilicon.
Some ambiq chips use Think silicon GPU, NEMA.

nicusorcitu, 2023-11-15:
Now i am wondering if the driver provided is the same.
________________________________
De la: Neo Xu ***@***.***>
Trimis: Wednesday, November 15, 2023 5:26:59 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] Unified VG-Lite implementation (Issue #4813)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

It's also Verisilicon.
Some ambiq chips use Think silicon GPU, NEMA.

XuNeo, 2023-11-17:
Hi @nicusorcitu ,

We are currently using VGLite SDK version 2.0, but will upgrade to 3.0 soon.

nicusorcitu, 2023-11-17:
The current vglite driver is 3.x but I intend to do the changes to address the vglite 4.x which is different in many ways.
So I to not believe it will work having the same VGlite acceleration implementation.

kisvegabor, 2023-11-18:
It's conceptual decision at this point and not limited to VG-Lite.  What we do now will be the base future GPU implementations. 

I think having a vendor independent VG-Lite v3 and v4 is still better than having e.g. VG-Lite v3 for multiple vendors. Maybe what all this means is just having an `src/draw/vglite_v4` folder instead of `src/draw/nxp/vglite`.

nicusorcitu, 2023-11-20:
It's a decision I can not take it by myself. We should discuss it internally in NXP before any agreement.

The code is opensource, hence anybody can use it. The key is also about maintaining it. Having different vendors that work on the same code base can be helpful (collaboration work) but at the same time it can become also harder to maintain - due to particular fixes based on different driver versions.

It is not guaranteed that vglite 3.x that Xiaomi have it is the same as vglite 3.x that NXP have it. VSI can (and I think it really does) deliver different set of patches to different vendors.

xiaoxiang781216, 2023-11-20:
@kisvegabor the description "VG-Lite is an IP of Think Silicon" is wrong. VG-Lite is an IP of Verisilicon, The IP from Think Silicon is named `NEMA`.

xlbao123, 2023-12-05:
Unified VGLite implementation is challenging. I have observed various VGLite SDKs, and while their APIs are similar, there are significant differences in internal logic and features. For instance, some VGLite IPs require premultiplied-alpha source surfaces, while others do not.

XuNeo, 2023-12-05:
> For instance, some VGLite IPs require premultiplied-alpha source surfaces, while others do not.

We have faced the same problem and that's why image flag is added.

https://github.com/lvgl/lvgl/blob/7b6f360d46f39d7d5237e27b291be74e208f94cc/src/draw/lv_image_buf.h#L47-L51

And every draw unit can custom how image decoder works by `decoder args`.

https://github.com/lvgl/lvgl/blob/7b6f360d46f39d7d5237e27b291be74e208f94cc/src/draw/lv_image_decoder.h#L52-L66

Ideally, we should be able to distinguish limitations of each VGLite IP and set parameters accordingly. I haven't tried this idea but it does sound challenge if VGLite provides similar API with different behavior.

XuNeo, 2023-12-13:
Some update on VG-Lite porting.
@FASTSHIFT is bringing up VG-Lite on Xiaomi Watch 3 model and will have initial result by this week.

We (actually chip vendors) are also upgrading internal VG-Lite SDK to version 4.0. We have seen three vendor's SDK and they are all looks similar. The only difference is hardware capability for different version of IP. It has function to detect which feature is supported or not.

On LVGL v8, @FASTSHIFT already made VG-Lite work with `lvgl`'s freetype in vector mode, thus reaching really high refresh rate when screen is full with text. This capability has now been ported to v9 using legacy interface. Still needs some time 
 upgrading to `lvgl`'s new vector API.

@FASTSHIFT will prepare a draft PR to show his work. We can then discuss how to collaborate.

kisvegabor, 2023-12-13:
It sounds really promising. 
I think it would be huge if multiple people could work on the VG-Lite implementation to make it as good as possible. 

@nicusorcitu What do you think? Is there any objection from NXP side to collaborate this way?

kisvegabor, 2023-12-13:
Makes sense!

nicusorcitu, 2023-12-14:
When I say "final revision" I do not mean with all the features set. There is plenty work to do. I mean to have a stable VGLite revision in January.

kisvegabor, 2023-12-14:
I agree. It doesn't have to be full featured but stable enough for production.

kisvegabor, 2024-01-01:
If I understand correctly this is the plan:
1. @nicusorcitu adds a not full featured PXP+VG-Lite implementation before January 15, and we merge it as NXP's GPU implementation
2. After that we will see if it can be unified with the generic VG-Lite implementation and if it's allowed by NXP  

Everyone is ok with that?

nicusorcitu, 2024-01-08:
Hi @kisvegabor 
Sorry, I've been late with my response due to the new year vacation.
Yes, I will start the release process this week and hopefully will be complete till this Friday.
From that moment we can start the discussion about the vglite implementation and how/if it can be unified. We have to clarify few things:
1. Which vglite driver version is being used by Xiaomi? NXP have currently vglite driver 4.0.58.
2. Which driver OS is being used? NXP use FreeRTOS.
3. Which platforms does it cover? NXP has GC255 GC355 and GC555.

kisvegabor, 2024-01-08:
Thank you, Nicu! 

cc @XuNeo, @FASTSHIFT

nicusorcitu, 2024-01-11:
Just created the PR
https://github.com/lvgl/lvgl/pull/5288

There is still work to do on the vglite, but this is the version prior of the official v9 release. Other features and fixes can fallow.

kisvegabor, 2024-01-14:
#5388 is merged, thank you Nicu. How can we proceed with figuring out the possibility of unifying the to VG-Lite drivers? Could you ask around if NXP supports to go into this direction?  If so we can check the technical feasibility as well.

XuNeo, 2024-01-29:
Not stale.

nicusorcitu, 2024-02-07:
@cristian-stoica

kisvegabor, 2024-02-24:
@nicusorcitu @cristian-stoica 
Are there any updates on this topic from NXP both technically and company policy wise?

nicusorcitu, 2024-02-27:
Hi, we have no decision so far about the unification of the VGLite support.
We are focused now on releasing the NXP VGLite support on our new platforms, so this is a short term fact. We are too close to the release to unify the implementation right now.

We keep the discussion opened, there is not yet a conclusion on it.

kisvegabor, 2024-03-03:
I see, thank you for the update.

I close this issue for now.

closed 2024-03-03 by kisvegabor