lvgl/lvgl #4821: Logs are printed only for the thread that makes the custom printf registration

nicusorcitu, 2023-11-17:
Original discussion was here:
https://github.com/lvgl/lvgl/issues/4624#issuecomment-1784695594

Usually, the registration for the custom printf it is done in the main lvgl thread.
```
void print_cb(lv_log_level_t level, const char * buf);

#if LV_USE_LOG
void print_cb(lv_log_level_t level, const char * buf)
{
    LV_UNUSED(level);

    PRINTF("%s\r\n", buf);
}
#endif

static void AppTask(void *param)
{
#if LV_USE_LOG
    lv_log_register_print_cb(print_cb);
#endif
    .....
}
````

Having this, the Log will only work for that thread. If I am trying to log anything from GPU draw thread, it will not work.

Have you seen this behavior?
Do I have to register the print_cb from every thread?
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

kisvegabor, 2023-11-18:
@bjsylvia Can it be related to multi-instance feature. 

@nicusorcitu Please confirm that `LV_ENABLE_GLOBAL_CUSTOM` is disabled for you.

bjsylvia, 2023-11-19:
Yes, in a multi-instance scenario, `custom_print_cb` is bound to the current instance. This configuration should be globally unique. @kisvegabor do you agree？ If yes, I can rewrite and verify it.

nicusorcitu, 2023-11-20:
> Please confirm that LV_ENABLE_GLOBAL_CUSTOM is disabled for you.
>
Yes, it is disabled. Should I just enable it to fix the problem?

kisvegabor, 2023-11-20:
> Yes, it is disabled. Should I just enable it to fix the problem?

No, when it's disabled it should work. Can you check why the logs are not printed? `custom_log_print_cb` is really `NULL` in case of an other thread?

nicusorcitu, 2023-11-21:
Yes, `custom_log_print_cb` is NULL in case of an other threads.

kisvegabor, 2023-11-21:
I wonder how it's possible. 

Normally `lv_global_t` where the log print callback is stored is a just a static variable [here](https://github.com/lvgl/lvgl/blob/master/src/lv_init.c#L52-L54). Can you check with the debugger when it becomes zero? 

The other variables are changing too? E.g. [inited](https://github.com/lvgl/lvgl/blob/master/src/core/lv_global.h#L69)?

nicusorcitu, 2023-11-22:
It does not change, nobody overwrites it.

It get set once by lvgl_main thread:
```
Thread 1 hit Breakpoint 1, lv_log_register_print_cb (print_cb=0x3000b6a5 <print_cb>) at ../lvgl/lvgl/src/misc/lv_log.c:76
76          custom_print_cb = print_cb;
```
Then when it is being called from the thread via `LV_LOG` it is reported correctly:
```
(gdb) p *custom_print_cb
$3 = {void (lv_log_level_t, const char *)} 0x3000b6a5 <print_cb>
(gdb) p &custom_print_cb
$4 = (lv_log_print_g_cb_t *) 0x803c4984 <lv_global+460>
```
but if it gets called from another thread, like GPU draw, it is NULL:
```
(gdb) p *custom_print_cb
$6 = {void (lv_log_level_t, const char *)} 0x0 <lv_strncpy>
(gdb) p &custom_print_cb
$7 = (lv_log_print_g_cb_t *) 0x803c4984 <lv_global+460>
```

kisvegabor, 2023-11-23:
Is it NULL even at the very beginning of the thread? 

Can it happen that [Thread Local Storage](https://www.freertos.org/thread-local-storage-pointers.html) is enabled in FreeRTOS somehow?

Other than that I can't see  how can it happen that a global variable is `NULL` in one thread and set in an other.

nicusorcitu, 2023-11-23:
I do not understand either. I will try to invalidate the cache for that area and see if there is any effect.

But, is this happening to other OS also? Can you log from other threads?

nicusorcitu, 2023-11-23:
What I did:
1. Made `custom_print_cb` volatile.
2. Made the entire structure `lv_global` volatile.
3. Called 'CleanInvalidateDCache() after 'custom_print_cb` is set.
4. Moved the member `custom_print_cb` on top of the structure.

None of the above fixed the issue. Honestly I am clueless.

nicusorcitu, 2023-11-23:
I just figured out. While looking with readelf to the binary, I found this function:
[image]
So I put a breakpoint and find out that lv_global_init() gets called after the `custom_print_cb` gets set.

nicusorcitu, 2023-11-23:
I am closing this issue. Thanks!

closed 2023-11-23 by nicusorcitu