lvgl/lvgl #4822: Rotating a style only works twice for the entire code.

XanCraft21, 2023-11-18:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-11-18:
I think it's an out of memory issue. Can you increase `LV_MEM_SIZE` in `lv_conf.h`?

XanCraft21, 2023-11-18:
Unfortunately that did not fix the problem. Then i tried increasing the size of LV_DISP_ROT_MAX_BUF and it still doesn’t work. The code still runs normally but no rotation happens.

I also tried enabling debug mode to see if any errors occur but i get no errors.

kisvegabor, 2023-11-20:
Please enable all `LV_USE_ASSERT`s in `lv_conf.h` and the [logging](https://docs.lvgl.io/8.3/porting/log.html). Hopefully it will tell something useful.

XanCraft21, 2023-11-21:
Sorry it too me all day to get back with you.

I turned on those settings and saw nothing. But i got it to work, but it took a while. I hope I didn’t waste an issue.

i had a warning saying i wasn’t calling lv_tick_inc(), so i did that and it started working, but it kept crashing after so long. I tried slowly increasing the memory usage and it was still acting strange.

I eventually turned on LV_TICK_CUSTOM and LV_MEM_CUSTOM and those problems went away.

To me it seems like using the built-in memory allocation is very buggy. I think i have fixed my problem now. This library is pretty hard to set up, but I usually use libraries that need little changes.

Thank you for your help, and sorry if i wasted an issue. I will close this soon.

kisvegabor, 2023-11-21:
Happy to hear that you could make it work. And also sorry for that it took more time than anticipated. I would like to understand what caused the biggest issue to get started, so that I can improve it. So could you tell how we could make the get started experience better?

XanCraft21, 2023-11-21:
I’m not the greatest at making ideas for certain things. I just think the getting started and setup stuff is a bit complicated. I personally prefer more simple things, but that’s just me. Is there a possible way to simplify things?

kisvegabor, 2023-11-22:
All the feedback matters. If something is too complicated for you, probably it is for other as well.

However I still can't see which part was too complicated:
- integrating LVGL into your project?
- configuring LVGL?
- making it work with your display?
- how to create widgets, styles, events, etc?
- where to learn the basics?
- language barriers?
- anything else?

XanCraft21, 2023-11-23:
To me it’s mostly the configuring part that’s complex. I currently don’t have an active project to integrate this into, I mostly just mess around with the code just for fun. I’ll need to dive deeper with using the library to find out what else is complex or what i might run into.

As i said before i mostly usually use libraries that already have most of the things ready to go, and this is pretty new for me. It will take me some time to get used to using this library.

It could be because i was trying to go into things too fast, my biggest problem was the timers and memory usage. If it would be ok to add extra instructions for that or possibly some extra preconfigured default settings to make it easier?

kisvegabor, 2023-11-23:
In this case I will think about how to make LVGL even less configuration free or more obvious to know what you can/should adjust. 

Thank you for the input!

XanCraft21, 2023-11-23:
You’re welcome! I think i will close this now.

closed 2023-11-23 by XanCraft21