lvgl/lvgl #4830: How to customize _lv_text_is_a_word ?

fuxiujun, 2023-11-20:
### Introduce the problem

_lv_text_is_a_word is a function defined inside lvgl, but the Unicode range inside is incomplete. [CJK characters](https://en.wikipedia.org/wiki/CJK_characters)

I want to complete it in an overseas project, but I can only modify the lvgl source code. One modification usually cannot take into account all situations. Is there any way we can customize it?

### Examples and cases

Japanese katakana and hiragana are not included.

```
    /*Hiragana and Katakana*/
    if (letter >= 0x3040 && letter <= 0x30FF) {
        return true;
    }

```

### Suggested solution

Implement this function outside lvgl

kisvegabor, 2023-11-20:
I would like to fix it in LVGL instead adding a config option or extension point for it.  Could you send a Pull request?

fuxiujun, 2023-11-21:
> I would like to fix it in LVGL instead adding a config option or extension point for it. Could you send a Pull request?

We don't know how sentences are segmented in all languages, but we just discovered this bug during testing. If you later find that other languages also have this problem, you will need to continue to add it.

Custom extensions can solve problems more efficiently and avoid frequent changes to lvgl. What do you think?

kisvegabor, 2023-11-21:
> Custom extensions can solve problems more efficiently and avoid frequent changes to lvgl. What do you think?

My approach is that there isn't much room for customization here. These are standard things and LVGL should work accordingly. If something is missing we should solve it in LVGL to have good out of the box experience for everyone's happiness :slightly_smiling_face:  

Thank you for the PR!

PR #4840 (2023-11-21): chore(txt): add cjk Unicode range
### Description of the feature or fix

fix #4830

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/l [... cut]
closed 2023-11-21 by kisvegabor