lvgl/lvgl #4838: test barcode won't work with get_area_cb

XuNeo, 2023-11-21:
While debugging CI issue when `LV_BIN_DECODER_RAM_LOAD` is set to `0`.

The issue is when using get_area_cb, decoder_dsc->img_data becomes not NULL after `img_decode_and_draw`, the next time it's called from `img_draw_tiled`,  It now thinks whole image is available(line 256) because `decoder_dsc->img_data` is not NULL.

https://github.com/lvgl/lvgl/blob/423c288070ccfb90977529aae425b72419733c77/src/draw/sw/lv_draw_sw_img.c#L256-L285

I currently hacked the code to mark `img_data` as NULL, so to continue debug CI test fail issues.

### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

XuNeo, 2023-11-26:
In order to solve this, the decoded image should not be a simple pointer of `img_data` but a complete `draw_buf` with decoded area. 

https://github.com/lvgl/lvgl/pull/4833

`img_decode_and_draw` should check if the area to be draw has already been decoded or not.

```c
/**Describe an image decoding session. Stores data about the decoding*/
typedef struct _lv_image_decoder_dsc_t {
    /**The decoder which was able to open the image source*/
    lv_image_decoder_t * decoder;

    /*A copy of parameters of how this image is decoded*/
    lv_image_decoder_args_t * args;

    /**The image source. A file path like "S:my_img.png" or pointer to an `lv_image_dsc_t` variable*/
    const void * src;

    /** Pointer to a buffer where the image's data (pixels) are stored in a decoded, plain format.
     *  MUST be set in `open` function*/
    const uint8_t * img_data;
   
    ...
} lv_image_decoder_dsc_t;
```

XuNeo, 2023-11-26:
A similar issue is for indexed image, it could decode to ARGB8888 image, which makes it inconsistent with header->cf. Adding `lv_draw_buf_t decoded` could also solve it. Decoded format could differ to original format.

kisvegabor, 2023-11-26:
> I currently hacked the code to mark img_data as NULL, so to continue debug CI test fail issues.

I don't think it's hacky, but rather a good clean up.

>  A similar issue is for indexed image, it could decode to ARGB8888 image, which makes it inconsistent with header->cf. Adding lv_draw_buf_t decoded could also solve it. Decoded format could differ to original format.

The header here should store data for the decoded image. 
https://github.com/lvgl/lvgl/blob/950cebf9710a8c845098852e89944e8737fec8bb/src/draw/lv_image_decoder.h#L136-L137

However considering it a `lv_draw_buf_t decoded_image` is definitely more elegant. Let's go for it!

XuNeo, 2023-11-27:
> The header here should store data for the decoded image.

If so, then `get_area_cb` will get lost because it should deal with indexed image for example, not ARGB8888.

kisvegabor, 2023-11-27:
It could use `const void * src;` to read the original color format. However I do agree to add `lv_draw_buf_t decoded_image` it's much less hacky.

XuNeo, 2023-12-20:
Will be fixed in https://github.com/lvgl/lvgl/pull/4948, close it for now.

PR #4948 (2024-01-18): fix(draw): use image area to check if decoded image usable 
### Description of the feature or fix
Fix #4838
When using get_area_cb, need to detect if the decoded image is ready to be used again(normally no).

> I thought `_lv_area_is_in` should be used to check decoded image fully covers the area to draw, but test show area to draw may larger than image. 
> 
> Thus use `_lv_area_intersect` instead.
> 

Now I have firstly intersect areas with full image, and then check if area to draw is inside decoded image area.

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](htt [... cut]
closed 2024-01-18 by kisvegabor