lvgl/lvgl #4851: After the size of the lable is changed, the position of other controls relative to the position of the lable cannot be changed.

sam131208, 2023-11-23:
### Describe the bug
   In general, when displaying a digital clock, AM/PM is displayed in the upper left or upper right of the time, and is immediately attached to the position of the time.
   There are also labels next to AM/PM. I also hope that when AM/PM changes position, it will change position along with it.
   example:
      lv_obj_set_align( digit_clock, LV_ALIGN_CENTER );
      lv_obj_align_to( time_ampm, digit_clock, LV_ALIGN_OUT_LEFT_TOP, 0, 0);
      lv_obj_align_to( bluetooth, time_ampm, LV_ALIGN_OUT_LEFT_TOP, 0, 0);

   I expect that when the time changes from 12:59 to 1:00, AM/PM, Bluetooth icon will follow and move closer to the right.
   I've tried lv_obj_mark_layout_as_dirty(), lv_obj_update_layout(), lv_obj_invalidate() and none of them worked.

### To Reproduce
```c
	lv_obj_t *digit_clock = lv_label_create(lv_scr_act());
	lv_obj_set_width( digit_clock, LV_SIZE_CONTENT);
	lv_obj_set_height( digit_clock, LV_SIZE_CONTENT);
	lv_obj_set_style_text_color(digit_clock, lv_color_hex(0), LV_PART_MAIN | LV_STATE_DEFAULT );
	lv_obj_set_style_text_opa(digit_clock, 255, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_obj_set_style_text_font(digit_clock, &lv_font_montserrat_48, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_label_set_text(digit_clock, "12:59");
	lv_obj_set_align( digit_clock, LV_ALIGN_CENTER );

	lv_obj_t *time_ampm = lv_label_create(lv_scr_act());
	lv_obj_set_width( time_ampm, LV_SIZE_CONTENT);
	lv_obj_set_height( time_ampm, LV_SIZE_CONTENT);
	lv_obj_set_style_text_color(time_ampm, lv_color_hex(0), LV_PART_MAIN | LV_STATE_DEFAULT );
	lv_obj_set_style_text_opa(time_ampm, 255, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_obj_set_style_text_font(time_ampm, &lv_font_montserrat_20, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_label_set_text(time_ampm, "PM");
	lv_obj_align_to( time_ampm, digit_clock, LV_ALIGN_OUT_LEFT_TOP, 0, 0);

	lv_obj_t *bluetooth = lv_label_create(lv_scr_act());
	lv_obj_set_width( bluetooth, LV_SIZE_CONTENT);
	lv_obj_set_height( bluetooth, LV_SIZE_CONTENT);
	lv_obj_set_style_text_color(bluetooth, lv_color_hex(0), LV_PART_MAIN | LV_STATE_DEFAULT );
	lv_obj_set_style_text_opa(bluetooth, 255, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_obj_set_style_text_font(bluetooth, &lv_font_montserrat_20, LV_PART_MAIN| LV_STATE_DEFAULT);
	lv_obj_add_flag(bluetooth, LV_OBJ_FLAG_HIDDEN);
	lv_label_set_text(bluetooth, LV_SYMBOL_BLUETOOTH);
	lv_obj_align_to( bluetooth, time_ampm, LV_ALIGN_OUT_LEFT_TOP, 0, 0);
```

### Expected behavior
  when the time changes from 12:59 PM to 1:00, PM, Bluetooth icon will follow and move closer to the right.

kisvegabor, 2023-11-23:
Automatically updating widgets with `lv_obj_align_to` is rally something that we need to solve soon.

Now you can do it with a flex layout like this:

```c
  lv_obj_t * cont = lv_obj_create(lv_scr_act());
  lv_obj_set_size(cont, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
  lv_obj_set_align(cont, LV_ALIGN_CENTER );
  lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_ROW);
  lv_obj_set_style_pad_column(cont, 5, 0); //Space between widgets

  lv_obj_t *digit_clock = lv_label_create(cont);
  lv_obj_set_style_text_color(digit_clock, lv_color_hex(0), LV_PART_MAIN | LV_STATE_DEFAULT );
  lv_obj_set_style_text_font(digit_clock, &lv_font_montserrat_48, LV_PART_MAIN| LV_STATE_DEFAULT);
  lv_label_set_text(digit_clock, "12:59");

  lv_obj_t *time_ampm = lv_label_create(cont);
  lv_obj_set_style_text_opa(time_ampm, 255, LV_PART_MAIN| LV_STATE_DEFAULT);
  lv_obj_set_style_text_font(time_ampm, &lv_font_montserrat_20, LV_PART_MAIN| LV_STATE_DEFAULT);
  lv_label_set_text(time_ampm, "PM");
  lv_obj_set_style_translate_y(time_ampm, 5, 0);    //Move down
```

sam131208, 2023-11-23:
Thank you for your help.

> Automatically updating widgets with `lv_obj_align_to` is rally something that we need to solve soon.

  Hopefully see this change soon. In more complex layouts, lv_obj_set_flex_flow is not the best choice. For example, when it is necessary to change the position dynamically.

kisvegabor, 2023-11-24:
After releasing v9 (planned for mid January) we a can work on these kind of things too.

Until then, checking out the [grid layout](https://docs.lvgl.io/8.3/layouts/grid.html) can be useful too.

sam131208, 2023-11-27:
Thank you and wait quietly for V9.

closed 2023-11-27 by sam131208