lvgl/lvgl #4855: Feature Request: Curved text support in LVGL.

XuNeo, 2023-11-23:
### Introduce the problem

Curved texts are a common design element in modern user interfaces, providing a more dynamic and engaging appearance. Introducing curved label support in LVGL would align the library with current design trends and make it more versatile for a wide range of applications.

[image]

### Examples and cases

### Suggested solution

kisvegabor, 2023-11-24:
Oh yes, it would be really great! I'm thinking about it for a long time and I have some ideas. But let's release v9 first, and get back to these goodies after that. :slightly_smiling_face:

XuNeo, 2023-12-10:
Not stale.

kisvegabor, 2023-12-26:
It could be a good candidate to try out. https://github.com/lvgl/lvgl/issues/5098

djixon, 2023-12-28:
Nice feature. I will be available after 17th January to help (if you need help) about this. Just as a reminder, this feature can be solved with two different approaches:

1. As a general BEND ..... where any rectangular canvas can be bent using some curve ..... so it can be an image too, slider or any rectangular area like single line text area. (if canvas is of w width and h height, then straight line on it, from (0, h/2) up to (w-1, h/2) is bent using any curve simply by counting ratio of length of that line and length of curve. Then each pixel of that line from original canvas is drawn at each pixel of the curve (taking into account that stretch factor). At each point on the curve, 2D tangent is calculated, which then allows calculation of 2D normal as a perpendicular 2D vector on that tangent. Content of canvas above that central line is then transformed as a displacement from the curve using positive normal taking into account proper interpolation for possible stretching. Same should be done with the content of the canvas below that central line, using inverted 2D normal vectors. There are fast tricks, for required pixel interpolations for both squeezing and stretching by involving an abstract access to canvas FLOATING coordinates. For example lv_color_t GetPixelColor(float x,y). Canvas IS WITH integer pixel coordinates but this function take 4 pixels from following coordinates: P1(trunc(x), trunc(y))  P2(trunc(x)+1, trunc(y)) P3(trunc(x),trunc(y)+1) P4(trunc(x)+1, trunc(y)+1).  Those 4 colors represent base for calculating resulting color of an imaginary pixel with floating point coordinates. Interpolation is done THIS way: hor_blend = frac(x); ver_blend = frac(y); Do you see? Fractional part from coordinates REPRESENTS blending factor(s). So P1 and P2 are mixed using hor_blend and giving color C1 then same is done with P3 and P4 also using hor_blend resulting in color C2, then C1 and C2 are mixed using ver_blend giving the final resulting color of an imaginary pixel at float coordinates lets say 17.351 and 83.192 Those fractional parts are normalized which means they are in range (0.0f - 0.999999f) and if you need them in range (0- 255) they have to be multiplied by 256. Etc. As you can see, BENDING transformation can be achieved by allowing "access" to canvas using floating coordinates because math behind bend requires such thing. Also you can allow virtual writing to canvas using FLOAT coordinates. Its simple, you find 4 pixels which will be affected by such float coordinates, then make proper RGBA distribution of passed color to all those 4 pixels using the same technique. Fractional parts of x and y determine weights of color distribution)
2.  Strictly as TEXT (font) deformation but not per pixel bases but rather as a single character transformation. So optional parameter could be: should TEXT be aligned to TOP of curve, BOTTOM of curve or CENTERED (verticaly) to curve. Also optional parameters could be TEXT mirrored vertically and/or TEXT mirrored horizontally. Math is simple, take the width of each character, "transfer" that width on the curve by measuring the same length on it. Then take a center point of that width on the curve, calculate 2D normal of that point on the curve, and that normal vector, with a zero angle vector, forms an angle of rotation required for that single character drawn on temporary canvas to be rotated and placed on the curve. Sure, proper displacement (depending on proposed align_to_curve enums) also has to be taken into account. 

Maybe both features, general BEND and TEXT bend could be implemented...

kisvegabor, 2023-12-30:
I don't see many real life use cases for a generic bend feature. I think it's not common on UIs. So I vote for text only. 

I'm not sure if we need to stretch/bend the character itself, but a simple rotation might do it. With this all we need is a curve, and a function the get the point and angle for any point. E.g.
```c
/*Get the points position and angle on `distance` length */
lv_curve_get_info(curve, distance, &p, &angle);

/*Shift the letter and it's pivot to top/bottom/center as needed*/
draw_letter(..., &p, angle, INNER/OUTER/CENTER);
```

djixon, 2023-12-31:
Well, it depends on the point of view. The full BEND feature is  applicable to lets say BORDER of components. Suppose some image (texture) of lets say 30 pixels width and 5 pixels height which is prepared such way that it can be perfectly tiled horizontally. That means you can repeat it as many times as it is required to form an image of N width and 5 pixels height. By implementing ceneral BEND using curve, you can replace your ROUNDED border on style (which is const color so far) by that image and have it perfectly bent on those rounded corner where radius is still changeable.  So such image can replace the border of any length and of any radius.... all you need is to calculate final length (circumference) of the border, make large enough length of those tiles, apply bend (with rescaling "on the fly") and have perfect border of that image ....

Bending TEXT (using that general BEND method) can also have applicability in some specific cases.... However, I agree that another, lets call it, text FOLLOW curves method should exists which is applicable only to the TEXT, where each character is treated as a RIGID sprite and only rotation and position of such a sprite has to be calculated .... it is much faster, because sprite is rigid and doesn't have deformation but only transformation is required.

XuNeo, 2024-01-15:
@kisvegabor Could we pin the feature requests?

kisvegabor, 2024-01-15:
Pinned

kisvegabor, 2024-09-05:
@XuNeo Do you plan to work on it in the next 3-6 month?

kisvegabor, 2024-09-11:
I also find it a great feature, but to keep the list of the issues cleaner let's close it for now and get beck to this e.g. in v9.4. 

For v9.3 I'm planning to add blurring which is an equally great feature.

closed 2024-09-11 by kisvegabor