lvgl/lvgl #4856: Invisible unicode characters will be displayed

fuxiujun, 2023-11-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Some characters in the unicode code are invisible, They will have some special functions. [example](https://invisible-characters.com)
On mainstream operating systems they will be parsed correctly and not displayed. But LVGL will render them. This will cause strings copied from other operating systems to not be displayed correctly.
### To Reproduce

```c
  lv_obj_t* label = lv_label_create(lv_screen_active());
  lv_obj_set_style_text_font(label, &lv_font_montserrat_48, 0);
  lv_label_set_text(label, "\u200E\u202AHELLO WORLD!\u202C\u200E");
  lv_obj_center(label);
```
### Expected behavior

[image]
VScode can preview them correctly

### Screenshots or video

[image]

XuNeo, 2023-11-24:
A method I can see is add `is_unicode_invisible` logic inside `_lv_text_encoded_next`, since lvgl won't care about invisible code.

```c
/**
 * Decode an VISIBLE UTF-8 character from a string.
 * @param txt pointer to '\0' terminated string
 * @param i start byte index in 'txt' where to start.
 *          After call it will point to the next UTF-8 char in 'txt'.
 *          NULL to use txt[0] as index
 * @return the decoded Unicode character or 0 on invalid UTF-8 code
 */
static uint32_t lv_text_utf8_next(const char * txt, uint32_t * i)
{
    uint32_t result = 0;
    /*Dummy 'i' pointer is required*/
    uint32_t i_tmp = 0;
    if(i == NULL) i = &i_tmp;

    do {
        /**
         * Unicode to UTF-8
         * 00000000 00000000 00000000 0xxxxxxx -> 0xxxxxxx
         * 00000000 00000000 00000yyy yyxxxxxx -> 110yyyyy 10xxxxxx
         * 00000000 00000000 zzzzyyyy yyxxxxxx -> 1110zzzz 10yyyyyy 10xxxxxx
         * 00000000 000wwwzz zzzzyyyy yyxxxxxx -> 11110www 10zzzzzz 10yyyyyy 10xxxxxx
         */
        ...
     } while (is_unicode_invisible(result));

    return result;
}
```

kisvegabor, 2023-11-24:
Makes sense, but I think it should be added [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_label.c#L371-L374).

closed 2023-11-27 by kisvegabor