lvgl/lvgl #4866: Run the image convert in the browser

W-Mai, 2023-11-25:
### Introduce the problem

Reference to this discussion https://github.com/lvgl/lvgl/pull/4414#issuecomment-1764819295

I wrote a simple demo to fit the requirements for this comment.

The repo is here

https://github.com/W-Mai/RunPyScriptInBrowserWithWasmLibCompiledByRust

@kisvegabor 

### Examples and cases

### Suggested solution

kisvegabor, 2023-11-26:
Thank you for dealing with it @W-Mai!

I've tried it based on the README and it worked well locally. 
I copied the 3 files from the `web` folder to lvgl.io's webserver but it doesn't work that well there: https://lvgl.io/tools/new_img_conv/

[image]

W-Mai, 2023-11-27:
I notice that the console shows an error message about MIME type mismatching
[image]

There is a `content-type` in the normal http request and response

[image]

But lvgl.io server seems not to provide the `content-type` property

[image]

I think this may be the real cause of this problem

kisvegabor, 2023-11-27:
I have no idea where to start, so it'd be great if you could provide an example for these 3 points.

W-Mai, 2023-11-28:
> I have no idea where to start, so it'd be great if you could provide an example for these 3 points.

Ok, I will do it later 😁

W-Mai, 2023-12-08:
[image]

Well done!

Now this demo supports dealing multiple images and also can download from a visible link.
And also it has the ability to get information form `select` component. 😁

kisvegabor, 2023-12-11:
Looks good! I'll try to integrate it soon!

kisvegabor, 2023-12-15:
Hi @W-Mai,

I started to integrate it to the online converter, however something still seems missing. I cannot see where you call the actual Image converter to which I can pass parameters like color format.

W-Mai, 2023-12-15:
Here is the calling entry
https://github.com/W-Mai/RunPyScriptInBrowserWithWasmLibCompiledByRust/blob/2b7e6351f4069b2f51173ece779579ae971b6c14/web/main.py#L89

And you will find the same function definition here https://github.com/W-Mai/RunPyScriptInBrowserWithWasmLibCompiledByRust/blob/2b7e6351f4069b2f51173ece779579ae971b6c14/src/main.rs#L10-L17

W-Mai, 2023-12-15:
oh, you mean where is the real image converter already in the lvgl repo.

It's a pity that this demo is an example of a simple version that I rewrote. It's just an example.

kisvegabor, 2023-12-15:
> oh, you mean where is the real image converter already in the lvgl repo.

Yes, I mean that :slightly_smiling_face: 
Where how can I add it?

If the whole thing is too complicated we can just slightly update the current image converter to use the new `LV_COLOR_FORMAT`  definitions and accept that it has some limitations. (E.g. not supporting stride). For a more complex converter people can use the the Python script.

W-Mai, 2023-12-15:
I don't think it should be very complicated, so please give me some time if you need to support the full script (I'm working on cache right now).

kisvegabor, 2023-12-15:
Okay, thank you! I would like to have a "call to test" for v9 on Monday, so some kind of converter should work by then. At worst case I can recommend to use the script directly, so there is no emergency. :slightly_smiling_face:

kisvegabor, 2024-01-01:
Not stale

kisvegabor, 2024-01-08:
@W-Mai As discussed on our meeting I attach the HTML file for the front-end of the current converter.
[index.html.zip](https://github.com/lvgl/lvgl/files/13860420/index.html.zip)

W-Mai, 2024-01-23:
Not Stale

kisvegabor, 2024-02-22:
I close it for now. The current simplified online converter should be ok for  while.

closed 2024-02-22 by kisvegabor