lvgl/lvgl #4868: `lv_draw_buf_malloc` does not return aligned draw buf

XuNeo, 2023-11-26:
### Introduce the problem

Despite the name is to malloc a draw buf, it actually could return non-aligned buffer and need `lv_draw_buf_align` to safely return the actual aligned address.

```c
/**
 * Allocate a buffer with the given size. It might allocate slightly larger buffer to fulfill the alignment requirements.
 * @param size          the size to allocate in bytes
 * @param color_format  the color format of the buffer to allocate
 * @return              the allocated buffer.
 * @note The returned value can be saved in draw_buf->buf
 * @note lv_draw_buf_align can be sued the align the returned pointer
 */
void * lv_draw_buf_malloc(size_t size_bytes, lv_color_format_t color_format);
```

### Examples and cases

I believe some of us haven't noticed this usage, and already misused it. It will work most of the time on simulator but will create bugs when running on hardware.

https://github.com/lvgl/lvgl/blob/de9c956c360d070a631100f4c41c1a1322ac4b34/src/libs/lodepng/lodepng.c#L5309-L5312

https://github.com/lvgl/lvgl/blob/de9c956c360d070a631100f4c41c1a1322ac4b34/src/libs/libjpeg_turbo/lv_libjpeg_turbo.c#L378-L379

https://github.com/lvgl/lvgl/blob/de9c956c360d070a631100f4c41c1a1322ac4b34/src/libs/tiny_ttf/lv_tiny_ttf.c#L161-L166

### Suggested solution

- Use `lv_draw_buf_t` instead of a raw pointer directly. 
https://github.com/lvgl/lvgl/pull/4833
- Add asserts where possible

kisvegabor, 2023-11-26:
Having `lv_draw_buf_t` really could be a solution, but note if have only the aligned pointer we cannot `free` it, as for free we need to original pointer.  `lv_draw_buf_t` we can store both, of course.

djixon, 2023-11-28:
@XuNeo
You can always perform safe manual alignment if malloc on the system is not capable to do that. For example suppose you need 4 byte aligned address from malloc and you need buff of 1KB so instead to write:
uint8_t * buff = malloc(1024 * sizeof(uint8_t));

you simply take into account all possible scenarios. In this example it would be 3 cases (forth case would be aligned and doesn't require adjustment. So you write:

uint8_t * buff = malloc((1024 +3) * sizeof(uint8_t));

now you actually got buffer of 1027 bytes but you still do not know is the beginning of the buffer 4 byte aligned. But here is the trick:

uint8_t * buff = malloc((1024 +3) * sizeof(uint8_t));
uint8_t tmp_offset =  (*((uint32_t*) (&buff)) & 0x03);  // take the last two bits of address returned from malloc and save
*((uint32_t *) (&buff)) &= 0xFFFFFFFC;  // clear the lowest two bits, now address is 4 bytes aligned 

You do not need to worrie about length. You allocated 3 bytes more so what ever the case from that, now aligned address, you will always have guarantied 1024 bytes length. Sure, that temporal tmp_offset must be kept all the time till the moment you want to release the buffer. Then you write

*((uint32_t *)(&buff)) |= tmp_offs;
 free(buff);

You simply re-adjusted address to original value returned by malloc and then its safe to free it.

XuNeo, 2023-11-28:
@djixon Thank you for your explanation. That's exactly current solution. You can check it in lv_draw_buf.c

XuNeo, 2023-12-18:
`lv_draw_buf_t` struct is already added. More modules are upgrading to use it directly instead of `void * buf`.

Note, aligned_alloc is not C99 available. Thus the only solution is wraping the unaligned pointer to a struct and avoid using `void * buf` directly.

XuNeo, 2024-01-09:
Remove public `lv_draw_buf_malloc` API is still ongoing. It needs `layer` and `label` modules migrate to `lv_draw_buf_t` firstly.

XuNeo, 2024-01-15:
As of now https://github.com/lvgl/lvgl/pull/5254 is merged. Now the `lv_draw_buf_malloc` can finally be a static function.

PR #5318 (2024-01-15): feat(draw_buf) make lv_draw_buf_malloc a static function

Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

Fix #4868

From now on, should use lv_draw_buf_create instead because lv_draw_buf_malloc does not return aligned pointer

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/m [... cut]
closed 2024-01-15 by kisvegabor