lvgl/lvgl #4869: `lv_image_dsc_t->data` could not meet buf align requirement

XuNeo, 2023-11-26:
### Introduce the problem

Current `lv_image_dsc_t` definition includes a simple data pointer for all kinds of images.
For `Indexed2` image, `palette` takes `4 x sizeof(lv_color32_t) = 16Bytes`.
It makes the image data following palette cannot meet requirement of `64byte` alignment.

It also makes decoding 8bit PNG a little more complicated. We have to alloc a new contiguous memory for pallete + img_data.

For RGB565A8 image, we cannot describe the stride for alpha map inside.

```c
typedef struct {
    lv_image_header_t header; /**< A header describing the basics of the image*/
    uint32_t data_size;     /**< Size of the image in bytes*/
    const uint8_t * data;   /**< Pointer to the data of the image*/
} lv_image_dsc_t;
```

### Examples and cases

### Suggested solution

Depending on the color format of the image, using different struct to describe data.
For example, an indexed image should use `indexed` field.

```c

typedef struct {
    lv_image_header_t header;           /*A header describing the basics of the image*/
    uint32_t data_size;                 /*Size of the data in bytes*/
    union {
        const uint8_t * data;           /*Uncompressed normal image data*/

        struct {
            uint32_t compress_method: 4;
            uint32_t flags: 4;
            uint32_t uncompress_size: 24;
            const uint8_t * compressed; /*Compressed raw data*/
        };

        struct {                        /*Special image of indexed format*/
            const uint8_t * palette;
            const uint8_t * data;
        } indexed;

        struct {                        /*Special image of RGB565A8 format*/
            const uint8_t * rgb;
            const uint8_t * alpha;      /*Alpha map of RGB565A8*/
            uint32_t alpha_stride;      /*Stride for the alpha map in bytes*/
        } rgb565a8;
    };
} lv_image_dsc_t;

```

kisvegabor, 2023-11-26:
I would approach it in a more generic way. What about
```c
typedef struct {
    lv_image_header_t header;           /*A header describing the basics of the image*/
    uint32_t data_size;                 /*Size of the data in bytes*/
    const void * buf;
    const void * buf_sup;           /*Optional supplementary buffer*/
}
```

Regarding `compressed`
```c
            uint32_t compress_method: 4;   /*Can we have it in lv_image_header_t*/
            uint32_t flags: 4;                            /*Can we have it in lv_image_header_t*/
            uint32_t uncompress_size: 24;   /*Why do we need to know this? It depends on stride on align too, so it might not be known*/
```

djixon, 2023-11-27:
There are 10 million of reasons of why alpha should not be separated from  rgb data: 
1. Possible miss-alignments of those two buffers
2.  in the most common R8G8B8 format, Alpha is extremely "handy" because it provides proper 4 byte alignment of pixels data
3.  Routine(s) which transfers buffer data, in case of separated buffers, now have to take care  of two "streams" and their possible miss-alignment. Suppose you have large image, but your routine only needs a small part of it to draw in some rectangular area, so your calculation for each row  "hits" miss-aligned in two of 3 cases (read it as, it is extremely slower because most of hardware has a huge drop in performance by accessing non-4 byte aligned data.)
4. If an image is stored on SDcard for example, and you only need small rectangular area of it, reading of two separated areas from different sectors on SD card by utilizing SEEK to provide proper location per pixel bases....
5. I can continue counting ..... but that is not the point. 

Simply, do not do that. It will make more trouble than benefits. Pixel data (all of its elements, doesn't matter how complicated its internal structure is, must stay stick together in that pixel chunk. Full stop)

One more suggestion, it is not necessary but can help in performance. A routines which perform anything with pixels buffer, should be AWARE of single pixel size. So it is not good enough to have pixels like void* buffer, or uint8_t * buffer passed to them! At hardware level, a lot of transfers are interrupted all the time, or are performed in chunks of DMA buffer sizes. Without knowing anything of "what's in that passed byte array", those low level routines can not guaranty that transfer is performed on integer count of pixels, because those routines are not aware of size of pixel. That, then requires special code for treatment those non-completed tails which introduce drop in performance at low level routines (for example, a display driver...) The goal is to make those low-level routines aware of "what is in a buffer" so they can guaranty that any transfer is performed with integer count of pixels (and not integer count of bytes). Structural transfers (pixel array is also one of them) MUST be handled to guaranty transfer on integer count of structure. Only that way, low-level routine can be written with maximal efficiency. Guru guys, made that 50 years ago or so, when they provided transfer routines (which can return non-completed what ever the reason for their interruption is) but they return integer quantity of amount of structures are transfer. For example in standard POSIX, beside pointer to data, you have to pass additional two parameters sizeof(struct) and count_of_structs you want to transfer. (sure internally, amount of bytes is calculated as multiply of those two, but that is not the point, the point is that, IF such a routine has to return non-completed, it will return with NUMBER which is integer count of amount of structures written/passed/transfer or what ever. Simply, its not possible it returns with last structure cut somewhere in the middle. 

There are a lot of benefits with passing two (instead  just one parameter like amount of bytes). For routines which have to perform mallocs, they can do that with guaranty that allocated memory block size will be exact multiple of structure size (because now they know that). Low level routines Internal DMA transfers (with linked buffers) and lot of other routines, can be written more efficiently by having that information.

XuNeo, 2023-11-27:
Hi @djixon 
I think you have misunderstand the issue here. We are dealing with `RGB565A8` which is wired because it's not a `pixel` format rather than buffer format. 
For other pixel formats(color format), the alpha comes with RGB map.

djixon, 2023-11-27:
Look, what ever the purpose of data stored in an image buffer is, fetching from/to buffer per pixel bases, should be performed in possibly one assembler instruction read/write from proper address (in memory) or sequence of consecutive bytes (if we are talking  about reads/writes from file on lets say disk/SD card/memory inside flash. If you have to separate those readings/writings to many different access (at lowest level) you are actually degraded performance of those routines with having separated buffers for rgb data and alpha data

XuNeo, 2023-11-27:
The only case RGB and alpha is separated is RGB565A8, because it's defined to be so. You may want to check ARGB8565.

kisvegabor, 2023-11-27:
> If you have to separate those readings/writings to many different accesses (at lowest level), you are actually degrading the performance of those routines with having separated buffers for rgb data and alpha data

It might be the case with SD cards, however reading from flash or RAM is faster because the RGB565  data on one place and A8 data on te other place is aligned, and each can be fetched with 1-1 instructions. Reading ARGB8565 is slower an there is fewer places for optimizations as reading 3 bytes is not optimal anyway.

ARGB8565 is not supported in v9 (it was in v8 but in v9 we have RGB565A8). We can consider adding ARGB8565 v9 but I would prefer more standard formats, like ARGB4444 or ARGB1555.

djixon, 2023-11-27:
Technically, it is true, but has sense only if the full image can be pulled into RAM.  Then, it would be performed by two separated loops one for rgb data and other for alpha. But the problem persists if you need partial access to image buffer, per pixel base. Such a "jumping" can target different sectors in SDcard/ flash/ disk .... not to mention "pollution" of cache data ....

djixon, 2023-11-27:
Also, as an instant suggestion for v9, based on really fast overview through full LVGL library, you should consider following concept. There should be an abstract class (similar to canvas) which persists in between all "paintable" classes and lv_obj class.

That would mean that such a base abstract class would provide an abstract functionalities like resize "buffer", paint "buffer" etc.
So all other classes like button, label, panel or any other "drawable" class is inherited from such a class. Then if you implement an abstract virtual methods to such a class like "paint", "resize" etc, those methods will be overriden in child classes (label, button panel etc). However, you could then provide public those methods for paint like OnPaint function, OnResize function etc which would be allow end developer to tweak original painting resizing or what ever in those published functions or to keep "original" paint, resize etc. methods by calling inherited method of ancestor class. Or by simple words, each class (label, button, panel etc would act also as an abstract canvas, allowing user to override default paint methods by its own or mixing its own by default. For an example:

function label_on_paint(lv_abstr*  )
{
 lv_canvas_drav(circle(lv_abstr);
lv_inherited_call;
}
 
would allow user to draw something to buffer of a label like he would do by accessing a canvas object and when it finish drawing he call inherited (default paint method which would perform drawing over his drawing). Or he also can switch order of operations, and first call inherited paint method to force default drawing in the buffer and then perform his own drawing over that. 
Such a concept eliminates needs to have two object on over another (lets say in many examples you have button andf label over it) ???? why???? You can simply have a button with overridden paint method where drawing a text on an abstract canvas pointing to buffer of that button will do the job much simpler

kisvegabor, 2023-11-27:
It already works like that. See this [example](https://docs.lvgl.io/master/widgets/chart.html#show-the-value-of-the-pressed-points)

djixon, 2023-11-27:
Nice to hear that ..... I am just a "new" to lvgl so most of introduce to it was through some examples provided for v7 .... last week I took v8.3 but I still didn't check all changes made or (its port to C) is not completed in complete because I didn't see any such ability.

kisvegabor, 2023-11-27:
> Anyway, net is flood by examples not utilizing it, but instead use two classes (button and label over it), that's why I thought its not supported

Having a separate label widget has some advantages:
1. The button's size can be set automatically to the label size (or an children size) by setting the button's width and/or height to `LV_SIZE_CONTENT`
2. The label can be targeted by styles
3. The label can be scrolled if it's too long by setting the `long_mode` property
4. The label has the versatile API to set and change text in various ways.

However, I understand what you say, and in most of the cases the text just needs to added to the center of the button without any magic. So I think displaying a label or image on the button should be a built-in button functionality in a future release.

djixon, 2023-11-27:
Sure, but those properties should be part of canvas class too ..... abstract canvas should be able to perform all the things a label can do. For it, drawing a text (with proper text_dsc_t) which contains all info about font, text wrapping etc should be enough to perform proper text drawing (same as label can do). Also, canvas should be capable of drawing a style on it etc ...... the only difference in between a canvas and a general "container" class is scrollable area (which should not be part of canvas object). Canvas should also be "scalable on demand", which means, in memory, it always occupies buffer determined by const width and height of the canvas, but method of drawing canvas to parent buffer uses stretching of canvas "on the fly" if parent container "asks for larger or smaller paintable area within itself. So yes, canvas can be made to be self-scaled (upper or down scale) if parent container "asks" for that. In memory, its buffer is not changed, only rescaling is performed on the fly (through another buffer or by proper pixel blittering function which can do proper aliasing and transparency keeping functionalities of such a rescale). Conceptually, its not too much different from DotPerInch functionality. It performs similar rescale of buffers for example in the case of small and large resolution monitors. Etc.

djixon, 2023-11-27:
..... ah I forgot, user should be able to get some very necessary info from canvas object .... text measurement ..... for example, once text_dsc_t info is populated (which can be general property of canvas object), function like getTextwith or get textHeight are required, so a canvas object can calculate those (with respect to text_dsc) and return values before it performs any painting.  That way user can adjust position of text he want to spit on canvas by knowing those two information got from canvas object. Canvas object ofcourse has to take info on possible style applied, borders, alignment of text in text_dsc  so it can return proper info of those two numbers. Also there could be one more optional function which would return the actual rectangle of "clean printable area" where all borders, paddings etc (if style exists on canvas object) are counted too. Etc

XuNeo, 2023-11-28:
> ```c
> uint32_t uncompress_size: 24;   /*Why do we need to know this? It depends on stride on align too, so it might not be known*/
> ```

There's situation uncompressed data size is slightly larger than needed. See https://github.com/lvgl/lvgl/pull/4870/commits/f3018b8dd1131fb72a4dd7983b97e2d70147cf66

kisvegabor, 2023-11-28:
@XuNeo One more note: nameless unions are C11 standard we are on C99. We can consider jumping to C11, but it requires serious consideration and research.

djixon, 2023-11-28:
> The only case RGB and alpha is separated is RGB565A8, because it's defined to be so.

@XuNeo Don't worry too much for that case, you can leave it literally RGB565A8 in single buffer. So it CAN be handled like standard prototype structure in single buffer. The trick to avoid non-aligned 3 byte data access is in this: You fetch literally chunks of 4 bytes into 3 32 bit registers (at assembler level), so your access to buffer is always aligned (you read 12 bytes into 3 32 bit registers with 4 byte alignment) then (depending on on target architecture, some can operate with 64 or even 128 bit registers) shuffling of bits among those 3 registers generates all 4 pixels ARGB data contained in those 3 32 registers) 
The only requirement is that width and height of image have to be even (and not odd). That guaranties the whole buffer size in bytes will be divisible by 12 so reading in chunks of 3x 4 bytes = 12 bytes into 3 32bit register will perfectly fit in buffer size

XuNeo, 2023-12-13:
Not stale.

This issue still exists. In case of indexed image. The image data start address is affected by palette length.

kisvegabor, 2023-12-13:
What is we move the palette to the end of the image?

djixon, 2023-12-13:
> What is we move the palette to the end of the image?

That's ok but also you can put proper amount (1, 2 or 3) of dummy bytes at palette end so image data starts at 4 byte aligned address

kisvegabor, 2023-12-14:
@djixon It's a little bit more complex because some GPUs can have different memory alignment requirements like 32 or 64 byte.

djixon, 2023-12-14:
@kisvegabor In that case you have to allocate 31 (63) bytes more per buffer (if compiler is not capable to generate malloc 32 or 64 bytes aligned) For 64 byte alignment  mask for pointer will be 0xFFFFFFC0  in case that the pointers are of 32 bits in size. For 64bit pointer size mask will be 0xFFFFFFFFFFFFFFC0
etc.

kisvegabor, 2023-12-15:
What is the advantage of this compared to moving the palette to end? 

With palette at the end we can tell the compiler to align the image itself to a given address and that's it. If the image is loaded from a file, we can just allocate an aligned memory aligned and read there.

djixon, 2023-12-15:
I am not sure we are talking about the same thing. If I understood properly, @XuNeo asked about image data stored in some indexed image format where palette (by format definition itself) is coded, for that format, before image data, and alignment of that image data beginning address, depends on the size of the palette itself. So you can not change definition of the format by swapping order of data and palette. 
On the other side all lv_xxxx_struct_t types contain pointers only (not data or whole palette inside) which means that initialization of those structures does dynamic allocation of buffers where those data is going to be load. So the only requirement is that those memory allocation routines provide properly aligned buffers for any of those data (what ever data is). Now, depending on the format itself, you can decide is it better to have those buffers interleaved (mixed data into single buffer for example in the case of RGB565A8 you can decide to mix 16 bits color + 8 bits alpha into single buffer (that means 3 bytes in sequence of each pixel in interleaved form or separated (chained form) but ALSO in a single buffer where each block inside the buffer begins at properly aligned address. But that is dependent on the implementation of routines which operates with that data. Similar way both palette and data can be load into single buffer (chained form) where buffer itself is properly aligned  and each block (in this case only two blocks) inside the buffer has also properly aligned beginning. 

In that sense, you can allocate a single buffer where where each block is "adjusted" to proper size by adding a required amount of dummy bytes at the end of that block, which will provide that every next block in the buffer begins at properly aligned address.

djixon, 2023-12-15:
Also, do not be "scared" of RGBA5658 (3 byte in size) format. There are tricks to use ALWAYS aligned access to that data (doesn't matter the pixel index!!!) I can describe it in short here. Lets call our 3 bytes of RGBA5658:  CH (represents high byte of color information in that 16 bits wide color), CL (represents low byte) and A represents 1 byte Alpha. So structure in the buffer is as:
CH,CL,A,CH    CL,A,CH,CL   A,CH,CL,A  **CH,CL,A,CH**  ....

or in the case if alpha is stored first

A,CH,CL,A    CH,CL,A,CH   CL,A,CH,CL  **A,CH,CL,A** ....  

I made (intentionally) those spaces to see how aligned to 4 byte access to such a buffer looks like. As you can see that every 3rd composition of 4 bytes is the same. So by modulo 3 (Pixel_Index % 3)  you have one of 3 possible cases of how "shuffling" of those bytes within that 32 bit access looks like. Total amount of bytes in buffer is 3*Amount_Of_Pixels however from the point of view of 32 bit access, there is total of (3 * Amount_Of_Pixels)/4 32 bit words. In the worst case, pixel data can be stored in not more than two of such 4 byte chunks. That is determined by (mod 3) value. So for an example, function which read pixel data (color + alpha) and uses STRICTLY aligned access to such a buffer can look like this:

lv_color_t Get_Pixel_RGBA8888_from_RGBA5658 (x, y){
 uint32_t Pixel_Index = y*width + x;
 uint32_t pix_byte_start = Pixel_index * sizeof(pixel);  // in this case size of pixel is 3 
 uint32_t* Addr_with_alligned_access = Aligned_BufferBeginingAddr + (pix_byte_start >> 2); 
 pixel_format_at_that_address = Pixel_Index % 3;
 
uint32_t value = *Addr_with_alligned_access;
   switch(pixel_format_at_that_address){
     0:   // pixel fully contained in the chunk  
           result =  convert  value from CH,CL,A,dummy to result;
          break; 
     1:  // pixel is spanned in two chunks
         value1 = *(Addr_with_alligned_access ++); // increment for 4 bytes
         result = convert from value dummy,dummy,dummy,CH and from value1 CL,A, dummy,dummy into result
         break; 
     2: // pixel is spanned too
         value1 = *(Addr_with_alligned_access ++); // increment for 4 bytes
         result = convert from value dummy,dummy,CH,CL and from value1 A,dummy, dummy,dummy into result
        break;
    }
return result;
}

   

 }  
}

etc.

That way only aligned access is possible to such a buffer.

kisvegabor, 2023-12-15:
> So you can not change definition of the format by swapping order of data and palette.

It's not standard format that we directly feed to an GPU but we set the palette for the GPU and then the data. So the actual order doesn't really matter. 

With `image data` followed by `palette` no dummy bytes are required at all, as there is no alignment requirement on the palette. So it doesn't matter is the image has 37 bytes, the palette is ok.

> Also, do not be "scared" of RGBA5658 (3 byte in size) format. 

Here the problem is that RGBA5658 is not standard therefore cannot be accelerated by GPUs. Probably we can figure out a not too bad SW implementation, but RGB565A8 seems superior to me.

djixon, 2023-12-15:
Ok. That was just an example of how 3 byte chunks can be accessed with always 4 byte aligned access .... you can see it also in RGB888 (without alpha) With the same technique you can make always aligned access to that 3 byte RGB888 buffer for example.

But anyway, I am thinking about some possible interesting thing. Suppose you can start LVGL in two different modes (switch in configuration). 
MODE1:
- Configuration like it is right now. You pick COLOR format and whole library and all functions operates on that color format. So all input images are converted into that format, all buffers are of that format, output to LCD driver is of that format .....

MODE2:
- You pick an internal operating mode of LVGL for example RGBA8888 (so all buffers, canvases widgets pretty everything) operates in chosen format. Any imported image is converted into that internal format and all internal calculation and intermediate buffers, blending routines etc  are in that format. However, output buffer, sent to display, can be of different format. That way you would be able to do complete internal work, in format independent of output display driver. (for now, resultant mixed buffer of chosen internal format would need to be converted to output display buffer by software, but once ESP32P4 is released, according to module they plan to add, such conversion will be possible "on fly" during DMA streaming)   

Correct me if I missed something and if such concept (internal format independent from output display driver) is already supported.

kisvegabor, 2023-12-16:
> Correct me if I missed something and if such concept (internal format independent from output display driver) is already supported.

Already implemented :blush: 

You can just use `lv_display_set_color_format(display, LV_COLOR_FORMAT_..)` and LVGL's SW render or GPUs will convert all the images during blending. See [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c).
You can also check out the updated [Porting guide](https://docs.lvgl.io/master/porting/disp.html).

XuNeo, 2024-01-15:
Not stale.

Ideally we could separate palette and image data. The compressed image format should be adjusted correspondingly.
The change is not started yet.

XuNeo, 2024-01-31:
Not stale. Got an idea to separate the palette and make corresponding adjustment to compressed bin image format. Will post once verified.

XuNeo, 2024-02-20:
Not stale.

closed 2024-05-27 by lvgl-bot