lvgl/lvgl #4874: Mess with SWAPPING and ENDIANNESS

djixon, 2023-11-26:
Guys, there are both 16 bit formats: RGB565 and BGR565. Swapping is introduced to support both of them. When swapping is OFF, format should  be RGB565. When swapping is ON, only R and B channels exchange positions so you have BGR565 format. That has nothing to do with endianness of byte order. Required endianness can be applied to any of those two formats. In nature, in RGB565 format, channels are spread from highest significant bit to lowest significant bit as order in name itself. So in RGB565 "left most bits" 15 to 11 are 5 bits of R channel while  in BGR565 format "leftmost" 15 to 11 bits are 5 bits from B channel. Green channel is always 6 bits in the middle. Once proper 16 bit value is calculated, any (required) endianness can be applied an has nothing to do with swapping channel  capability which exists to support bot RGB565 and BGR565 formats.

kisvegabor, 2023-11-27:
I agree that we need to support it in some form. Probably adding many branched for this during rendering would be slow, so vote for having a helper function in v9. Similar to [these](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.h#L92-L101).

djixon, 2023-11-27:
Sure, functions which iterate through pixel buffer should be separated for different formats. So its much better having two different functions lets say FunctionNameRGB565 and FunctionNameBGR565 to do their jobs. What-ever a function performs, it would be waste of CPU resources to perform branches per pixel format inside the function  itself

closed 2023-12-20 by lvgl-bot