lvgl/lvgl #4879: v9 `lv_obj_set_style_clip_corner` not support bigger obj?

tkernelcn, 2023-11-27:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

Thanks.

kisvegabor, 2023-11-27:
Probably it's an out of memory issue. In v9 clip corner is handled in a more generic way which is less optimal in terms of memory and speed but can be accelerated by GPUs. 

I see some optimization opportunities here,  but until that could you try it with a larger `LV_MEM_SIZE`?

tkernelcn, 2023-11-27:
Thanks reply
very strange：LV_MEM_SIZE is 4MB, I think is enough
```c
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_STRING    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_BUILTIN

#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
#ifdef BUILD_ARM
    #define LV_MEM_SIZE (46 * 1024U)          /*[bytes]*/
#else
    #define LV_MEM_SIZE (4096 * 1024U)          /*[bytes]*/
#endif
    /*Size of the memory expand for `lv_malloc()` in bytes*/
    #define LV_MEM_POOL_EXPAND_SIZE 0

    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
    #define LV_MEM_ADR 0     /*0: unused*/
    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
    #if LV_MEM_ADR == 0
        #undef LV_MEM_POOL_INCLUDE
        #undef LV_MEM_POOL_ALLOC
    #endif
#endif  /*LV_USE_MALLOC == LV_STDLIB_BUILTIN*/
```

tkernelcn, 2023-11-27:
if not use load api：
```c
void lv_example_scroll_6(void)
{
    lv_obj_t * cont = lv_obj_create(lv_scr_act());
    lv_obj_set_size(cont, 480, 480);
    lv_obj_center(cont);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_COLUMN);
    lv_obj_add_event(cont, scroll_event_cb, LV_EVENT_SCROLL, NULL);
    lv_obj_set_style_radius(cont, LV_RADIUS_CIRCLE, 0);
    //lv_obj_set_style_clip_corner(cont, true, 0);
    lv_obj_set_scroll_dir(cont, LV_DIR_VER);
    lv_obj_set_scroll_snap_y(cont, LV_SCROLL_SNAP_CENTER);
    lv_obj_set_scrollbar_mode(cont, LV_SCROLLBAR_MODE_OFF);

    lv_obj_set_style_border_width(cont, 1, 0);
    lv_obj_set_style_border_color(cont, lv_color_hex(0xFF0000), 0);

    uint32_t i;
    for(i = 0; i < 20; i++) {
        lv_obj_t * btn = lv_button_create(cont);
        lv_obj_set_width(btn, lv_pct(100));

        lv_obj_t * label = lv_label_create(btn);
        lv_label_set_text_fmt(label, "Button %"LV_PRIu32, i);
    }

    /*Update the buttons position manually for first*/
    lv_obj_send_event(cont, LV_EVENT_SCROLL, NULL);

    /*Be sure the fist button is in the middle*/
    lv_obj_scroll_to_view(lv_obj_get_child(cont, 0), LV_ANIM_OFF);

    //lv_screen_load_anim(cont, LV_SCR_LOAD_ANIM_NONE, 500, 0, false);
}
```

[image]

kisvegabor, 2023-11-28:
```c
    lv_obj_t * cont = lv_obj_create(NULL);
    lv_obj_set_size(cont, 480, 480);
    lv_obj_center(cont);
```

Here you have created a screen and tried to set its size and position which is not valid.

In the last code it's already correct:
```c
    lv_obj_t * cont = lv_obj_create(lv_scr_act());
    lv_obj_set_size(cont, 480, 480);
    lv_obj_center(cont);
```

Here are you sure the `#else` branch is active? What if you add `#error "test"` there?
```c
#ifdef BUILD_ARM
    #define LV_MEM_SIZE (46 * 1024U)          /*[bytes]*/
#else
    #define LV_MEM_SIZE (4096 * 1024U)          /*[bytes]*/
#endif

tkernelcn, 2023-11-28:
Hi @kisvegabor 

yes I know below can show correctly 
```c
    lv_obj_t * cont = lv_obj_create(lv_scr_act());
    lv_obj_set_size(cont, 480, 480);
    lv_obj_center(cont);
```
why I try to use below: I have developed a simple framework to create container , and load it as my logic
and the load mechanism can work correctly on v8, so in v9 will not supported? or how to enable it.
```c
    lv_obj_t * cont = lv_obj_create(NULL);
    lv_obj_set_size(cont, 480, 480);
    lv_obj_center(cont);
    //......
   lv_screen_load_anim(cont, LV_SCR_LOAD_ANIM_NONE, 500, 0, false);
```

regarding below, sure I already test with code like `#error "test"`,  4MB it is.
```c
#ifdef BUILD_ARM
    #define LV_MEM_SIZE (46 * 1024U)          /*[bytes]*/
#else
    #define LV_MEM_SIZE (4096 * 1024U)          /*[bytes]*/
#endif
```
Thanks a lot.

kisvegabor, 2023-11-28:
> why I try to use below: I have developed a simple framework to create container , and load it as my logic
and the load mechanism can work correctly on v8, so in v9 will not supported? or how to enable it.

It should work in v9 too, but you just shouldn't set the size and position of a screen.

>  sure I already test with code like #error "test", 4MB it is.

Please enable `LV_USE_MEM_MONITOR` to see what's happening here, and try with a smaller radius first. (`LV_USE_SYSMON` and `LV_USE_OBSERVER` needs to be enabled too)

tkernelcn, 2023-11-29:
> It should work in v9 too, but you just shouldn't set the size and position of a screen.
my purpose is create a 480x480 circle in the center， and support clip corner
in this case， also support multi screen load/switch;  so how to setup size and position and load it?
thanks.

enabled `LV_USE_MEM_MONITOR`,  `LV_USE_SYSMON `  and `LV_USE_OBSERVER `
seems There is still sufficient memory available
[image]

tkernelcn, 2023-11-29:
seems v8 works correctly, would you please try it on your side， just lv_example_scroll_6（）on v9
change `lv_obj_set_size(cont, 200, 200);` to `lv_obj_set_size(cont, 480, 480);`
thanks.

[image]

kisvegabor, 2023-11-30:
Thanks for the update. 
I consider it as an optimization opportunity or bug. Now we add the last required new features to v9 and after that we can look into these kind of issues too.

tkernelcn, 2023-11-30:
Hi I just update today‘s commit，It‘s OK now
issue on:
commit 5c05a341b94695bb1c177d4ca5ba0d96e4c28a8c
latest OK one:
commit 8094cd1acbad237effed053f7d4c4f5d645510dc
I don't know which code line fix it.
Thanks for your support!
[image]

kisvegabor, 2023-11-30:
Amazing! I close this issue then.