lvgl/lvgl #4880: lv_disp_get_default always returns 0

HugoWeijersContecto, 2023-11-27:
### Introduce the problem
I am trying to use LVGL to program a display with an ILI9488 display module and a FT6x63 touch module in ESP-IDF 5.1 in VSCode. I'm working in LVGL version 8.3 and I'm trying to get the code working for 'lv_example_get_started_1.c'. When I try to run the code, the terminal keeps repeating the following error: Guru Meditation Error: 

```
__Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200a195  PS      : 0x00060930  A0      : 0x82009991  A1      : 0x3fca60a0
0x4200a195: lv_obj_mark_layout_as_dirty at C:/Users/hugo/sample_project/components/lvgl/src/core/lv_obj_pos.c:290

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fca5f30  A5      : 0x00000243
A6      : 0x3c03a154  A7      : 0x00000000  A8      : 0x3fca24b8  A9      : 0x3fca6080
A10     : 0x00000000  A11     : 0x3fc96a78  A12     : 0x00000024  A13     : 0x0000036f
A14     : 0x3c03a154  A15     : 0x00000048  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000022  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000
0x40056f5c: memcpy in ROM

0x40056f72: memcpy in ROM

Backtrace: 0x4200a192:0x3fca60a0 0x4200998e:0x3fca60c0 0x42013c6f:0x3fca60e0 0x4201134e:0x3fca6100 0x42008a21:0x3fca6120 0x42008a13:0x3fca6140 0x42027f57:0x3fca6160 0x4037d159:0x3fca6190
0x4200a192: lv_obj_mark_layout_as_dirty at C:/Users/hugo/sample_project/components/lvgl/src/core/lv_obj_pos.c:289

0x4200998e: lv_obj_class_init_obj at C:/Users/hugo/sample_project/components/lvgl/src/core/lv_obj_class.c:106

0x42013c6f: lv_btn_create at C:/Users/hugo/sample_project/components/lvgl/src/widgets/lv_btn.c:53

0x4201134e: lv_example_get_started_1 at C:/Users/hugo/sample_project/components/lvgl/examples/get_started/lv_example_get_started_1.c:23

0x42008a21: c_screen_example1 at C:/Users/hugo/sample_project/main/lib/c_screen/c_screen.c:14

0x42008a13: app_main at C:/Users/hugo/sample_project/main/main.c:45

0x42027f57: main_task at C:/Users/hugo/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)

0x4037d159: vPortTaskWrapper at C:/Users/hugo/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162_
```

### Examples and cases

Every time I run the code the issue occurs.

### Suggested solution
I have read that this error results in a NULL pointer, so I started debugging the code. I found out that going through the functions as followed, beginning from the lv_example_get_started_1.c: 

lv_btn_create() -> lv_obj_class_create_obj() -> lv_disp_get_default(). Here I found out, that the variable disp gets set to disp_def. After that, in the function lv_obj_class_create_obj() it gets checked if disp is 0 and if it iss, the function itseld will return NULL.

So I know what causes my issue, but I have no clue how to fix it. The code can be found from the LVGL github repository version 8.3: https://github.com/lvgl/lvgl.git

HugoWeijersContecto, 2023-11-27:
Okay, I just turned on the logging correctly, and this is my output, maybe this means more to someone than my explanation: 
[image]

kisvegabor, 2023-11-27:
I seems you haven't created a display with `lv_disp_drv_register`. See [here](https://docs.lvgl.io/8.3/porting/display.html).

HugoWeijersContecto, 2023-11-27:
Yes, you are right. Just before you responded I found this out. Apologies for my ignorance. However I now do have a question. In the LVGL version 8.3 github there is a porting template with everything described in the link you sent. Everything in there is very clear, except for where I need to place the copies of these templates and what I need to put in disp_init(). I think it is the ILI9488_init() in the LVGL_drivers, but I'm uncertain as I don't know if I can include this file without knowing where to put the lv_port_disp files. 

So could you please tell me where the lv_port_disp files should go and what I should put in the disp_init()?

HugoWeijersContecto, 2023-11-27:
Hi, I added the ili9488_init(), but I still keep getting this error:

```
FAILED:` main.elf
cmd.exe /C "cd . && C:\Users\hugo\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/hugo/sample_project/build/main.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\main.elf.rsp -o main.elf  && cd ."
c:/users/hugo/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(c_screen.c.obj):(.literal.c_screen_example1+0x0): undefined reference to `lv_port_disp_init'
c:/users/hugo/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(c_screen.c.obj): in function `c_screen_example1':
C:/Users/hugo/sample_project/main/lib/c_screen/c_screen.c:12: undefined reference to `lv_port_disp_init'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

My code now is: 

```
void c_screen_example1(void){
    
    lv_init();
    lvgl_driver_init();

    lv_port_disp_init();

    lv_example_get_started_1();
}
```

kisvegabor, 2023-11-28:
Probably too large display draw buffer.

HugoWeijersContecto, 2023-11-28:
Thank you very much, after your reaction I went to the display draw buffers and I realised that the lv_port_disp_template. shows off the 3 different ways you can use these buffers, using a total of 5 buffers. This obviously cost me way too much memory. Thank you for your help!