lvgl/lvgl #4885: lvgl micropython

klora-sun, 2023-11-28:
hi, i have put [lv_binding_micropython](https://github.com/lvgl/lv_binding_micropython)as lv_bingings into microrpython lib,and i write this `#LittlevGL
LVGL_BINDING_DIR = $(TOP)/lib/lv_bindings
LVGL_DIR = $(LVGL_BINDING_DIR)/lvgl
LVGL_GENERIC_DRV_DIR = $(LVGL_BINDING_DIR)/driver/generic
INC += -I$(LVGL_BINDING_DIR)
ALL_LVGL_SRC = 
(LVGL_DIR) -type f) $(LVGL_BINDING_DIR)/lv_conf.h
LVGL_PP = $(BUILD)/lvgl/lvgl.pp.c
LVGL_MPY = $(BUILD)/lvgl/lv_mpy.c
LVGL_MPY_METADATA = $(BUILD)/lvgl/lv_mpy.json
QSTR_GLOBAL_DEPENDENCIES += $(LVGL_MPY)
CFLAGS_MOD += $(LV_CFLAGS)

(ALL_LVGL_SRC) $(LVGL_BINDING_DIR)/gen/gen_mpy.py
@"
(dir $@)
(CPP) 
(LVGL_BINDING_DIR)/pycparser/utils/fake_libc_include 
(LVGL_DIR)/lvgl.h > $(LVGL_PP)
(PYTHON) 
(LVGL_MPY_METADATA) -E 
(LVGL_DIR)/lvgl.h > $@

CFLAGS_MOD += -Wno-unused-function
SRC_MOD += 
(TOP)/,,$(shell find 
(LVGL_GENERIC_DRV_DIR) -type f -name "*.c") $(LVGL_MPY))
in py.mk.then wehen i make BOARD,the error showUse make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LVGL-GEN build-REGh743LQ208RMI_lvgl/lvgl/lv_mpy.c
Traceback (most recent call last):
File "/home/admin1/micropython/ports/stm32/../../lib/lv_bindings/gen/gen_mpy.py", line 299, in
ast = parser.parse(s, filename='')
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/c_parser.py", line 149, in parse
return self.cparser.parse(
^^^^^^^^^^^^^^^^^^^
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/ply/yacc.py", line 331, in parse
return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/ply/yacc.py", line 1199, in parseopt_notrack
tok = call_errorfunc(self.errorfunc, errtoken, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/ply/yacc.py", line 193, in call_errorfunc
r = errorfunc(token)
^^^^^^^^^^^^^^^^
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/c_parser.py", line 1861, in p_error
self._parse_error(
File "/home/admin1/micropython/lib/lv_bindings/gen/../pycparser/pycparser/plyparser.py", line 67, in _parse_error
raise ParseError("%s: %s" % (coord, msg))
pycparser.plyparser.ParseError: ../../lib/lv_bindings/lvgl/src/misc/lv_log.h:72:39: before: attribute
make: *** [../../py/py.mk:41：build-REGh743LQ208RMI_lvgl/lvgl/lv_mpy.c] 错误 1
make: *** 正在删除文件“build-REGh743LQ208RMI_lvgl/lvgl/lv_mpy.c”`
could someone help

XuNeo, 2023-11-28:
You can follow https://github.com/lvgl/lv_micropython for a quickstart.
Note if you want to test latest master branch of lvgl, switch to branch `feat/multi-instance` for `lv_micropython` and `lv_micropython/lib/lv_bindings`.

SL2021-Dev, 2023-11-29:
How would I go about doing that.
This is how I pulled in the past...
`
git clone https://github.com/lvgl/lv_micropython.git
cd lv_micropython
git submodule update --init --recursive lib/lv_bindings
`

closed 2023-12-22 by lvgl-bot