lvgl/lvgl #4895: Demo widgets chart coordinates

anaGrad, 2023-11-29:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I ran the lv_demo_benchmark and when it got to the widgets demo I saw that in the chart drawn with triangles and rectangles is a missing line of pixels between the two. I attached a picture with the result:
[image]

Then I took a look at the coordinates of the triangles and rectangles:
First triangle:
[image]

Corresponding rectangle: 
[image]

Second triangle:
[image]

Corresponding rectangle:
[image]

And I noticed that between the end of a triangle and beginning of a rectangle is a difference of 1 on Y axis, but on the X axis the shapes start where the previous ended:
[image]

Is this an error in the coordinates of the rectangles or is the expected behaviour that the two of them shouldn't have that 1px difference? If that is the expected behaviour, then why isn't there a 1px difference in the X coordinates? 
With the CPU the problem doesn't reproduce since the CPU is drawing including the line corresponding to the bottom of the shape.

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-11-30:
Hi, 

It works well for me with the latest version from `master`. Please be suer that you are also using the latest.

kisvegabor, 2023-12-04:
I see. Really there was a bug in SW render. Could you try with this patch?
```diff
diff --git a/src/draw/sw/lv_draw_sw_mask.c b/src/draw/sw/lv_draw_sw_mask.c
index 74c0869f6..d9152d5c4 100644
--- a/src/draw/sw/lv_draw_sw_mask.c
+++ b/src/draw/sw/lv_draw_sw_mask.c
@@ -451,7 +451,7 @@ LV_ATTRIBUTE_FAST_MEM static lv_draw_sw_mask_res_t lv_draw_mask_line(lv_opa_t *
             /*Non sense: Can't be on the right/left of a horizontal line*/
             if(p->cfg.side == LV_DRAW_SW_MASK_LINE_SIDE_LEFT ||
                p->cfg.side == LV_DRAW_SW_MASK_LINE_SIDE_RIGHT) return LV_DRAW_SW_MASK_RES_FULL_COVER;
-            else if(p->cfg.side == LV_DRAW_SW_MASK_LINE_SIDE_TOP && abs_y + 1 < 0) return LV_DRAW_SW_MASK_RES_FULL_COVER;
+            else if(p->cfg.side == LV_DRAW_SW_MASK_LINE_SIDE_TOP && abs_y < 0) return LV_DRAW_SW_MASK_RES_FULL_COVER;
             else if(p->cfg.side == LV_DRAW_SW_MASK_LINE_SIDE_BOTTOM && abs_y > 0) return LV_DRAW_SW_MASK_RES_FULL_COVER;
             else {
                 return LV_DRAW_SW_MASK_RES_TRANSP;
diff --git a/src/draw/sw/lv_draw_sw_triangle.c b/src/draw/sw/lv_draw_sw_triangle.c
index f2b1eaec8..ee24ca0c8 100644
--- a/src/draw/sw/lv_draw_sw_triangle.c
+++ b/src/draw/sw/lv_draw_sw_triangle.c
@@ -104,9 +104,14 @@ void lv_draw_sw_triangle(lv_draw_unit_t * draw_unit, const lv_draw_triangle_dsc_
                                      p[2].x, p[2].y,
                                      right ? LV_DRAW_SW_MASK_LINE_SIDE_LEFT : LV_DRAW_SW_MASK_LINE_SIDE_RIGHT);
 
-    lv_draw_sw_mask_line_points_init(&mask_bottom, p[1].x, p[1].y,
-                                     p[2].x, p[2].y,
-                                     right ? LV_DRAW_SW_MASK_LINE_SIDE_LEFT : LV_DRAW_SW_MASK_LINE_SIDE_RIGHT);
+    if(p[1].y == p[2].y) {
+        lv_draw_sw_mask_line_points_init(&mask_bottom, p[1].x, p[1].y,
+                                         p[2].x, p[2].y, LV_DRAW_SW_MASK_LINE_SIDE_TOP);
+    } else {
+        lv_draw_sw_mask_line_points_init(&mask_bottom, p[1].x, p[1].y,
+                                         p[2].x, p[2].y,
+                                         right ? LV_DRAW_SW_MASK_LINE_SIDE_LEFT  : LV_DRAW_SW_MASK_LINE_SIDE_RIGHT);
+    }
 
     masks[0] = &mask_left;
     masks[1] = &mask_right;
diff --git a/demos/widgets/lv_demo_widgets.c b/demos/widgets/lv_demo_widgets.c
index d1190f84e..9878eb9c7 100644
--- a/demos/widgets/lv_demo_widgets.c
+++ b/demos/widgets/lv_demo_widgets.c
@@ -1452,7 +1452,7 @@ static void chart_event_cb(lv_event_t * e)
             lv_area_t rect_area;
             rect_area.x1 = (int32_t)draw_line_dsc->p1.x;
             rect_area.x2 = (int32_t)draw_line_dsc->p2.x;
-            rect_area.y1 = (int32_t)LV_MAX(draw_line_dsc->p1.y, draw_line_dsc->p2.y) + 1;
+            rect_area.y1 = (int32_t)LV_MAX(draw_line_dsc->p1.y, draw_line_dsc->p2.y);
             rect_area.y2 = (int32_t)obj->coords.y2;
             lv_draw_rect(base_dsc->layer, &rect_dsc, &rect_area);
         }
```

I believe now none of the coordinates are included.

closed 2023-12-05 by anaGrad

kisvegabor, 2023-12-05:
The patch is added in [3c7e8bd](https://github.com/lvgl/lvgl/commit/3c7e8bd0d4ba4de931682aa89a7492cc00f34797)