lvgl/lvgl #4901: Vote for addon in future releases, some kind of LVGL virtual display driver

djixon, 2023-11-30:
It would be nice to have some (integrated into LVGL) virtual display driver capable to route final rendered buffer not to a real LCD driver but rather to virtual driver on WiFi or LAN module and having "catcher" of the buffer at "other side", let's say some web browser capable to stream such buffer and "show" uncompressed content in web page. For example, according to math, if you configure LVGL for true color RGB888 format, some display of let's say 480x320 pixels has 153600 pixels. By multiplying with 24 bits per pixels it ends to 3686400 bits per buffer. To achieve, let's say, 25 fps, it ends at 92160000 which is under 100Mbits per second (without any kind of compression). 

It would be a huge benefit for developing, a graphical application targeting  ESP32 families, already equipped with WiFi modules, where you would be able to use LVGL even without implemented LCD on the board, but rather connecting an ESP32 wirelessly to your web browser on mobile phone or PC.  Maybe an indev driver could be also advanced to accept, lets say, actions from the touch of your mobile phone (also wirelessly) or mouse actions from remote web page, which would then make it truly interactive., opening door to many new kind of IoT or other really small devices (now without LCD) but still with graphical capabilities through a wireless connection to any device with LCD display and web browser

kisvegabor, 2023-11-30:
It would be rally amazing. It would be also great to use a standard remote desktop protocol, so it could be used with the existing tools.

djixon, 2023-11-30:
It would be even better because of already established standards. There's a lot of open source versions supporting RDP like TightVNC, MightyViewer, XRDP, X2Go, Nomachine, Anydesk etc. But I don't know how robust server side of any of those is. Can it fit into tight memory model of ESP32 (or similar modules)?

kisvegabor, 2023-12-01:
I don't know ether, but I know for sure that it would be great and theoretically LVGL is fully capable of doing it. We could both stream the image of an existing display, or have only a virtual display, where each remote user has their own "session".  It could be a good starting point: https://github.com/LibVNC/libvncserver/tree/master

closed 2023-12-24 by lvgl-bot