lvgl/lvgl #4905: MSVC will return error C4576 for the following code

MouriNaruto, 2023-12-01:
https://github.com/lvgl/lvgl/blob/ba229b0708fe3b893a179da3b3635a807b046c4b/src/misc/lv_area.h#L267-L279

Kenji Mouri

kisvegabor, 2023-12-01:
Thank you for reporting it!  Do you see an easy fix?
cc @XuNeo

XuNeo, 2023-12-01:
It has been discussed before. 
https://forum.lvgl.io/t/visual-studio-2019-compile-errors/1379/23 
https://stackoverflow.com/questions/33270731/error-c4576-in-vs2015-enterprise/33271165#33271165

@MouriNaruto Could you try if this fix it?

MouriNaruto, 2023-12-01:
@XuNeo 

We shouldn't force users use LVGL in pure C projects. Especially it will limit the usage case of LVGL.

Kenji Mouri

MouriNaruto, 2023-12-01:
@XuNeo 

> Could you try if this fix it?

I will try you patch. Please wait.

Kenji Mouri

PR #4907 (2023-12-01): fix: workaround MSVC C4576 error

### Description of the feature or fix

Fix #4905

Simply initialize a struct and return it instead
### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` r [... cut]