lvgl/lvgl #4910: [Feature Request] Create and draw widget on canvas

XuNeo, 2023-12-01:
### Introduce the problem

For some complicated widget like chart, It's expensive to redraw them especially when scrolling page.
Cache the draw result into image and simple redraw the image could be faster.

### Examples and cases

In v8, I have tried to rewrite the canvas widget to include a virtual display, and any children created on virtual display will be drawn directly to the canvas. All the input event canvas received will be forwarded to the widgets inside virtual root.
See simplified version ready on simulator.
https://github.com/XuNeo/lv_sim_eclipse_sdl/commit/043754cb3dd32aaa98c3e3e6882c55bab361ceda

### Suggested solution

For v9, we need to discuss the possible solutions. 
We may considering adding the ability to draw any `widget tree` to a specified layer.
Or we can add widget level cache mechanism.

kisvegabor, 2023-12-03:
> Or we can add widget level cache mechanism.

I vote for this one. For me this idea seems independent from the layers.  However, I think it should be a v9.x feature. 

What do you think?

XuNeo, 2023-12-04:
Agree. 

Another relevant topic is supporting hardware layers. Utilizing the hardware ability to blend layers to final frame buffer.
It should be a hardware resource registered to `lvgl`, and be usable by applications.

Let's consider this feature after v9 release.

kisvegabor, 2023-12-04:
> Utilizing the hardware ability to blend layers to final frame buffer.

So far I imagined these layers as separate `lv_display`s because from LVGL's point of view these are not related to each other.  What do you think about it?

XuNeo, 2023-12-05:
cc @terry0012

terry0012, 2023-12-16:
> > Or we can add widget level cache mechanism.
> 
> I vote for this one. For me this idea seems independent from the layers.  However, I think it should be a v9.x feature. 
> 
> What do you think?

Nice!  this method can save a lot time in scroll mode.
FPS will be better in complex UI pages.

terry0012, 2023-12-16:
> cc @terry0012 

I think each hardware layer is a lv_display_t is ok, we can discuss it internal next week first.

kisvegabor, 2024-01-01:
Not stale

kisvegabor, 2024-01-17:
In comes done to widget cashing which is an important feature so I pinned this issue.

kisvegabor, 2024-01-17:
It can be a good candidate for the next round of shaping.

kisvegabor, 2024-09-05:
Added to the v9.3 Roadmap as widget caching.
https://github.com/lvgl/lvgl/issues/6763

halyssonJr, 2026-11-16:
@kisvegabor , given that the v9.3 roadmap has been completed, should we close this issue?

AndreCostaaa, 2026-11-16:
It hasn't been implemented @halyssonJr