lvgl/lvgl #4917: The issue of using scale to replace meter

lhdjply, 2023-12-03:
In v9, the **meter**  has been removed and replaced with **scale.** However, if you want to use **needle,** currently it can only be achieved through images and rotation functions. In V8, the meter can directly set numerical values, and the needle can display the corresponding position, making it very convenient to use.
Therefore, should we restore the **meter** control or add a separate **needle** control? This way, users can also reduce the difficulty of use.

kisvegabor, 2023-12-03:
It's an interesting situation. There are cases when the scale is bound to an other object (e.g. [chart scale](https://docs.lvgl.io/master/widgets/chart.html#axis-ticks-and-labels-with-scrolling)), and in other cases widgets are bound to the scale (meter-like behavior).

This issue is about the second case, and I agree to make it simpler for the users. What about something like this:
- The user can create an images on the scale.
- There is a function like `lv_scale_set_needle_value(scale, image, value)` which set the image rotation 
- Note that this way the value is not stored anywhere but directly mapped to an angle in `lv_scale_set_needle_value`.

What do you think?

lhdjply, 2023-12-04:
This is indeed a good method. However, I think there are two ways to add it. 
The first method is through images, such as 
- `lv_scale_set_needle_value_image (scale, image, value);`

Another method is through lines, such as 
- `lv_scale_set_needle_value_line (scale, line_length, line_width, value);`

kisvegabor, 2023-12-04:
> Another method is through lines, such as
lv_scale_set_needle_value_line (scale, line_length, line_width, value);

If we want to support lines too we can assume that the line widget is created on the scale with the desired width and other style properties. 

So it can be simplified to:
```c
lv_scale_set_needle_value_line (scale, line_length, value);
```

If you agree, do you have time to send a PR with this update?

lhdjply, 2023-12-05:
> If you agree, do you have time to send a PR with this update?

Of course, but I'll have to wait a few days. I've been a bit busy lately.

> There is a function like lv_scale_set_needle_value(scale, image, value) which set the image rotation

I found a difficulty. If we use images, we cannot know which direction the user is using. The images they are using may be vertical, horizontal, or diagonal.

kisvegabor, 2023-12-05:
> I found a difficulty. If we use images, we cannot know which direction the user is using. The images they are using may be vertical, horizontal, or diagonal.

In meter it was mandatory to use an image pointing to the right. E.g. `-O------>`
I think it's an acceptable request from the user.

lhdjply, 2023-12-05:
> > I found a difficulty. If we use images, we cannot know which direction the user is using. The images they are using may be vertical, horizontal, or diagonal.
> 
> In meter it was mandatory to use an image pointing to the right. E.g. `-O------>` I think it's an acceptable request from the user.

OK👌

lhdjply, 2023-12-06:
continue #4938

closed 2023-12-06 by lhdjply

xinsuinizhuan, 2024-03-06:
could add the meter come back?