lvgl/lvgl #4926: Finalize v9

kisvegabor, 2023-12-04:
Continue #3298 (see for the changes too)

### Schedule
1. **December 4**: Feature stop, start updating the docs and testing
2. **December 18**: Release candidate version and call to test
3. **January 15**: Release v9.0

### 1. Features to finalize
We don't add any new features but finalize these ongoing ones 
- [ ] Image decoder fine tuning (@XuNeo )
- [ ] Online image converter (@W-Mai)
- [ ] Cache update (@W-Mai and @FASTSHIFT )
- [ ] Scale needle handling (@lhdjply ? #4917)
- [ ] Pointer scroll support (@kisvegabor / @XuNeo #4909)
- [ ] Add ESP-LCD support (@zjanosy  #4507)
- [ ] Add support for at least 3 display controller (@zjanosy)
- [ ] Remove the MicroPython examples and replace them with a test (@kdschlosser #4645)

### 2. Prepare for release candidate (by December 18)
Docs review
- [ ] [Porting](https://docs.lvgl.io/master/porting/index.html) (@kisvegabor )
- [ ] [Drivers](https://docs.lvgl.io/master/integration/index.html) ( :warning: Help wanted)
- [ ] [Widgets](https://docs.lvgl.io/master/widgets/index.html) (@kisvegabor )
- [ ] [Overview](https://docs.lvgl.io/master/overview/index.html) (especially the image part)  ( :warning: Help wanted)

Demos end examples
- [ ] [Examples](https://docs.lvgl.io/master/examples.html) (@kisvegabor )
- [ ] Review and update demos (@kisvegabor )

### 3. Fixes and last touches (by January 15)
Mainly for January
- [ ] CI update with more config options (@XuNeo and @kisvegabor)
- [ ] Final NXP drivers (@nicusorcitu)
- [ ] Arm2D support? (@GorgonMeducer )
- [ ] Project update (E.g. Eclipse, VSCode, etc)  ( :warning:  Help wanted)
- [ ] Changelog and migration guide (@kisvegabor)
- [ ] **Release**

nicusorcitu, 2023-12-05:
Is screen rotation still a task for v9?

GorgonMeducer, 2023-12-05:
OK, I will update the cmsis-pack by end of this week.
In the coming two weeks, I will finish the porting for Helium.

kisvegabor, 2023-12-06:
Amazing, thank you Gabriel!

nicusorcitu, 2023-12-06:
> Earlier we have decided to provide platform specific helper functions. I've added these for SW [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.c#L169-L188).
>

Ok, I will add too for VGLite and PXP. But as I see, there is no callback mechanism yet, just the implementation.

Till 15Jan I will post another VGLite update to github, not sure is going to be the final one because I have to migrate to vglite 4.x and solve the issues. But I can do another one after the official v9 release, correct?

kisvegabor, 2023-12-06:
> Ok, I will add too for VGLite and PXP. But as I see, there is no callback mechanism yet, just the implementation.

Yes, this way the user can select what he wants in a transparent way. Do you agree with this?

> Till 15Jan I will post another VGLite update to github, not sure is going to be the final one because I have to migrate to vglite 4.x and solve the issues. But I can do another one after the official v9 release, correct?

Ideally we should have a stable version by Jan 15 even if doesn't support everything. Of course updates can be added later too.

MouriNaruto, 2023-12-07:
When I am writing the resizing or change LVGL resolution support for new Windows backend, I found it will cause memory access violation because the rendering thread is running and we cannot make it stop temporarily.

But I hope LVGL can apply and fire the resolution changed event at the proper timing automatically and it will reduce lots of dirty hack for backends.

Kenji Mouri

nicusorcitu, 2023-12-07:
>>    Ok, I will add too for VGLite and PXP. But as I see, there is no callback mechanism yet, just the implementation.
>>
> Yes, this way the user can select what he wants in a transparent way. Do you agree with this?
>

I though the decision was to make the screen rotation a task that shall be executed as any other task by the unit thread.

espzav, 2023-12-07:
Hi @kisvegabor, I started with investigation of LVGLv9 for using with Espressif's examples, IDF and BSP. 
I found one issue. I cannot use this: `lv_display_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED)` because `LV_COLOR_FORMAT_NATIVE_REVERSED` is not exists. 
Please, could you help me?
Thank you very much!

espzav, 2023-12-07:
@kisvegabor Please, I have another question about images. When I generated image on lvgl.io website, I got some code. There are two defines ,which not exists: `LV_IMG_PX_SIZE_ALPHA_BYTE` and `LV_IMG_CF_TRUE_COLOR_ALPHA`. Is there any simple way, how to change it? I tried lot of `LV_COLOR_FORMAT_`, but not good results. It seems, that there is idefs about color and swap, which is not available now.
Please, could you help me?

kdschlosser, 2023-12-07:
With the esp idf I am making an assumption here that you are using the spi bus driver from the esp-lcd component. If you are you can set the byte order in the config structure for the bus.

kdschlosser, 2023-12-07:
I would guess using the config structure would also be faster. It should do the byte swap when the data is being collected to be transfered.

kisvegabor, 2023-12-07:
@MouriNaruto 
> When I am writing the resizing or change LVGL resolution support for new Windows backend, I found it will cause memory access violation because the rendering thread is running and we cannot make it stop temporarily. (It looks timer pause is not work. If we follow the document we find we cannot restore the original render timer elegantly.)

I can't see how can it happen that the rendering is still running when the resolution change event is fired.  In the refresh timer all rendering should be ready when the flush_cb is called in the end. I don't know exactly where do you call `lv_display_set_resolution` but during rendering you shouldn't do anything else. If it's coming from a different thread then `lv_timer_handler` and the other should be protected by a mutex to avoid concurrent running.  

@nicusorcitu 
> I though the decision was to make the screen rotation a task that shall be executed as any other task by the unit thread.

I haven't searched for the conversation, but it still make sense to me to provide only helper for maximal flexibility. Do you think it's a bad approach?  

@espzav 
> Hi @kisvegabor, I started with investigation of LVGLv9 for using with Espressif's examples, IDF and BSP.
I found one issue. I cannot use this: lv_display_set_color_format(disp, LV_COLOR_FORMAT_NATIVE_REVERSED) because LV_COLOR_FORMAT_NATIVE_REVERSED is not exists.

It's great you have tried out v9! I've just updated the comment and docs in 892eca900

 > Please, I have another question about images. When I generated image on lvgl.io website, I got some code.

See the [comment](https://github.com/lvgl/lvgl/issues/4926#issuecomment-1845610631) of @XuNeo

lhdjply, 2023-12-08:
* [x]  Scale needle handling (@lhdjply https://github.com/lvgl/lvgl/issues/4917 https://github.com/lvgl/lvgl/pull/4938)

“Scale needle handling ”The feature has been completed

kdschlosser, 2023-12-08:
Think you could make the meter more flexible with what can be done with it? Something along these lines?

https://www.youtube.com/watch?v=NxQ7cLobr_Q

lhdjply, 2023-12-08:
> Think you could make the meter more flexible with what can be done with it? Something along these lines?
> 
> https://www.youtube.com/watch?v=NxQ7cLobr_Q

@kdschlosser I think the effects in the video can be temporarily achieved through scale and arc
```c
lv_obj_t * scale;
lv_obj_t * label;

static void value_changed_event_cb(lv_event_t * e)
{
    uint32_t value;
    lv_obj_t * arc = lv_event_get_target(e);
    value=lv_arc_get_value(arc);

    lv_scale_set_total_tick_count(scale, value+1);
    lv_scale_set_major_tick_every(scale, 5);
    lv_scale_set_round_props(scale, value*2.7, 135);

    lv_label_set_text_fmt(label, "%"LV_PRId32"%%", value);
}

void lv_example_scale(void)
{
    scale=lv_scale_create(lv_screen_active());

    lv_obj_set_size(scale, 200, 200);
    lv_scale_set_mode(scale, LV_SCALE_MODE_ROUND_OUTER);
    lv_obj_set_style_bg_opa(scale, LV_OPA_COVER, LV_PART_MAIN);
    lv_obj_set_style_bg_color(scale, lv_palette_lighten(LV_PALETTE_RED, 5), LV_PART_MAIN);
    lv_obj_set_style_radius(scale, LV_RADIUS_CIRCLE, LV_PART_MAIN);
    lv_obj_set_style_clip_corner(scale, true,LV_PART_MAIN);
    lv_obj_align(scale,LV_ALIGN_CENTER,0,0);

    lv_scale_set_label_show(scale, false);
    lv_scale_set_range(scale, 0, 100);

    lv_obj_t * arc = lv_arc_create(scale);
    lv_obj_set_size(arc, 100, 100);
    lv_arc_set_rotation(arc, 135);
    lv_arc_set_bg_angles(arc, 0, 270);
    lv_arc_set_value(arc, 10);
    lv_obj_center(arc);

    lv_scale_set_total_tick_count(scale, 11);
    lv_scale_set_major_tick_every(scale, 5);
    lv_scale_set_round_props(scale, 27, 135);

    label=lv_label_create(scale);
    lv_obj_center(label);
    lv_label_set_text(label,"10%");

    lv_obj_add_event_cb(arc, value_changed_event_cb, LV_EVENT_VALUE_CHANGED, 0);
}
```

kdschlosser, 2023-12-08:
That example was made using LVGL. It's blurry because of the bad screen recorder program I used. (wish I knew of a free one that didn't look like poop). I didn't know if manipulating the meter widget could be done or maybe the addition of a "knob" widget might be an idea.

XuNeo, 2023-12-08:
> > The online converter has not been upgraded to v9, you can for now using the script LVGLImage.py in lvgl/scripts/ to convert PNG images to binary. Here's the help message.
> 
> Thank you @XuNeo! But is there a way to use old C array of image and transform it to new C array? For example when I have some old projects and I want to use new LVGL9 and I haven't got original images?

Yes, you can use  `--ofmt C`. Find more info using `--help`

nicusorcitu, 2023-12-08:
The key is that `disp_drv->full_refresh=1` doesn't support screen rotation.

espzav, 2023-12-08:
> With the esp idf I am making an assumption here that you are using the spi bus driver from the esp-lcd component. If you are you can set the byte order in the config structure for the bus.

@kdschlosser Please, did you try it? I found, that I can change endianess in LCD controller, but it is not working. For example in ST7789 is register RAMCTRL inside command table 2 and it is available only when EXTC pin is in LOW, but this pin is not available on my HW. And not all LCD drivers support swap endianess.

kisvegabor, 2023-12-08:
@nicusorcitu 
> The key is that disp_drv->full_refresh=1 doesn't support screen rotation.
Please read the next 5 comments starting from https://github.com/lvgl/lvgl/issues/4060#issuecomment-1484546542

I think there were a more recent discussion about it too. I'll search for it later. Anyway, with the helper functions even full refresh can be rotated if there is a large enough destination buffer. For PARTIAL mode the user can decide how to allocate the destination buffer and with what size. 

So it seems really flexible and explicit to me. What would be the benefit of using using a draw_task for it?

kdschlosser, 2023-12-09:
esp_lcd_panel_io_spi_config_t.flags.lsb_first

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/lcd.html?highlight=esp_lcd_panel_io_spi_config_t#_CPPv429esp_lcd_panel_io_spi_config_t

GorgonMeducer, 2023-12-11:
@kisvegabor  the cmsis-pack script is updated for the latest version.
#4977

espzav, 2023-12-11:
> @espzav
> 
> esp_lcd_panel_io_spi_config_t.flags.lsb_first
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/lcd.html?highlight=esp_lcd_panel_io_spi_config_t#_CPPv429esp_lcd_panel_io_spi_config_t

@kdschlosser Are you sure, that it is same as swap bytes? I tried it and it is not working (nothing on the screen).  I think, that this is swap bits not bytes.

nicusorcitu, 2023-12-11:
> So it seems really flexible and explicit to me. What would be the benefit of using using a draw_task for it?
>
1. It would made the life much easier for the users. The screen rotation would be handled internal in lvgl and only triggered by a config or an API call: `lv_display_set_resolution`
2. We could split rotation in two half: one side to be rotated by GPU, the other by CPU.

Having only the API set, the screen rotation will be made outside the lvgl library. The screen rotation elapsed time will not get measured in FPS at all, the users might see wrong FPS calculated by lvgl which measure only the draw task executions.

kisvegabor, 2023-12-11:
> It would made the life much easier for the users. The screen rotation would be handled internal in lvgl and only triggered by a config or an API call: lv_display_set_resolution

I know a customer who uses PXP to rotate the screen sized rendered image into the real frame buffer. In this case he can disable PXP in `lv_conf.h` and use only VGLite and SW rendering. With the helpers it's very simple to use PXP freely. With the draw tasks it's not possible to have this fine grade customization. It's the same for rotating the half with PXP and the other half with SW. It's much less trivial how to configure with draw tasks.

> Having only the API set, the screen rotation will be made outside the lvgl library. The screen rotation elapsed time will not get measured in FPS at all, the users might see wrong FPS calculated by lvgl which measure only the draw task executions.

It's still done in the `flush_cb` which is measured as `flush_avg_time`. The overall FPS is calculated like [this](https://github.com/lvgl/lvgl/blob/master/src/others/sysmon/lv_sysmon.c#L147-L148).

nicusorcitu, 2023-12-11:
> With the draw tasks it's not possible to have this fine grade customization.
>
You can set a preferred score as you wish for every task on each draw unit.

So bottom line, the rotation will continue to be handled externally to the library? Same as it was on lvgl v8?

kisvegabor, 2023-12-11:
> You can set a preferred score as you wish for every task on each draw unit.

However it requires the modification of LVGL itself by the user.

> So bottom line, the rotation will continue to be handled externally to the library same as in v8?
What is the argument to not handle it inside the library as a task?

I'm open for both, but display rotation is not a standard thing that can be done in only one specific way, but the user can use the available peripheries in a creative and special way. By having a draw_task we loose this flexibility.

nicusorcitu, 2023-12-12:
OK, we can leave it this way for now, we can discuss further how can we improve it.

kdschlosser, 2023-12-14:
@espzav 

I believe I know the solution to your color issue.

If you are using one of the ESP-IDF component displays try changing the `color_space` field in the `esp_lcd_panel_dev_config_t` structure to `ESP_LCD_COLOR_SPACE_BGR`

I think that might solve your problem

espzav, 2023-12-14:
> @espzav
> 
> I believe I know the solution to your color issue.
> 
> If you are using one of the ESP-IDF component displays try changing the `color_space` field in the `esp_lcd_panel_dev_config_t` structure to `ESP_LCD_COLOR_SPACE_BGR`
> 
> I think that might solve your problem

@kdschlosser I am not sure, but it is not right. Here is the picture of swap bytes: 
[image]

But when you change color space from BGR to RGB, you only swap red and blue color. 
I have the solution with `lv_draw_sw_rgb565_swap` in `flush` callback. It seems, that there is not HW solution.

kisvegabor, 2023-12-30:
Not stale.

kisvegabor, 2024-01-11:
@MouriNaruto We are getting close to the release. I wonder if you could add even only a SW render based window driver to the lvgl repo. It would be important to have something to get started with for the users.

MouriNaruto, 2024-01-11:
@kisvegabor 

I think after I finished the LVGL display creation function abstraction. I will create a PR. I will do this in this week as soon as possible.

Kenji Mouri

kisvegabor, 2024-01-11:
Thank you! Note that #5204 will be reverted. See here https://github.com/lvgl/lvgl/issues/5273

MouriNaruto, 2024-01-11:
@kisvegabor 

I will keep up that change if the changes are merged.

Kenji Mouri

kisvegabor, 2024-01-11:
Here is the revert PR: https://github.com/lvgl/lvgl/pull/5274
I'll merge it when the CI  passes.

MouriNaruto, 2024-01-12:
@kisvegabor 

https://github.com/lvgl/lvgl/issues/5293

Here is the issue which notes the new Windows backend development progress. Maybe people who care about that can review my finished API design first.

Kenji Mouri

espzav, 2024-01-12:
Hi, please, is it any similar solution for color text in LVGL9 like this:
```
lv_label_set_recolor(label, true);
lv_label_set_text(label, "#FF0000  text #");
```

GorgonMeducer, 2024-01-13:
@kisvegabor The arm-2d support has been added to LVGL9. #5231 #5045

kisvegabor, 2024-01-14:
> Hi, please, is it any similar solution for color text in LVGL9 like this:

It was removed to simply the text rendering. `lv_span` can be used instead. I'm planning to improve it after v9 to make it less verbose to use, and faster to render.

kisvegabor, 2024-01-14:
>I found an issue in the Widgets demo: the bar used to change the UI colour shows no colour dot.

Fixed here #5309

GorgonMeducer, 2024-01-15:
> I'll release v9. @GorgonMeducer Please check if the CMSIS pack are up to date.

No problem, I will do it ASAP.

GorgonMeducer, 2024-01-15:
@kisvegabor  A draft PR(#5323) is created just in case there are other last-minute updates

kisvegabor, 2024-01-15:
@GorgonMeducer 
@XuNeo just mentioned that the current Helium accelerator is actually an ARM2D acceleration and the naming might be misleading. So do you agree to update the  naming from Helium to Arm2D? If so, could you prepare a PR still today to have the release with the new naming?

GorgonMeducer, 2024-01-15:
@kisvegabor When creating the acceleration, I have the idea in mind that:

- Arm-2D provides the helium acceleration for LVGL9
- Arm-2D is not the only possible way of providing helium acceleration to LVGL9
- Whether Arm-2D is present or not, users can add their own helium acceleration 
- When Arm-2D is present, users can use arm-2d helium acceleration as the default ones and override some LVGL intrinsics with their own implementations.
- User define Helium acceleration can be added with the definition of macro `LV_DRAW_SW_HELIUM_CUSTOM_INCLUDE`

Based on aforementioned, I think there is no misleading content and I propose to keep current version.

kisvegabor, 2024-01-15:
> Arm-2D is NOT the only possible way of providing helium acceleration to LVGL9

I think that's why it might be misleading a little bit. Arm2d is one implantation of helium so it's not fortunate to call it the helium implementation.

GorgonMeducer, 2024-01-15:
@kisvegabor Although I reserve the idea to myself, I have made the change you requested, please review it here:
https://github.com/lvgl/lvgl/pull/5323

nicusorcitu, 2024-01-15:
draw_buf have been again updated and nxp code is broken. I will try to fix it up quickly.

nicusorcitu, 2024-01-15:
I do not understand:
1. `layer->draw_buf.header.cf` is the same as `layer->color_format`? Why there are two?
2. `layer->draw_buf.header.w` is the same as `lv_area_get_width(&layer->buf_area)`?

FASTSHIFT, 2024-01-15:
> 1. `layer->draw_buf.header.cf` is the same as `layer->color_format`? Why there are two?

Because when setting `layer->color_format,` `layer->draw_buf` may be a null pointer and cannot be written immediately.

> 2. `layer->draw_buf.header.w` is the same as `lv_area_get_width(&layer->buf_area)`? If so, why the area is separated by w/h?

The reason for retaining `layer->buf_area` is that `buf_area` has xy coordinate information, and the drawing area needs to be offset based on this information.

XuNeo, 2024-01-15:
layer->buf_area may be updated during partial update mode. The layer->draw_buf will follow up the areas to refresh by reshape draw buffer to specified area.

See
https://github.com/lvgl/lvgl/blob/adfd8c2e8bb151fabc63d0e38fe9db48b5a53d88/src/core/lv_refr.c#L641-L658

So the area gets updated firstly and then layer->draw_buf adjusted accordingly.

kisvegabor, 2024-01-15:
It's really redundant information now and buf_area could be replaced by a point mark the start x,y coordinates. And function could craft the actual area.

Guys, I feel like we are having too many last minute changes. Making the release now would not be the responsible decision. I recommend having a release candidate version today and make the real release next Monday. This week we wouldn't add any internal or API  changes. Just be sure everything is smooth and tested. 

It's a little bit of reputation loss to delay the release, but having a broken release would be even worse.

What do you think?

It was my responsibility to organize the work for the release and although it was a big challenge, I could do it better. After the release my top priority will be to figure out how to do things in more predicable, transparent and organized way.

GorgonMeducer, 2024-01-15:
> I recommend having a release candidate version today and make the real release next Monday. This week we wouldn't add any internal or API changes. Just be sure everything is smooth and tested.

I have no problem with this.

IAMMX, 2024-01-15:
>  I recommend having a release candidate version today and make the real release next Monday. This week we wouldn't add any internal or API changes. Just be sure everything is smooth and tested.

Sorry to hear the launch has been delayed. However, since the consideration has been postponed for a week, I think this week we should not add new features for the time being, and should focus more on testing the code and fixing bugs to make the release more stable.
Finally, I wish the v9 version can be released smoothly.💪

espzav, 2024-01-16:
Hi, please, is it supported to use monochromatic screen in LVGL9? I tried to `CONFIG_LV_COLOR_DEPTH_1=y` and I got compilation error: 
```
lvgl__lvgl/src/core/lv_refr.c:924:104: error:LV_COLOR_FORMAT_NATIVE' undeclared (first use in this function); did you mean 'LV_COLOR_FORMAT_UYVY'?
  924 |                                                           area_need_alpha ? LV_COLOR_FORMAT_ARGB8888 : LV_COLOR_FORMAT_NATIVE, &layer_area_act);
```

And second question: What is the right new callback instead of `disp_drv.set_px_cb`?

espzav, 2024-01-16:
@kisvegabor Thank you for the info. I am reworking the `esp_lvgl_port` to support both LVGL versions. The last thing missing is a monochromatic screen.

It seems that it will be more complicated now, I will try to find a solution tomorrow. The function `lv_rgb565_to_l1(buf, buf_size)` would be really helpful. 
Thank you.

kisvegabor, 2024-01-17:
>  It seems that it will be more complicated now, I will try to find a solution tomorrow. 

Note that `LV_COLOR_DEPTH 1` in v8 rendered in RGB332, which is also counter intuitive. Once we will have the A8 or L8 renderer in v9 it will be cleaner. After that having A1 and L1, doesn't seem to complicated either. The two issue I'm not sure about are
1. placing the bits next to each other is simpler (compared to placing them into columns) but I'm not sure if all controllers are supporting row format. 
2. starting with MSB or LSB could be an issue too.

> The function lv_rgb565_to_l1(buf, buf_size) would be really helpful.

Added in #5355

IAMMX, 2024-01-17:
Can you add a formatting function to **textarea?** For example, `lv_textarea_set_text_fmt`.
Because I need to set a default value for textarea depending on the situation. But there is no such formatting function, I need to `lv_snprintf` on one buff at a time, and then write it using `lv_textarea_set_text`. This is especially troublesome.

kisvegabor, 2024-01-18:
I think it can be added without any issues. Let's do it when v9 is released.

MouriNaruto, 2024-01-18:
@kisvegabor 

I'm care about the release date of LVGL v9 stable. I know LVGL v9 release candidate is announced but there is no git tag for that.

Kenji Mouri

[30 more replies]

closed 2024-01-29 by kisvegabor