lvgl/lvgl #4945: obj cannot receive LV_EVENT_LONG_PRESSED

HongChao6, 2023-12-06:
Unable to receive LV_EVENT_LONG_PRESSED using following code.

`
static void event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * label = lv_event_get_user_data(e);

    switch(code) {
        case LV_EVENT_PRESSED:
            lv_label_set_text(label, "The last button event:\nLV_EVENT_PRESSED");
            break;
        case LV_EVENT_CLICKED:
            lv_label_set_text(label, "The last button event:\nLV_EVENT_CLICKED");
            break;
        case LV_EVENT_LONG_PRESSED:
            lv_label_set_text(label, "The last button event:\nLV_EVENT_LONG_PRESSED");
            break;
        case LV_EVENT_LONG_PRESSED_REPEAT:
            lv_label_set_text(label, "The last button event:\nLV_EVENT_LONG_PRESSED_REPEAT");
            break;
        default:
            break;
    }
}

void lv_example_event_2(void)
{
    lv_obj_t * btn = lv_button_create(lv_screen_active());
    lv_obj_set_size(btn, 100, 50);
    lv_obj_center(btn);

    lv_obj_t * btn_label = lv_label_create(btn);
    lv_label_set_text(btn_label, "Click me!");
    lv_obj_center(btn_label);

    lv_obj_t * info_label = lv_label_create(lv_screen_active());
    lv_label_set_text(info_label, "The last button event:\nNone");

    lv_obj_add_event(btn, event_cb, LV_EVENT_ALL, info_label);
}

`

kisvegabor, 2023-12-06:
Thank you for reporting it.  It's not working for me either. I assume that you are using the Eclipse project with SDL.

@bjsylvia I've found that the problem is that in `_lv_sdl_mouse_handler` ` lv_indev_read(indev);` is not called  while the mouse button is being pressed and LVGL cannot measure the pressing time and send long press event. 

To solve instead of deleting the timer we could just pause it when release happens and resume it on press. 

What do you think?

bjsylvia, 2023-12-06:
I have an idea: First, we should unify the behavior of reading input devices by standardizing the poll and timer mechanisms for input devices, such as processing a mouse button press only once when it is not moved. Secondly, we should standardize the logic for handling long press events, that is, starting a long press timer when a button is pressed and stopping it when it is released ？

kisvegabor, 2023-12-21:
Solved by #4949

closed 2023-12-21 by kisvegabor